mikakaraila/node-red-contrib-opcua #233: Issue with SIEMENS-NodeRED OPCUA Connection and Question Regarding NodeRED Client Certificate

halcyon0606, 2020-12-23:
Hello Mika,

Recently I started using a SIEMENS PLC model S7-1200, and I was testing out its OPC UA Server with the node red OPCUA client. If the security policy and mode are both set to none, connecting to this server has no issues. However, when the security policy and mode are set to any other policy/mode, the node returns the error: 
[image]
On TIA portal, I have already enabled the server to automatically accept all certificates, but the error still persists. What could cause the issue to occur? Is it related to SIEMENS side or NodeRED side or is it a misunderstanding on my part about how certificates work? 

Also, while we're on the topic of certificates, how is the local certificate and private key entry in the node client used? As in what should be entered here because it's not immediately apparent to me.
[image]

Thank you!

mikakaraila, 2020-12-23:
Did you check on server side that certificate is on rejected folder?
You can test perhaps moving it to trusted folder will it work then?
Client gives it´s own certificate to server and server decides it if it will trust or reject it. This can be also some timing problem.
Test bit more and provide some debug log files I could try to find where the problem lies...

halcyon0606, 2020-12-23:
I am not sure where to find the rejected folder in the TIA portal, I've tried to search up but I can't find the PKI folder that would contain this, any ideas? On another note, is there a possibility that it could be the client that rejects the server certificate? Because when I'm attempting to use UaExpert to connect to the SIEMENS PLC OPC UA Server, if the security 'Basic256Sha256-Sign and Encrypt' was used, there is an error:
[image]

mikakaraila, 2020-12-23:
Have you changed TIA Server hostname? Certificate contain Fully Qualified Domain Name and it´s not matching to hostname.
Problem is now on TIA and OPC UA Server certificate.

If you run UaExpert on TIA then you can try this:
[image]

halcyon0606, 2020-12-24:
So I resolved the issue with the BadCertificateHostnameInvalid, turns out I forgot to update the server certificate's URI after changing IP addresses, so now UaExpert does not have this error anymore. However, the node red client still cannot connect to the server. That being said, when connecting with UaExpert, there is still a warning of the server having a time difference of more than 30 minutes from local system time, is this relevant to the issue right now?
[image]

halcyon0606, 2020-12-24:
Update: After fixing the PLC's system time the node red opcua client can now connect without any issues, thank you and enjoy your holidays!

closed 2020-12-24 by halcyon0606