mikakaraila/node-red-contrib-opcua #234: Problem with writing multiple data

halcyon0606, 2020-12-29:
Hello Mika,

I am currently working on controlling a smart light using the OPCUA client but I have run into an issue. The process of lighting up the smart light is as such, there are 3 segments of the smart light, each segment has a nodeId that can be written to with a byte variable. When using the write multiple function to write to the addresses of the smart light segments, the smart light tends to only light up 2 segments, missing out the middle segment. Now, there is another method to light up the segments which is to write using either the byte array type or the string byte type, both different addresses. However, writing using this method does not seem to work at all. Thus, I am wondering what could the issue be, is the first problem related to latency? If you require more information on this, please do let me know!

mikakaraila, 2021-01-04:
Can you provide flow file? Are you trying to write one byte / multiple bytes? It depends also how server uses the actual bytes.

If each segment has own nodeId then writing them one by one or write multiple should not cause any problem on OPC UA level.
So I expect problem is on server how it uses bytes.

halcyon0606, 2021-01-05:
I am trying to write multiple bytes, one byte to each nodeId. Each segment has their own nodeId, however, there is also a nodeId that groups all three segments together.

One other thing to mention is that when UaExpert controls the nodeId of the segments group, it is able to turn on all 3 segments simultaneously, but the client in NodeRED can only control 2 segments, missing out the middle one. 

The flow used in node red:
[
    {
        "id": "17f0aba3.52a7d4",
        "type": "tab",
        "label": "Flow 4",
        "disabled": false,
        "info": ""
    },
    {
        "id": "b8f20b1d.cd6b08",
        "type": "inject",
        "z": "17f0aba3.52a7d4",
        "name": "",
        "topic": "ns=1;s=IOLM/Port 2/Attached Device/PDO Fields/V_PdT_Segment/888.330242.V_PdT_Segment.TI_PD_Color_Segment2.12.3",
        "payload": "",
        "payloadType": "date",
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "x": 360,
        "y": 280,
        "wires": [
            [
                "203eb3a4.5366fc"
            ]
        ]
    },
    {
        "id": "497512b1.750d4c",
        "type": "OpcUa-Client",
        "z": "17f0aba3.52a7d4",
        "endpoint": "a9d85020.79c69",
        "action": "writemultiple",
        "deadbandtype": "a",
        "deadbandvalue": 1,
        "time": "100",
        "timeUnit": "ms",
        "certificate": "n",
        "localfile": "",
        "localkeyfile": "",
        "securitymode": "None",
        "securitypolicy": "None",
        "name": "",
        "x": 800,
        "y": 280,
        "wires": [
            []
        ]
    },
    {
        "id": "203eb3a4.5366fc",
        "type": "function",
        "z": "17f0aba3.52a7d4",
        "name": "SMART_LIGHT",
        "func": "msg.payload=[];\n\nmsg.payload[0]=\n{\n    nodeId:\"ns=1;s=IOLM/Port 2/Attached Device/PDO Fields/V_PdT_Segment/888.330242.V_PdT_Segment.TI_PD_Color_Segment1.8.3\",\n    datatype:\"Byte\",\n    value:4\n};\n\nmsg.payload[1]=\n{\n    nodeId:\"ns=1;s=IOLM/Port 2/Attached Device/PDO Fields/V_PdT_Segment/888.330242.V_PdT_Segment.TI_PD_Color_Segment2.12.3\",\n    datatype:\"Byte\",\n    value:4\n};\n\n\nmsg.payload[2]=\n{\n    nodeId:\"ns=1;s=IOLM/Port 2/Attached Device/PDO Fields/V_PdT_Segment/888.330242.V_PdT_Segment.TI_PD_Color_Segment3.0.3\",\n    datatype:\"Byte\",\n    value:4\n};\n\nreturn msg;\n\n\n",
        "outputs": 1,
        "noerr": 0,
        "x": 580,
        "y": 280,
        "wires": [
            [
                "497512b1.750d4c"
            ]
        ]
    },
    {
        "id": "a9d85020.79c69",
        "type": "OpcUa-Endpoint",
        "endpoint": "opc.tcp://192.168.0.250:4840",
        "secpol": "None",
        "secmode": "None",
        "login": false
    }
]

halcyon0606, 2021-01-05:
Just in case you are interested in knowing what device is hosting the OPC UA Server, it is the PNIO/MOD Fieldbus module ICE3-8IOL-G65L-V1D from Pepperl+Fuchs, I shall provide the link below:
https://www.pepperl-fuchs.com/global/en/classid_4996.htm?view=productdetails&prodid=96759#documents

mikakaraila, 2021-01-05:
Hmm, I expect you are building write multiple items wrong way. I will re-factor your flow.

halcyon0606, 2021-01-05:
I have compared the nodeId in UaExpert with the nodeId in Node RED and they are the same. Furthermore, it works when writing individually to the segment that does not light up when writing multiple. Could it be a problem of writing multiple bytes? As all the data written to the nodeIds are bytes.

mikakaraila, 2021-01-05:
Can you provide screenshot from UaExpert that will show Dataview and those nodeIds as I made from Prosys server?

halcyon0606, 2021-01-06:
Here are the screenshots:
[image]
[image]
A case has been tested in which the middle segment which is missed out is written to twice, once more after the third segment, the light manages to light up but this is undesired. Just to illustrate the write multiple function code would look something like this:

msg.payload=[];

msg.payload[0]=
{
    nodeId:"ns=1;s=IOLM/Port 2/Attached Device/PDO Fields/V_PdT_Segment/888.330242.V_PdT_Segment.TI_PD_Color_Segment1.8.3",
    datatype:"Byte",
    value:4
};

msg.payload[1]=
{
    nodeId:"ns=1;s=IOLM/Port 2/Attached Device/PDO Fields/V_PdT_Segment/888.330242.V_PdT_Segment.TI_PD_Color_Segment2.12.3",
    datatype:"Byte",
    value:4
};

msg.payload[2]=
{
    nodeId:"ns=1;s=IOLM/Port 2/Attached Device/PDO Fields/V_PdT_Segment/888.330242.V_PdT_Segment.TI_PD_Color_Segment3.0.3",
    datatype:"Byte",
    value:4
};

msg.payload[3]=
{
    nodeId:"ns=1;s=IOLM/Port 2/Attached Device/PDO Fields/V_PdT_Segment/888.330242.V_PdT_Segment.TI_PD_Color_Segment2.12.3",
    datatype:"Byte",
    value:4
};

return msg;

This works but unfortunately this is undesired.

mikakaraila, 2021-01-06:
BUG is on the server side, you have to report it to P&F.

As client uses writemultiple to Prosys UA Simulation server all items are written.

BUT in your case the last item on array is NOT written by server as I started to expect as I proposed this test.

halcyon0606, 2021-01-08:
I will report this to P&F and will get back to you on what they think the issue could be. In the meanwhile, would it be possible to include some new things so that I can test this extensively? Just so I can cover all grounds. Firstly, it would be great if the server is able to create a node with the "Byte" datatype so that I can test writemultiple on it. In addition to that, could the ByteString datatype be included in the OPCUA item node so that this datatype can be written to? And lastly, could you provide an example of how would one write a ByteArray datatype in node red using the item node (I'm not getting the expected output)? Anyways, I appreciate all the help and support! Thank you so much!

mikakaraila, 2021-01-18:
I made first other updates. I will take a look new/missing features.

mikakaraila, 2021-01-18:
Arrays are a lot more work, I have implemented those to our server. This will need changes multiple places in the code.
Perhaps later...

halcyon0606, 2021-01-19:
Ahhh I see, thank you so much for the implementation! Anyways, the arrays are not urgent, just a suggestion as P+F server also has array type data.

mikakaraila, 2021-01-19:
I just released 0.2.95 but ByteString does not work... have to study it later today more

mikakaraila, 2021-01-27:
Array support is now done v0.2.98

mikakaraila, 2021-02-08:
Any news? I will close this as no activity for a while. Open if needed.

closed 2021-02-08 by mikakaraila