mikakaraila/node-red-contrib-opcua #235: Add Variable on OPCUA Server

henjoe, 2021-01-10:
Hi Mika, 

Happy New year to you.
I just wanna ask if how can I achieve this using the OPCUAServer node. 

I want to have an OPC Server with variable "Counter" that has value of 25. 
Upon reading the documentation, I just need to pass this JSON message to OPCUA server node.
[image]

and here is my flow:
[image]

and this is the function node:
[image]

When I press the Inject, I am hoping that there will be a "Counter" variable on the OPC UA server with value equal to 25. Unfortunately nothing is in the server, the nodered has NO error too.
[image]

Hoping for your kind help.

THank you.

With kind regards,
Henjoe

henjoe, 2021-01-10:
Hi Mika,

I found the variable Counter and I am successfully setting the value to 25. It's under the Vendor Name (it's kinda weird)
but anyways the command works. 

[image]

However, of course I want to add some "MyOwnVariables". What command should I use? 
So far I tried to use the "addEquipment" command.
[image]

There were NO errors, but nothing happens on the server.

Thank you Mika.

With kind regards,
Henjoe

mikakaraila, 2021-01-11:
I expect there should be example how to use server node. There are "commands" that you can inject:
- restartOPCUAServer
- addEquipment
- addPhysicalAsset
- setFolder
- addFolder
- addVariable
- installHistorian
- deleteNode

henjoe, 2021-01-11:
Hi Mika,

Thanks for the reply.
On the information tab only 4 commands are there, never knew about
-addVariable
-addfolders
And etc commands.

By the wat, do you have some example on how to use these commands? (Add variable,add folder) because it's not in the information tab.

Again, thank you Mika.

With kind regards,
Henjoe

mikakaraila, 2021-01-12:
Ack, I will update help.

Here is quick help:
- setFolder parameter folder name == msg.topic; // Example topic format ns=1;s=FolderName (note: uses this as parentFolder)
- addFolder parameter folder name == msg.topic // Example topic format ns=1s=NewFolder (parent is objects if not set by setFolder)
- addVariable parameter variable ==msg.topic // Example topic format ns=1;s=VariableName;datatype=Double
- installHistorian parameter variable == msg.topic // Example topic format ns=1;s=VariableName;datatype=Double
- deleteNode parameter nodeId == msg.payload // Example payload format ns=1;s=VariableName

mikakaraila, 2021-01-12:
Added to v0.2.92

henjoe, 2021-01-12:
Hi Mika,

Highly appreciated all your efforts! 

Cheers!

With kind regards,
Henjoe

henjoe, 2021-01-13:
Hi Mika, 
Just one more question before we close this issue.

Will it be possible to set the "sourceTimestamp"? when setting a value of a variable?
[image]

for example:
msg = { 
payload: {
messageType: "Variable",
variableName: "Counter",
variableValue: 25,
timeStamp: unixTimestamp or any other format of timestamp
}
}

I hope this is possible on the OPC technology.

Thank you.

With kind regards,
Henjoe

mikakaraila, 2021-01-13:
OPC UA supports serverTimestamp and sourceTimestamp. I can add those at some point as optional property not mandatory.

closed 2021-01-13 by mikakaraila

henjoe, 2021-01-13:
Thanks Mika! 

Highly Appreciated! 

With kind regards,
Henjoe