mikakaraila/node-red-contrib-opcua #236: Certificate error: can't subscribe or read

fssantos95, 2021-01-11:
Hello Mika,

Today after the update to 0.2.91 i have been geting these certificates errors:
[image]

Befora that i just created a client and it was all good. My OPC server has no credentials.
Can you help?

fssantos95, 2021-01-11:
Already tried to delete certificates so that they regenerate but no success.

fssantos95, 2021-01-11:
[image]

mikakaraila, 2021-01-12:
This requires a bit more, node-opcua and node-opcua-pki contains config.js that is used with command to create certificate.
openssl needs many parameters and they must be correct and run in certain folder.

I will look requirements to create certificate for the client...

mikakaraila, 2021-01-12:
Found:
https://stackoverflow.com/questions/50328537/generate-certificate-for-opc-client

fssantos95, 2021-01-12:
But where do i get all those parametrs? Is there a reason why the node-red OPC certificate stopped working? I already uninstaled everything and still get the same error.....

mikakaraila, 2021-01-13:
Check first certificate folder exists:
c:/Users/admin.exporplas/.node-red/node_modules/node-opcua-client/certificates

Then there should be certificate file:
client_selfsigned_cert_2048.pem

Please report your findings.

mikakaraila, 2021-01-13:
There is breaking change in node-opcua v2.29.0 (downgrade to node-opcua v2.28.3 for a while).

OPCUAClient & OPCUAServer : the location of the default PKI is now created in .config folder or in the %APPDATA% folder instead of inside the node_module itself.

fssantos95, 2021-01-13:
> Check first certificate folder exists:
> c:/Users/admin.exporplas/.node-red/node_modules/node-opcua-client/certificates
> 
> Then there should be certificate file:
> client_selfsigned_cert_2048.pem
> 
> Please report your findings.

I didn't have that folder Mika.

> There is breaking change in node-opcua v2.29.0 (downgrade to node-opcua v2.28.3 for a while).
> 
> OPCUAClient & OPCUAServer : the location of the default PKI is now created in .config folder or in the %APPDATA% folder instead of inside the node_module itself.

I downgraded node-red-contrib-opcua to v0.2.89 and node-opcua to v2.82.3 and all is working again. Should i wait to update?

mikakaraila, 2021-01-13:
I need couple of days to make migration, but you can install manually older node-opcua v2.28.3

mikakaraila, 2021-01-13:
I made new version 0.2.93 that uses node-opcua v2.28.3
I need more time to do all changes.

fssantos95, 2021-01-14:
> I made new version 0.2.93 that uses node-opcua v2.28.3
> I need more time to do all changes.

Ty Mika!

mikakaraila, 2021-01-18:
Latest version is now v0.2.94 but it uses also new node-opcua that uses certificate manager for client.
There are some topics (issues) on node-opcua so I expect I have to update node-opcua later.

closed 2021-01-18 by mikakaraila