mikakaraila/node-red-contrib-opcua #238: After update, client cannot connect to server when there is security policy and mode selected

halcyon0606, 2021-01-13:
Hello Mika,

After updating the package to the latest version, I am now unable to use the node red OPCUA client to connect to the server. The following are the options selected:

[image]

[image]

And the error message is as shown:

[image]

After looking at the error message, I've noticed that the usual directory for the client's self signed certificate is no longer there. Hope that a fix can be added soon! Thank you.

mikakaraila, 2021-01-13:
Strange, can you check node-opcua version? 

Certificate file should be in pem format, now it seems to use .pem.csr

Check one more time your environment!

halcyon0606, 2021-01-13:
The node-opcua version is given as 6.14.8. Is that the current version that is being used?

[image]

Also, the node-opcua-client file does not have the certificate folder anymore, is that also an issue?

[image]

mikakaraila, 2021-01-13:
OK that explains why client is not anymore working.

node-opcua should be v2.28.3

Your command shows npm version. 
Use npm list command:

C:\Users\karaimi\.node-red>npm list
node-red-project@0.0.1 C:\Users\karaimi\.node-red
+-- async@3.2.0
+-- chalk@4.1.0
| +-- ansi-styles@4.2.1
| | +-- @types/color-name@1.1.1
| | `-- color-convert@2.0.1
| |   `-- color-name@1.1.4
| `-- supports-color@7.1.0
|   `-- has-flag@4.0.0
+-- circular-json@0.5.9
+-- flatted@3.1.0
+-- node-opcua@2.28.3
| +-- chalk@4.1.0 deduped
| +-- node-opcua-address-space@2.28.2

halcyon0606, 2021-01-13:
Ahh I see, sorry about that, my npm list output shows that node opcua version is 2.29.0

| +-- node-opcua@2.29.0
| | +-- chalk@4.1.0 deduped
| | +-- node-opcua-address-space@2.29.0
| | | +-- @types/lodash@4.14.167
| | | +-- async@3.2.0 deduped
| | | +-- chalk@4.1.0 deduped
| | | +-- dequeue@1.0.5

So if that's the case, do I have to regress to v2.28.3?

mikakaraila, 2021-01-13:
There are breaking change in new v2.29.0

OPCUAClient & OPCUAServer : the location of the default PKI is now created in .config folder or in the %APPDATA% folder instead of inside the node_module itself.

halcyon0606, 2021-01-13:
Oh, I see, in that case do I wait for you to fix the default PKI path in a newer version or is there a way for me to fix this currently?

mikakaraila, 2021-01-13:
Just install older node-opcua v2.28.3 then it should work again. I need couple of days to fix code to use new node-opcua.

mikakaraila, 2021-01-13:
I made quick fix, 0.2.93 uses node-opcua v2.28.3 

I need more time to do all needed changes.

mikakaraila, 2021-01-18:
Now new version 0.2.94 uses node-opcua 2.29.0 let´s see if it helps now.

closed 2021-01-18 by mikakaraila