mikakaraila/node-red-contrib-opcua #239: Subscription RangeError

jelmerthijsen, 2021-01-13:
Hello Mika,

I am trying to subscribe to an item from a Siemens Simatic Comfort panel configured as OPC UA server. This raises the error:
RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 105. Received 4_294_967_365

When i subscribe to a Prosys OPC UA Simulation Server, it works fine. Do you know what can cause this error? I am using V0.2.92.

[subscription error.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/5807371/subscription.error.txt)

[image]

mikakaraila, 2021-01-13:
Do you have any more log to show?

Error message is new for me, I haven´t seen it before.

jelmerthijsen, 2021-01-13:
UaExpert view:

[image]

Attributes view of the Objects, Wincc comfort panel and tags objects:

[image]
[image]
[image]

mikakaraila, 2021-01-13:
What is item Einde cyclus (nodeId)? Can you find and show it with UaExpert?

It contains something that is not decoded correctly...

mikakaraila, 2021-01-14:
Can you find corresponding nodeId and look what it contains like this:

[image]

jelmerthijsen, 2021-01-14:
This is the tag i was trying to subscribe to. All the tags from the tags folder are empty:

[image]
Also the views folder is empty

mikakaraila, 2021-01-14:
OK this case is then something that Etienne (node-opcua) should look...

jelmerthijsen, 2021-01-14:
Oke, thanks for your time :)

closed 2021-01-14 by jelmerthijsen

ziceva, 2022-05-13:
I know this is a closed issue, but I also came upon this same problem, and I think I have an insight that might help others, that, like me, got here from googling (I am still looking for a solution, but I think I have pinpointed the problem)

The problem seems to be that some servers (in my case a very old version of a Siemens OPCUA server running on a HMI Panel) do not compute the length of the DataChangeNotification body as it can be decoded without the length (it is not free-length, and UAExpert has no issue in decoding it). These servers just fill the field with 0xFFFFFFFF.

I did find some more info here: https://www.eclipse.org/lists/milo-dev/msg00176.html

This seems to indeed be an issue in the underlying node-opcua, not in node-red-contrib-opcua (again, posting here because of google).

mikakaraila, 2022-05-16:
Thanks, I have noticed v1.02 <> v1.04 have some differences in diagnostics.
Siemens old version uses static message sequence number in the header, it will produce warning.