mikakaraila/node-red-contrib-opcua #240: What´s the difference between using multiple Item Nodes and using the function node

HappyP4ncake, 2021-01-13:
Hello, 

i want to know if there are any differences in the connection if one of these two options is used: 

Option1: Using Item Nodes, for each value a seperate Item Node is used and all Item Nodes are connected to one single client.

Option2: Using A function node and node.send({...}) where all paths to the Nodes (on the OPC UA Server) are send to one single client. 

Afaik messages from the second option are similar to the first... 

Additionally: When the Client Node is deactivated, is there still a connection between the Client and the Server? I had the issue that the connection between Node-RED and the OPC UA Server blocked other PLC´s and Clients, which caused a shutdown of production (I used one client only). Only by killing the Node-RED Docker Container the connection was terminated. 

Thanks!

mikakaraila, 2021-01-13:
Depends on client node action. If you use READ it will cause load to server, SUBSCRIPTION allows server to publish data.
In other words: READ will wait until server sends response. Subscription will be received as server will publish data.

Client keeps session open to server but it should not cause anything on server side.

There must be some resource leak or blocking service (bug) on server code if client can cause PLC to block execution / other clients. It will be good to run long stress testing in other than production environment.

mikakaraila, 2021-01-13:
Another case can be if you have wrong nodeIds and it repeats those read actions server can have problem inside error handling.
Good to do also some negative testing.

HappyP4ncake, 2021-01-14:
Thank you. I´ll come back after testing and checking the OPC UA Server configuration.

HappyP4ncake, 2021-01-25:
Update: I could not reproduce the error/ problem in an testing environment yet. 

During regular maintenance we did some testing with a subset of datapoints (subset <<original set) and no problems occurred. We testet with READ and SUBSCRIBE. Testing with false NodeIDs didn´t cause any problems either. It worked pretty good.  After maintenance we deleted the Client Node in Node-RED to ensure we won´t cause a shutdown of production. Surprisingly we still had a session open... 
It might be a problem with Docker or permissions, as the connection is not terminated properly.

mikakaraila, 2021-01-25:
OK then focus on session testing. Run client up/down several times. What happens on server side?
Can it release old sessions?

HappyP4ncake, 2021-02-17:
I still couldn´t find out what caused the problem which drives me crazy....

We put a subset into production and to this day no problems occurred. Extending the set of datapoints and observe the behavior.
If Client is deactivated or deleted, the number of sessions is constant. After reactivating or run Client in short period of Time the number increases by one and will go down by one again. So it seems to release old sessions. So I guess it is a problem on the server side. Maybe I find the root cause...

Anyway I really appreciate your support. Thanks

closed 2021-06-10 by mikakaraila