mikakaraila/node-red-contrib-opcua #242: Dinamically change Endpoint

LucaCarminati-it, 2021-01-18:
Hi to all,

I connect to the previously open issue :  

#59 
#92 

I have tryed to do what is explained in the issue #59 but every time that I try to change the Endpoint while there is a connection yet estabilished it works and I am able to switch and connect dinamically two or more (existing) device.
If I try to insert a new or change the Endpoint of a connection that is not yet estabilished, or is closed, it doesn't work, and in the debug node I receive the error :

TypeError: Cannot read property 'close' of undefined

Does somebody find a solution about that ?
I attach my node flow :

`[{"id":"6a1449cb.e08848","type":"tab","label":"Test OPC","disabled":false,"info":""},{"id":"cb81f563.4ae138","type":"debug","z":"6a1449cb.e08848","name":"","active":true,"tosidebar":true,"console":true,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":790,"y":240,"wires":[]},{"id":"3ca4393b.270696","type":"inject","z":"6a1449cb.e08848","name":"Read","props":[{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"ns=3;s=\"opctest\".\"var\"","x":310,"y":220,"wires":[["2c0201c3.5dd42e"]]},{"id":"6c05a536.d4203c","type":"function","z":"6a1449cb.e08848","name":"","func":"msg.topic=\"Reconnect\";\nmsg.action=\"reconnect\";\nmsg.OpcUaEndpoint = {\n credentials: {},\n endpoint: 'opc.tcp://172.31.224.21:4840',\n //endpoint: 'opc.tcp://172.31.224.21:4840/simulation/',\n securityPolicy: 'None',\n securityMode: 'NONE',\n login: false,\n user: undefined,\n password: undefined \n}\n\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","x":460,"y":180,"wires":[["2c0201c3.5dd42e"]]},{"id":"13565688.dc6f29","type":"inject","z":"6a1449cb.e08848","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"","payloadType":"str","x":310,"y":180,"wires":[["6c05a536.d4203c"]]},{"id":"938471da.1bf71","type":"function","z":"6a1449cb.e08848","name":"","func":"msg.topic=\"Reconnect\";\nmsg.action=\"reconnect\";\nmsg.OpcUaEndpoint = {\n credentials: {},\n endpoint: 'opc.tcp://172.31.224.2:4840',\n //endpoint: 'opc.tcp://172.31.224.2:4840/simulation/',\n securityPolicy: 'None',\n securityMode: 'NONE',\n login: false,\n user: undefined,\n password: undefined \n}\n\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","x":460,"y":300,"wires":[["2c0201c3.5dd42e"]]},{"id":"e456f2c9.24d4c","type":"inject","z":"6a1449cb.e08848","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"","payloadType":"str","x":310,"y":300,"wires":[["938471da.1bf71"]]},{"id":"29107ebe.5c7c72","type":"inject","z":"6a1449cb.e08848","name":"Read","props":[{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"ns=3;s=\"opcua\".\"plc2\"","x":310,"y":260,"wires":[["2c0201c3.5dd42e"]]},{"id":"2c0201c3.5dd42e","type":"OpcUa-Client","z":"6a1449cb.e08848","endpoint":"94949813.9a64e8","action":"read","deadbandtype":"a","deadbandvalue":1,"time":10,"timeUnit":"s","certificate":"n","localfile":"","localkeyfile":"","securitymode":"None","securitypolicy":"None","name":"","x":640,"y":240,"wires":[["cb81f563.4ae138"]]},{"id":"94949813.9a64e8","type":"OpcUa-Endpoint","endpoint":"opc.tcp://0.0.0.0:0000","secpol":"None","secmode":"None","login":false}]`

Thanks to all!

mikakaraila, 2021-01-18:
Small bug in the code. As client is not connected then session is undefined and it cannot be used to close session.
Fixed to next version. I will test as there are more changes than just this one...

LucaCarminati-it, 2021-01-18:
> Small bug in the code. As client is not connected then session is undefined and it cannot be used to close session.
> Fixed to next version. I will test as there are more changes than just this one...

Thanks a lot for the reply !
So, sorry for the misunderstanding, this 'bug' is fixed in the version 0.2.93 or will be fixed with a next version 0.X.XX ?

mikakaraila, 2021-01-18:
I just published v0.2.94 and fixed also event listeners will be removed as new client will be created.

closed 2021-01-18 by mikakaraila