mikakaraila/node-red-contrib-opcua #243: - [error] [OpcUa-Client:d289327c.3f175] TypeError: Cannot read property 'match' of undefined

hj91, 2021-01-18:
18 Jan 17:24:29 - [info] Node-RED version: v0.20.5
18 Jan 17:24:29 - [info] Node.js  version: v13.14.0
18 Jan 17:24:29 - [info] Linux 4.15.0-130-generic x64 LE
18 Jan 17:24:29 - [info] Loading palette nodes
18 Jan 17:24:30 - [warn] rpi-gpio : Raspberry Pi specific node set inactive
18 Jan 17:24:30 - [warn] rpi-gpio : Cannot find Pi RPi.GPIO python library
18 Jan 17:24:31 - [info] Settings file  : /home/harshad/.node-red/settings.js
18 Jan 17:24:31 - [info] Context store  : 'default' [module=memory]
18 Jan 17:24:31 - [info] User directory : /home/harshad/.node-red
18 Jan 17:24:31 - [warn] Projects disabled : editorTheme.projects.enabled=false
18 Jan 17:24:31 - [info] Flows file     : /home/harshad/.node-red/start
18 Jan 17:24:31 - [info] Server now running at http://127.0.0.1:1880/
18 Jan 17:24:31 - [warn] 

18 Jan 17:24:31 - [info] Starting flows
18 Jan 17:24:31 - [info] [OpcUa-Client:922a9ac9.bdcfe8] 	Local 'own' certificate is NOT used.
18 Jan 17:24:31 - [info] [OpcUa-Client:d289327c.3f175] 	Local 'own' certificate is NOT used.
18 Jan 17:24:31 - [info] Started flows
18 Jan 17:24:31 - [info] [mqtt-broker:45d57805.32a478] Connected to broker: mqtt://192.168.1.10:1883
18 Jan 17:24:42 - [error] [OpcUa-Client:d289327c.3f175] TypeError: Cannot read property 'match' of undefined

Installed the latest version by using npm install node-red-contrib-opcua

I am getting issues while writing to the OPC server

hj91, 2021-01-18:
This is the sample code

[
    {
        "id": "ee848bb2.635a78",
        "type": "tab",
        "label": "Flow 1",
        "disabled": false,
        "info": ""
    },
    {
        "id": "d162599f.3def08",
        "type": "inject",
        "z": "ee848bb2.635a78",
        "name": "",
        "topic": "ns=1;s=Control Relay number 0.",
        "payload": "false",
        "payloadType": "bool",
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "x": 141,
        "y": 215,
        "wires": [
            [
                "d289327c.3f175"
            ]
        ]
    },
    {
        "id": "27ef455c.53cfda",
        "type": "debug",
        "z": "ee848bb2.635a78",
        "name": "",
        "active": false,
        "tosidebar": true,
        "console": true,
        "tostatus": true,
        "complete": "payload",
        "targetType": "msg",
        "x": 989,
        "y": 107,
        "wires": []
    },
    {
        "id": "922a9ac9.bdcfe8",
        "type": "OpcUa-Client",
        "z": "ee848bb2.635a78",
        "endpoint": "389acf53.a2add",
        "action": "subscribe",
        "deadbandtype": "a",
        "deadbandvalue": 1,
        "time": 10,
        "timeUnit": "s",
        "certificate": "n",
        "localfile": "",
        "localkeyfile": "",
        "name": "",
        "x": 650,
        "y": 395,
        "wires": [
            [
                "571061c6.54b8c",
                "768ee612.fad558"
            ]
        ]
    },
    {
        "id": "571061c6.54b8c",
        "type": "debug",
        "z": "ee848bb2.635a78",
        "name": "",
        "active": true,
        "tosidebar": true,
        "console": true,
        "tostatus": true,
        "complete": "payload",
        "targetType": "msg",
        "x": 935,
        "y": 455,
        "wires": []
    },
    {
        "id": "7f192d74.c850b4",
        "type": "debug",
        "z": "ee848bb2.635a78",
        "name": "",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": true,
        "complete": "payload",
        "targetType": "msg",
        "x": 634,
        "y": 120,
        "wires": []
    },
    {
        "id": "ca2787c4.939748",
        "type": "inject",
        "z": "ee848bb2.635a78",
        "name": "",
        "topic": "ns=1;s=Control Relay number 0.",
        "payload": "true",
        "payloadType": "bool",
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "x": 303,
        "y": 434,
        "wires": [
            [
                "922a9ac9.bdcfe8"
            ]
        ]
    },
    {
        "id": "b591ec01.a8a64",
        "type": "mqtt in",
        "z": "ee848bb2.635a78",
        "name": "",
        "topic": "/test/opcua0",
        "qos": "2",
        "datatype": "auto",
        "broker": "45d57805.32a478",
        "x": 110,
        "y": 60,
        "wires": [
            [
                "7f192d74.c850b4",
                "6dc58b2e.1c2bb4"
            ]
        ]
    },
    {
        "id": "768ee612.fad558",
        "type": "mqtt out",
        "z": "ee848bb2.635a78",
        "name": "",
        "topic": "opcua/relayon",
        "qos": "2",
        "retain": "false",
        "broker": "45d57805.32a478",
        "x": 999,
        "y": 372,
        "wires": []
    },
    {
        "id": "5b44b931.5c60a8",
        "type": "switch",
        "z": "ee848bb2.635a78",
        "name": "",
        "property": "payload",
        "propertyType": "msg",
        "rules": [
            {
                "t": "eq",
                "v": "",
                "vt": "str"
            }
        ],
        "checkall": "true",
        "repair": false,
        "outputs": 1,
        "x": 400,
        "y": 260,
        "wires": [
            []
        ]
    },
    {
        "id": "d289327c.3f175",
        "type": "OpcUa-Client",
        "z": "ee848bb2.635a78",
        "endpoint": "9933c3d6.b3c67",
        "action": "write",
        "deadbandtype": "a",
        "deadbandvalue": 1,
        "time": 10,
        "timeUnit": "s",
        "certificate": "n",
        "localfile": "",
        "localkeyfile": "",
        "securitymode": "None",
        "securitypolicy": "None",
        "name": "",
        "x": 633.5,
        "y": 214,
        "wires": [
            [
                "27ef455c.53cfda"
            ]
        ]
    },
    {
        "id": "6dc58b2e.1c2bb4",
        "type": "change",
        "z": "ee848bb2.635a78",
        "name": "",
        "rules": [
            {
                "t": "set",
                "p": "topic",
                "pt": "msg",
                "to": "ns=1;s=Control Relay number 0 .",
                "tot": "str"
            }
        ],
        "action": "",
        "property": "",
        "from": "",
        "to": "",
        "reg": false,
        "x": 423.5,
        "y": 162,
        "wires": [
            []
        ]
    },
    {
        "id": "389acf53.a2add",
        "type": "OpcUa-Endpoint",
        "z": "",
        "endpoint": "opc.tcp://192.168.1.2:4840/",
        "secpol": "None",
        "secmode": "NONE",
        "login": false
    },
    {
        "id": "45d57805.32a478",
        "type": "mqtt-broker",
        "z": "",
        "name": "",
        "broker": "192.168.1.10",
        "port": "1883",
        "clientid": "",
        "usetls": false,
        "compatmode": true,
        "keepalive": "60",
        "cleansession": true,
        "birthTopic": "opcua/relayon",
        "birthQos": "0",
        "birthRetain": "false",
        "birthPayload": "ns=1;s=Control Relay number 0.",
        "closeTopic": "",
        "closeQos": "0",
        "closePayload": "",
        "willTopic": "",
        "willQos": "0",
        "willPayload": ""
    },
    {
        "id": "9933c3d6.b3c67",
        "type": "OpcUa-Endpoint",
        "z": "",
        "endpoint": "opc.tcp://192.168.1.2:4840",
        "secpol": "None",
        "secmode": "None",
        "login": false
    }
]

mikakaraila, 2021-01-18:
You flow contain 2 endpoint, I expect you should have this one at write node (original was missing / after 4840):

[image]

hj91, 2021-01-18:
[image]
i added the / but still getting error...

hj91, 2021-01-18:
[
    {
        "id": "c448d578.f2f368",
        "type": "tab",
        "label": "MQTT_to_OPCUA_server-read-write",
        "disabled": false,
        "info": ""
    },
    {
        "id": "18741b49.3c0d45",
        "type": "OpcUa-Client",
        "z": "c448d578.f2f368",
        "endpoint": "b53b9f5f.3cdbf",
        "action": "subscribe",
        "deadbandtype": "p",
        "deadbandvalue": 1,
        "time": "10",
        "timeUnit": "s",
        "certificate": "n",
        "localfile": "",
        "localkeyfile": "",
        "name": "esp32_opcua_read",
        "x": 657,
        "y": 735,
        "wires": [
            [
                "a0cac204.00363",
                "6d25bdd3.9ccf84",
                "14604232.65a93e",
                "c4a7844f.3b0a58"
            ]
        ]
    },
    {
        "id": "6d25bdd3.9ccf84",
        "type": "debug",
        "z": "c448d578.f2f368",
        "name": "",
        "active": false,
        "tosidebar": true,
        "console": true,
        "tostatus": true,
        "complete": "payload",
        "targetType": "msg",
        "x": 1312,
        "y": 309,
        "wires": []
    },
    {
        "id": "cc3faf11.4660b",
        "type": "mqtt in",
        "z": "c448d578.f2f368",
        "name": "",
        "topic": "ns=1;s=Control Relay number 0.",
        "qos": "0",
        "datatype": "auto",
        "broker": "5a29021e.ef250c",
        "x": 149,
        "y": 689,
        "wires": [
            [
                "f09fbf3d.0af2b"
            ]
        ]
    },
    {
        "id": "a0cac204.00363",
        "type": "mqtt out",
        "z": "c448d578.f2f368",
        "name": "",
[... 311 lines cut ...]
        "keepalive": "60",
        "cleansession": true,
        "birthTopic": "opcua/relayon",
        "birthQos": "0",
        "birthRetain": "false",
        "birthPayload": "ns=1;s=Control Relay number 0.",
        "closeTopic": "",
        "closeQos": "0",
        "closePayload": "",
        "willTopic": "",
        "willQos": "0",
        "willPayload": ""
    },
    {
        "id": "c2357342.c319a",
        "type": "OpcUa-Endpoint",
        "z": "",
        "endpoint": "opc.tcp://192.168.1.2:4840/",
        "secpol": "None",
        "secmode": "None",
        "login": false
    }
]

This flow worked after i added an opcua-item node before write opcua-client node for writing. The reading operation does not require any extra node

closed 2021-01-18 by hj91