mikakaraila/node-red-contrib-opcua #244: [Question] variable created is gone everytime OPCUA server start

henjoe, 2021-01-18:
Hi Mika,

I don't if this is normal. But what I simply want is if added a variable in the OPCUA server, I want to retain those variable everytime I restarted the OPCUA server.

What I notice is, everytime my OPCUAserver restarted, all Folders and Variables that I have created is also gone.

Is this normal? 

Thanks Mika.

With kind regards,
Henjoe

mikakaraila, 2021-01-18:
Yes of course as server re-starts it will start beginning. Very natural. It is not restart if it keeps variables.
Why you need restart actually. What is action you want to make?

henjoe, 2021-01-18:
Hi Mika,

That's make sense :) 
Thanks for clarifying this.

With kind regards,
Henjoe

closed 2021-06-10 by mikakaraila