mikakaraila/node-red-contrib-opcua #245: Subscribe not working correctly in v0.2.94

halcyon0606, 2021-01-19:
Hello Mika,

I've verified all other client actions in the new update and everything seems to be working fine. However, the subscribe action has an issue. It is able to detect whenever a value has changed, but now it does not seem to display the changed value, even in the full msg object I am not able to find the value of the node. 

Here is the output from the subscribed client in the node red debug log:
[image]

In any case, thank you for your hard work thus far!

mikakaraila, 2021-01-19:
Subscription is fixed to 0.2.95, it didn´t show value correctly. I will take a look to browse later today.

mikakaraila, 2021-01-19:
What was actual problem with browse?

For me it seems to work:
[image]

mikakaraila, 2021-01-19:
One difference could be that now node-opcua returns browseName as object:

[image]

halcyon0606, 2021-01-19:
For browse it is not the Browser Node that has the error, that node provides correct output. It is the Client's Browse Action that I have an issue with, the following is the JSON flow for the client node with Browse Action:

[{"id":"e612ebe3.8c4e38","type":"OpcUa-Client","z":"2d630062.17aca","endpoint":"c06892f0.58fd5","action":"browse","deadbandtype":"a","deadbandvalue":1,"time":10,"timeUnit":"s","certificate":"n","localfile":"","localkeyfile":"","securitymode":"None","securitypolicy":"None","name":"","x":640,"y":440,"wires":[["8478cb37.731108"]]},{"id":"159f2bae.342ac4","type":"inject","z":"2d630062.17aca","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"ns=0;i=84","payload":"","payloadType":"date","x":360,"y":440,"wires":[["e612ebe3.8c4e38"]]},{"id":"8478cb37.731108","type":"debug","z":"2d630062.17aca","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":910,"y":440,"wires":[]},{"id":"4fe2cd82.bcf154","type":"comment","z":"2d630062.17aca","name":"Browsing root node","info":"","x":310,"y":400,"wires":[]},{"id":"c06892f0.58fd5","type":"OpcUa-Endpoint","endpoint":"opc.tcp://MSI:54880","secpol":"Basic256Sha256","secmode":"SignAndEncrypt","login":false}]

mikakaraila, 2021-01-19:
OK thanks for clarifying. Now I can look correct code.

closed 2021-01-19 by mikakaraila

halcyon0606, 2021-01-19:
Ahh I see, but why does it return so many items? I was under the impression that if you browsed the root node, you should be getting back the three folders and not subsequent browsing of the browsed folders.

mikakaraila, 2021-01-19:
Browse action is recursive so it will go through all nodeIds under given nodeId.

halcyon0606, 2021-01-19:
Alright, that explains a lot. Thank you so much!