mikakaraila/node-red-contrib-opcua #246: Configuration Error certificate and private key doesn't match

manciown, 2021-01-19:
Hi Sir,
I use this package for a long time... today I made the update from node-red-opcua-client version 0.2.52 to last version 0.2.95.
My Enviroment is Raspberry Pi 3b raspbian version 8 "jessie", node-red version 0.20.7 .

I've got many flows that now are not working properly anymore... The Opc-Ua-Client node is showing error: 
"Configuration error : the certificate and the private key do not match !".

Running manually node-red I got some error in the console log:
:verify :24    [NODE-OPCUA-E01] Configuration error : the certificate and the private key do not match !
:verify :25    please check the configuration of the OPCUA Server
:verify :26    privateKey=  /home/pi/.config/NodeOPCUA-Default-nodejs/certificates/own/private/private_key.pem
:verify :27    certificateManager.privateKey=  /home/pi/.config/NodeOPCUA-Default-nodejs/certificates/own/private/private_key.pem
:verify :28    certificateFile=  /home/pi/.config/NodeOPCUA-Default-nodejs/certificates/client_certificate.pem

How can I solve?

mikakaraila, 2021-01-19:
Here is solution:
https://github.com/node-opcua/node-opcua/issues/930

At least case is closed.

HansDamph, 2021-01-20:
> Hi Sir,
> I use this package for a long time... today I made the update from node-red-opcua-client version 0.2.52 to last version 0.2.95.
> My Enviroment is Raspberry Pi 3b raspbian version 8 "jessie", [...]

I have the same problem since udpdatet to 0.2.95 yesterday. I deletet everything. Did reinstall of package via palette manager. It still does not connect "Configuration error : the certificate and the private key do not match !".
It did work with previous version (maybe 2.8x, not really sure).
I still can connect with UAexpert.

Did node-opcua/node-opcua#930 solve your issue?

Is there anything else i can do?

Thank you mikakaraila for your great work!

mikakaraila, 2021-01-20:
Did you check and delete from /home/pi/.config/NodeOPCUA-Default-nodejs/

Folder: NodeOPCUA-Default-nodejs ??

HansDamph, 2021-01-21:
I am not using raspbian. My collegue (who has access to file system) found following keys:

/usr/src/node-red/.config/NodeOPCUA-Default-nodejs/certificates/trusted/certs/MyOPCUAClientApplicationName[3e3bb97402ddf25ce46893f3e3c3c4acb365f9].pem
/usr/src/node-red/.config/NodeOPCUA-Default-nodejs/certificates/client_certificate.pem
/usr/src/node-red/.config/NodeOPCUA-Default-nodejs/certificates/own/private/private_key.pem
/usr/src/node-red/.config/NodeOPCUA-Default-nodejs.wontwork/certificates/trusted/certs/MyOPCUAClientApplicatioame[d09016b5522d7aad06e65a95caefea099c31872c].pem
/usr/src/node-red/.config/NodeOPCUA-Default-nodejs.wontwork/certificates/client_certificate.pem
/usr/src/node-red/.config/NodeOPCUA-Default-nodejs.wontwork/certificates/own/private/private_key.pem

After deleting, a new key appeared but the error message stays the same.

mikakaraila, 2021-01-21:
Then update issue on node-opcua as it runs certificate verification.

Is certificate own self-signed or official?

IbramT15, 2021-01-21:
Hi,
i'm using the opcua-library on a nodered, which is running on a PLC. I don`t have access to the plc root folder, so i'm not able to delete the certificates.
My application is not working with the newest update of the opcua-node.
[image]

Is there any possibility to fix it, without an access to the root?

jumo-sballing, 2021-01-22:
Same here. Deleting the complete folder `./nodered/.config/NodeOPCUA-Default-nodejs` doesn't help. At the next start the folder is generated again. But the error "certificate and private key doesn't match" is still displayed.

mikakaraila, 2021-01-22:
Cannot fix issue on node-red-contrib-opcua -> report to node-opcua as it makes automated certificate management.
There is something wrong... but should folder be under /home/pi/.config/ ???

jumo-sballing, 2021-01-25:
The .config folder is in the home path of the Node-RED user. At my system the node-red user has the name nodered and this is it home path.
Thank you for your hint. I will ask at node-opcua for help.

jumo-sballing, 2021-01-25:
I've done a few tests and with node-opcua v2.31.0 the error is gone. It would be very nice of you to raise the package version of node-opcua. If you want I can send you a pull request.

mikakaraila, 2021-01-25:
Merged and new v0.2.97 available

closed 2021-01-25 by mikakaraila

jumo-sballing, 2021-01-25:
You're great. Thank you very much for your fast response.

paulsalibe, 2021-03-17:
I have version 0.2.113 and this is happening to me now. I delete /home/pi/.config/node-opcua-default-nodejs folder and it works for a second but immediately comes back.

mikakaraila, 2021-03-17:
Strange, code uses hostname that is coming from the computer and same hostname should be also inside certificate.
Can you study it more with openssl to look what you have inside certificate?
Or did you install node-opcua-contrib-opcua earlier with some other network settings? Then it could have caused mismatch.

paulsalibe, 2021-03-17:
I installed node-red-contrib-opcua yesterday on a new Raspberry Pi but imported a flow from a Raspberry Pi running a previous version. Not sure if this would have caused issues...

danmarsic, 2021-03-17:
I have the same problem after updating some older version of node-red-contrib-opcua to the newest version 0.2.113 (Error: node-opcua-e01 configuration error the certificate and the private key do not match). I am using nodered as a Docker container running on a Debian Linux IoT Gateway from the Company TTTech. 
I cannot find the folder ./nodered/.config/NodeOPCUA-Default-nodejs on my system in order to delete it:
**- NodeRED container** : I entered the container from terminal (docker exec -it containername sh) and there I have only the following: 
drwxr-xr-x    1 node-red root          9216 Feb  4 08:24 node_modules (and )
-rw-rw-r--    1 node-red root          1038 Feb  4 08:23 package.json
**-Docker-Volume:** Nodered persistent volume data is bind mounted to the host file system under ./opt/data/var/lib/docker/volumes/nodered/_data/.  Here I have te following folders:
[image]
In the folder node_modules I have many files starting the name with node-opcua... but I file or folder with the name node-opcua-default-nodejs.
[image]

Can some please gove me an advice on how to locate /NodeOPCUA-Default-nodejs in order to delete it?

mikakaraila, 2021-03-17:
At windows:
C:\Users\%USER%\AppData\Roaming\node-opcua-default-nodejs

Linux, I expect, didn´t find from the certificate manager code (no Linux running just now to check it online):
/home/$USER

But you can give root folder for the certificate manager to use your own location for the PKI folde.

jumo-sballing, 2021-03-18:
Still have problems with this sometimes. I found two solutions. Solution 1 is the best:
1. Create your own certificates, put it to your Node-RED instance and add the path to it in your opc-ua client node
2. Delete all of your OPC-Client Nodes from all of your Flows **incl.** the config of the client nodes. Then shut down node-red and clear the auto generated certificates.Windows: C:\Users%USER%\AppData\Roaming\node-opcua-default-nodejs Linux:

danmarsic, 2021-03-18:
I also created my own certificates  add the path to it in your opc-ua client node. Now it works. Thanks. 😊

However, I have problems with using OpcUa-Method Node.
I am using RFID module with OPC UA Server supporting AutoID Companion specification to read and write RFID HF Tags.
https://www.turck.de/en/product/00000040000381bb0003003a
 
The method ReadTag has six input arguments, where the first and the last one are not basic data types but structs specific to AutoID Companion specification.
[image]

On the other hand, OpcUa-Method Node in NoreRED has only three input arguments of a basic data type. I tried only to enter Offest and Length input arguments because all the other are optional and use default values. Unfortunately, this does not work.
[image]

@mikakaraila is it even possible to call method ReadTag with your OpcUa-Method Node?

mikakaraila, 2021-03-18:
Not yet, follow this one:
https://github.com/mikakaraila/node-red-contrib-opcua/issues/270

I have to add support for extension object.