mikakaraila/node-red-contrib-opcua #247: FULL Deploy shuts down OPC Server. Restart does not work.

youngkzy, 2021-01-19:
FULL Deploy shuts down OPC Server. Shuts down OPC Server.  "restartOPCUAServer" command does not work.  You have to FULL DEPLOY to reset. or Restart Flows.   So for now, I am careful not to use "FULL DEPLOY" or "FLOW DEPLOY".
[image]

mikakaraila, 2021-01-19:
Do you have latest version? I quickly tested and my flow works event after full deploy.

mikakaraila, 2021-01-19:
Please check you node-red console. In some cases I have seen that server port is still in use. 

Operating system has not released port and it causes that port is already in use and server cannot use it.

mikakaraila, 2021-01-19:
Use default values or similar to those that are "placeholders".

[image]

mikakaraila, 2021-01-20:
Updated dialog and help:

[image]

closed 2021-01-20 by mikakaraila