mikakaraila/node-red-contrib-opcua #248: Question on OPC UA Server Node

youngkzy, 2021-01-19:
@mikakaraila,
   I can write to the OPC Server using example 

msg = { payload : { "messageType" : "Variable", "variableName": "Counter", "variableValue": msg.payload }};

but when I try to use an OPC UA Client Node set to WRITE, it does not respond with any error, but does not WRITE the value to the tag. 

Node-RED v1.2.7  
Node.JS v14.15.4
node-red-contrib-opcua 0.2.93
Windows_NT 10.0.162999 x64 LE

```[{"id":"259b1b07.22d9b4","type":"function","z":"b76887cc.51a578","name":"Translation","func":"msg = { payload : { \"messageType\" : \"Variable\", \"variableName\": \"Counter\", \"variableValue\": msg.payload }};\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","x":470,"y":420,"wires":[["5166e9e5.9ea0b","2651fb5d.5dff4c"]]},{"id":"a13e828f.6d6dc8","type":"inject","z":"b76887cc.51a578","name":"restartOPCUAServer","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":"","topic":"","payload":"{ \"opcuaCommand\": \"restartOPCUAServer\" }","payloadType":"json","x":311.5,"y":486,"wires":[["2651fb5d.5dff4c"]]},{"id":"fc02345f.b48c8","type":"inject","z":"b76887cc.51a578","name":"addEquipment","props":[{"p":"payload","v":"{ \"opcuaCommand\": \"addEquipment\", \"nodeName\": \"Machine\" }","vt":"json"},{"p":"topic","v":"","vt":"string"}],"repeat":"","crontab":"","once":false,"topic":"","payload":"{ \"opcuaCommand\": \"addEquipment\", \"nodeName\": \"Machine\" }","payloadType":"json","x":320.5,"y":528,"wires":[["2651fb5d.5dff4c"]]},{"id":"22aed14c.ebf2fe","type":"inject","z":"b76887cc.51a578","name":"deleteNode PhysicalAsset2","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":"","topic":"","payload":"{ \"opcuaCommand\": \"deleteNode\", \"nodeId\": \"ns=1;s=PhysicalAsset2\" }","payloadType":"json","x":280.5,"y":571,"wires":[["2651fb5d.5dff4c"]]},{"id":"b5c2f4ea.b45ac","type":"inject","z":"b76887cc.51a578","name":"addPhysicalAsset","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":"","topic":"","payload":"{ \"opcuaCommand\": \"addPhysicalAsset\", \"nodeName\": \"PhysicalAsset\" }","payloadType":"json","x":309,"y":612,"wires":[["2651fb5d.5dff4c"]]},{"id":"7cba5107.bf17f","type":"inject","z":"b76887cc.51a578","name":"addPerson","props":[{"p":"payload","v":"{ \"opcuaCommand\": \"addPerson\", \"nodeName\": \"Person\" }","vt":"json"},{"p":"topic","v":"","vt":"string"}],"repeat":"","crontab":"","once":false,"topic":"","payload":"{ \"opcuaCommand\": \"addPerson\", \"nodeName\": \"Person\" }","payloadType":"json","x":338.5,"y":653,"wires":[["2651fb5d.5dff4c"]]},{"id":"5166e9e5.9ea0b","type":"debug","z":"b76887cc.51a578","name":"","active":false,"tosidebar":true,"console":false,"complete":"false","statusVal":"","statusType":"auto","x":790,"y":420,"wires":[]},{"id":"2651fb5d.5dff4c","type":"OpcUa-Server","z":"b76887cc.51a578","port":"53880","name":"","endpoint":"opc.tcp://127.0.0.1:53880","autoAcceptUnknownCertificate":true,"registerToDiscovery":true,"maxNodesPerBrowse":"","maxNodesPerHistoryReadData":"","maxNodesPerHistoryReadEvents":"","maxNodesPerHistoryUpdateData":"","maxNodesPerRead":"","maxNodesPerWrite":"0","maxNodesPerMethodCall":"","maxNodesPerRegisterNodes":"","maxNodesPerNodeManagement":"","maxMonitoredItemsPerCall":"","maxNodesPerHistoryUpdateEvents":"","maxNodesPerTranslateBrowsePathsToNodeIds":"","x":600,"y":520,"wires":[["faca9f93.8972d"]]},{"id":"7c78c000.27c19","type":"OpcUa-Item","z":"b76887cc.51a578","item":"ns=1;s=Counter","datatype":"Int32","value":"","name":"","x":330,"y":280,"wires":[["a10e0f63.185948"]]},{"id":"a10e0f63.185948","type":"OpcUa-Client","z":"b76887cc.51a578","endpoint":"d651828c.3518b","action":"read","deadbandtype":"a","deadbandvalue":1,"time":10,"timeUnit":"s","certificate":"n","localfile":"","localkeyfile":"","securitymode":"None","securitypolicy":"None","name":"","x":600,"y":280,"wires":[["2a40b5ae.41c732"]]},{"id":"3448170.4d9c26a","type":"inject","z":"b76887cc.51a578","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"","payloadType":"str","x":110,"y":280,"wires":[["7c78c000.27c19"]]},{"id":"2a40b5ae.41c732","type":"debug","z":"b76887cc.51a578","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","x":790,"y":280,"wires":[]},{"id":"faca9f93.8972d","type":"debug","z":"b76887cc.51a578","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","x":790,"y":520,"wires":[]},{"id":"b421fc3b.460538","type":"function","z":"b76887cc.51a578","name":"INCREMENT","func":"var count = context.get('FlowCount')||0;\nif (count>10)\n{\n    count=0;\n}\nelse{\n    count++;\n}\ncontext.set('FlowCount',count);\nmsg.payload=count;\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","x":290,"y":420,"wires":[["259b1b07.22d9b4"]]},{"id":"f3b4c376.afb178","type":"inject","z":"b76887cc.51a578","name":"INC","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"","payloadType":"str","x":110,"y":420,"wires":[["b421fc3b.460538"]]},{"id":"764974b6.aafa1c","type":"inject","z":"b76887cc.51a578","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"","payloadType":"str","x":110,"y":340,"wires":[["51f63353.0f6ca4"]]},{"id":"33d2e7dc.4e2e88","type":"OpcUa-Item","z":"b76887cc.51a578","item":"ns=1;s=Counter","datatype":"Int32","value":"","name":"","x":410,"y":340,"wires":[["177109d2.f5ed0e"]]},{"id":"177109d2.f5ed0e","type":"OpcUa-Client","z":"b76887cc.51a578","endpoint":"d651828c.3518b","action":"write","deadbandtype":"a","deadbandvalue":1,"time":10,"timeUnit":"s","certificate":"n","localfile":"","localkeyfile":"","securitymode":"None","securitypolicy":"None","name":"","x":600,"y":340,"wires":[["804915c6.c2b788"]]},{"id":"51f63353.0f6ca4","type":"function","z":"b76887cc.51a578","name":"INCREMENT","func":"var count = context.get('FlowCount')||0;\nif (count>10)\n{\n    count=0;\n}\nelse{\n    count++;\n}\ncontext.set('FlowCount',count);\nmsg.payload=count;\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","x":250,"y":340,"wires":[["33d2e7dc.4e2e88","e3d8b1b9.6cdee8"]]},{"id":"804915c6.c2b788","type":"debug","z":"b76887cc.51a578","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","x":800,"y":340,"wires":[]},{"id":"e3d8b1b9.6cdee8","type":"debug","z":"b76887cc.51a578","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","x":450,"y":300,"wires":[]},{"id":"d651828c.3518b","type":"OpcUa-Endpoint","endpoint":"opc.tcp://127.0.0.1:53880","secpol":"None","secmode":"None","login":false}]```

mikakaraila, 2021-01-19:
Should work, your flow does not define type for the Counter. Use inject JSON this way:

[image]

mikakaraila, 2021-01-19:
Couple of errors in the flow, server endpoint is as in the help resource path not opc.tcp:// endpoint.
I will update help.

[image]

mikakaraila, 2021-01-19:
Here is fixed flow:

[{"id":"af23020e.02ad9","type":"tab","label":"Flow 1","disabled":false,"info":""},{"id":"d651828c.3518b","type":"OpcUa-Endpoint","z":"","endpoint":"opc.tcp://127.0.0.1:53881","secpol":"None","secmode":"None","login":false},{"id":"e302be0d.0adfa","type":"OpcUa-Endpoint","z":"","endpoint":"opc.tcp://127.0.0.1:53881","secpol":"None","secmode":"None","login":false},{"id":"259b1b07.22d9b4","type":"function","z":"af23020e.02ad9","name":"Translation","func":"msg = { payload : { \"messageType\" : \"Variable\", \"variableName\": \"Counter\", \"variableValue\": msg.payload }};\nreturn msg;","outputs":1,"noerr":0,"x":470,"y":420,"wires":[["5166e9e5.9ea0b"]]},{"id":"a13e828f.6d6dc8","type":"inject","z":"af23020e.02ad9","name":"restartOPCUAServer","topic":"","payload":"{ \"opcuaCommand\": \"restartOPCUAServer\" }","payloadType":"json","repeat":"","crontab":"","once":false,"onceDelay":"","x":311.5,"y":486,"wires":[["2651fb5d.5dff4c"]]},{"id":"5166e9e5.9ea0b","type":"debug","z":"af23020e.02ad9","name":"","active":true,"tosidebar":true,"console":false,"complete":"false","x":790,"y":420,"wires":[]},{"id":"2651fb5d.5dff4c","type":"OpcUa-Server","z":"af23020e.02ad9","port":"53881","name":"Local Server","endpoint":"","autoAcceptUnknownCertificate":true,"registerToDiscovery":true,"maxNodesPerBrowse":"","maxNodesPerHistoryReadData":"","maxNodesPerHistoryReadEvents":"","maxNodesPerHistoryUpdateData":"","maxNodesPerRead":"","maxNodesPerWrite":"0","maxNodesPerMethodCall":"","maxNodesPerRegisterNodes":"","maxNodesPerNodeManagement":"","maxMonitoredItemsPerCall":"","maxNodesPerHistoryUpdateEvents":"","maxNodesPerTranslateBrowsePathsToNodeIds":"","x":590,"y":520,"wires":[["faca9f93.8972d"]]},{"id":"7c78c000.27c19","type":"OpcUa-Item","z":"af23020e.02ad9","item":"ns=1;s=MyCounter","datatype":"Int32","value":"","name":"MyCounter","x":710,"y":120,"wires":[["a10e0f63.185948"]]},{"id":"a10e0f63.185948","type":"OpcUa-Client","z":"af23020e.02ad9","endpoint":"e302be0d.0adfa","action":"read","deadbandtype":"a","deadbandvalue":1,"time":10,"timeUnit":"s","certificate":"n","localfile":"","localkeyfile":"","securitymode":"None","securitypolicy":"None","name":"Read","x":960,"y":120,"wires":[["2a40b5ae.41c732"]]},{"id":"3448170.4d9c26a","type":"inject","z":"af23020e.02ad9","name":"","topic":"","payload":"","payloadType":"str","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":500,"y":120,"wires":[["7c78c000.27c19"]]},{"id":"2a40b5ae.41c732","type":"debug","z":"af23020e.02ad9","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":1180,"y":120,"wires":[]},{"id":"faca9f93.8972d","type":"debug","z":"af23020e.02ad9","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","x":770,"y":520,"wires":[]},{"id":"b421fc3b.460538","type":"function","z":"af23020e.02ad9","name":"INCREMENT","func":"var count = context.get('FlowCount')||0;\nif (count>10)\n{\n count=0;\n}\nelse{\n count++;\n}\ncontext.set('FlowCount',count);\nmsg.payload=count;\nreturn msg;","outputs":1,"noerr":0,"x":290,"y":420,"wires":[["259b1b07.22d9b4"]]},{"id":"f3b4c376.afb178","type":"inject","z":"af23020e.02ad9","name":"INC","topic":"","payload":"","payloadType":"str","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":110,"y":420,"wires":[["b421fc3b.460538"]]},{"id":"764974b6.aafa1c","type":"inject","z":"af23020e.02ad9","name":"","topic":"","payload":"","payloadType":"str","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":450,"y":200,"wires":[["51f63353.0f6ca4"]]},{"id":"33d2e7dc.4e2e88","type":"OpcUa-Item","z":"af23020e.02ad9","item":"ns=1;s=MyCounter","datatype":"Int32","value":"","name":"MyCounter","x":810,"y":200,"wires":[["177109d2.f5ed0e"]]},{"id":"177109d2.f5ed0e","type":"OpcUa-Client","z":"af23020e.02ad9","endpoint":"d651828c.3518b","action":"write","deadbandtype":"a","deadbandvalue":1,"time":10,"timeUnit":"s","certificate":"n","localfile":"","localkeyfile":"","securitymode":"None","securitypolicy":"None","name":"Write","x":950,"y":200,"wires":[["804915c6.c2b788"]]},{"id":"51f63353.0f6ca4","type":"function","z":"af23020e.02ad9","name":"INCREMENT","func":"var count = context.get('FlowCount')||0;\nif (count>10)\n{\n count=0;\n}\nelse{\n count++;\n}\ncontext.set('FlowCount',count);\nmsg.payload=count;\nreturn msg;","outputs":1,"noerr":0,"x":610,"y":200,"wires":[["33d2e7dc.4e2e88","e3d8b1b9.6cdee8"]]},{"id":"804915c6.c2b788","type":"debug","z":"af23020e.02ad9","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":1180,"y":200,"wires":[]},{"id":"e3d8b1b9.6cdee8","type":"debug","z":"af23020e.02ad9","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":810,"y":280,"wires":[]},{"id":"bdfb767e.3c1798","type":"inject","z":"af23020e.02ad9","name":"add MyCounter","topic":"ns=1;s=MyCounter;datatype=Int32","payload":"{\"opcuaCommand\":\"addVariable\"}","payloadType":"json","repeat":"","crontab":"","once":false,"onceDelay":"","x":360,"y":540,"wires":[["2651fb5d.5dff4c"]]}]

youngkzy, 2021-01-19:
please add the 3 "`" little "single preceding quotes" before and after the JSON.  As it is,it gives syntax error from import copy/paste.

youngkzy, 2021-01-19:
I do not understand what is meant by the OPC Server Property "Endpoint"   the example gives a value of  "UA/SimpleNodeRedServer"... Do I just use that? or is there something with reference to my configuration?  no explanation and term Endpoint is ambiguous.

youngkzy, 2021-01-19:
@mikakaraila ,
   Thanks for that correction,  Figured it out from your screen shots.  Still need some explaination on the term OPC Server Property "Endpoint", but went through and added deliberate new MyCounter and both means of updating the tag works.  I also updated OPC UA to newest version.

mikakaraila, 2021-01-20:
Endpoint is actually ResourcePath. code build up actual endpoint this way:

"opc.tcp://localhost:" + **PORT** + "**/ResourcePath/**"

closed 2021-01-20 by mikakaraila