mikakaraila/node-red-contrib-opcua #249: Create new folder in OPC UA Server

NicolaiWrt, 2021-01-20:
Hello Mika,

I have problems with the setFolder-Command. I created two more Objects under Equipment, there should be also some new variables.
Therefore I send a setFolder-Command to the Server Node, but the next variable is also written in VendorName.
[image]
[image]

How can I create the new Variable under my new objects? (Megomat 1 and 2)
Here my Code:
[{"id":"5aede970.315ee","type":"tab","label":"Flow 1","disabled":false,"info":""},{"id":"99b874a5.e4e818","type":"OpcUa-Server","z":"5aede970.315ee","port":"53880","name":"","endpoint":"UA/SimpleNodeRedServer","autoAcceptUnknownCertificate":true,"registerToDiscovery":false,"maxNodesPerBrowse":0,"maxNodesPerHistoryReadData":0,"maxNodesPerHistoryReadEvents":0,"maxNodesPerHistoryUpdateData":0,"maxNodesPerRead":0,"maxNodesPerWrite":0,"maxNodesPerMethodCall":0,"maxNodesPerRegisterNodes":0,"maxNodesPerNodeManagement":0,"maxMonitoredItemsPerCall":0,"maxNodesPerHistoryUpdateEvents":0,"maxNodesPerTranslateBrowsePathsToNodeIds":0,"x":300,"y":420,"wires":[["4a6897b1.d2d7a8"]]},{"id":"feb0639e.10e988","type":"inject","z":"5aede970.315ee","name":"{ \"opcuaCommand\": \"addEquipment\",  \"nodeName\": \"Megomat 1\" }","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"{ \"opcuaCommand\": \"addEquipment\",  \"nodeName\": \"Megomat\" }","payloadType":"json","x":320,"y":40,"wires":[["99b874a5.e4e818"]]},{"id":"4a6897b1.d2d7a8","type":"debug","z":"5aede970.315ee","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":470,"y":420,"wires":[]},{"id":"e4eba1d0.42dc98","type":"inject","z":"5aede970.315ee","name":"{ \"opcuaCommand\": \"setFolder\" } \"Megomat 1\"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"\"ns=1;s=Megomat1\"","payload":"{ \"opcuaCommand\": \"setFolder\" }","payloadType":"json","x":260,"y":160,"wires":[["99b874a5.e4e818"]]},{"id":"d546a956.fddca8","type":"inject","z":"5aede970.315ee","name":"{ \"opcuaCommand\": \"addEquipment\",  \"nodeName\": \"Megomat 2\" }","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"{ \"opcuaCommand\": \"addEquipment\",  \"nodeName\": \"Megomat\" }","payloadType":"json","x":320,"y":80,"wires":[["99b874a5.e4e818"]]},{"id":"60eb41fc.5cb61","type":"inject","z":"5aede970.315ee","name":"{ \"opcuaCommand\": \"addVariable\" } ns=1;s=Test\";datatype=Int32","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"ns=1;s=Test\";datatype=Int32","payload":"{ \"opcuaCommand\": \"addVariable\" }","payloadType":"json","x":310,"y":120,"wires":[["99b874a5.e4e818"]]},{"id":"6d6fb9ae.36f908","type":"inject","z":"5aede970.315ee","name":"{ \"opcuaCommand\": \"addVariable\" } ns=1;s=Test_Value\";datatype=Int32","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"ns=1;s=Test_Value\";datatype=Int32","payload":"{ \"opcuaCommand\": \"addVariable\" }","payloadType":"json","x":330,"y":200,"wires":[["99b874a5.e4e818"]]}]

Thanks in advance for you help

Best regards

Nicolai

mikakaraila, 2021-01-20:
Works:

[image]

Updated flow file:
[{"id":"5aede970.315ee","type":"tab","label":"Flow 1","disabled":false,"info":""},{"id":"99b874a5.e4e818","type":"OpcUa-Server","z":"5aede970.315ee","port":"53881","name":"","endpoint":"UA/SimpleNodeRedServer","autoAcceptUnknownCertificate":true,"registerToDiscovery":false,"maxNodesPerBrowse":0,"maxNodesPerHistoryReadData":0,"maxNodesPerHistoryReadEvents":0,"maxNodesPerHistoryUpdateData":0,"maxNodesPerRead":0,"maxNodesPerWrite":0,"maxNodesPerMethodCall":0,"maxNodesPerRegisterNodes":0,"maxNodesPerNodeManagement":0,"maxMonitoredItemsPerCall":0,"maxNodesPerHistoryUpdateEvents":0,"maxNodesPerTranslateBrowsePathsToNodeIds":0,"x":1060,"y":40,"wires":[["4a6897b1.d2d7a8"]]},{"id":"feb0639e.10e988","type":"inject","z":"5aede970.315ee","name":"Add Equiment: Megomat","topic":"","payload":"{\"opcuaCommand\":\"addEquipment\",\"nodeName\":\"Megomat\"}","payloadType":"json","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":510,"y":40,"wires":[["99b874a5.e4e818"]]},{"id":"4a6897b1.d2d7a8","type":"debug","z":"5aede970.315ee","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","x":1230,"y":40,"wires":[]},{"id":"e4eba1d0.42dc98","type":"inject","z":"5aede970.315ee","name":"Set Folder: Megomat1","topic":"ns=1;s=Megomat1","payload":"{\"opcuaCommand\":\"setFolder\"}","payloadType":"json","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":520,"y":160,"wires":[["99b874a5.e4e818"]]},{"id":"d546a956.fddca8","type":"inject","z":"5aede970.315ee","name":"Add Equipment: Megomat","topic":"","payload":"{\"opcuaCommand\":\"addEquipment\",\"nodeName\":\"Megomat\"}","payloadType":"json","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":510,"y":80,"wires":[["99b874a5.e4e818"]]},{"id":"60eb41fc.5cb61","type":"inject","z":"5aede970.315ee","name":"Add Variable: Test","topic":"ns=1;s=Test;datatype=Int32","payload":"{\"opcuaCommand\":\"addVariable\"}","payloadType":"json","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":530,"y":120,"wires":[["99b874a5.e4e818"]]},{"id":"6d6fb9ae.36f908","type":"inject","z":"5aede970.315ee","name":"Add Variable: Test_Value","topic":"ns=1;s=Test_Value;datatype=Int32","payload":"{\"opcuaCommand\":\"addVariable\"}","payloadType":"json","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":510,"y":200,"wires":[["99b874a5.e4e818"]]},{"id":"396ed080.13625","type":"comment","z":"5aede970.315ee","name":"Objects as default folder","info":"","x":210,"y":120,"wires":[]},{"id":"1d68ba68.daab46","type":"comment","z":"5aede970.315ee","name":"set folder to Megomat1","info":"","x":200,"y":160,"wires":[]},{"id":"acbf22d5.25f0a","type":"comment","z":"5aede970.315ee","name":"Test_Value under Megomat1","info":"","x":220,"y":200,"wires":[]},{"id":"c6ca9470.7fe018","type":"comment","z":"5aede970.315ee","name":"Megomat1","info":"As equipments are auto renumerated with increment 1","x":160,"y":40,"wires":[]},{"id":"129b7930.762c97","type":"comment","z":"5aede970.315ee","name":"Megomat2","info":"As equipments are auto renumerated with increment 1","x":160,"y":80,"wires":[]}]

NicolaiWrt, 2021-01-21:
Hey Mika,

thanks for your fast answer.
I try to import your code, but NodeRed gives me an error.
[image]
I use NodeRed on Windows 10.

Another thing:
Do you have a manuall for writing data to the server?

Thanks in advance.
Best regards

Nicolai

mikakaraila, 2021-01-21:
Copy & paste have removed something.  Rename SERVER.txt SERVER.json

[SERVER.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/5851537/SERVER.txt)

[image]

Writing depends on your nodeId & type, you can use UaExpert or client node with node-red.

mikakaraila, 2021-01-21:
Just a final screenshot: change value with UaExpert and subscription will show value on node-red:

[image]

closed 2021-01-21 by mikakaraila

NicolaiWrt, 2021-01-21:
Thank you really much.

jhonnyPalas, 2023-04-27:
Hi everyone,

if you wanted to create a tree structure with folders like :

Folder 1:
        subfolder1_child_1
        subfolder1_child_2

Folder 2:
        subfolder2_child_1
        subfolder2_child_2

how should I do?

mikakaraila, 2023-04-27:
Trial & error or use addFolder / setFolder by default addFolder will set new folder as parent folder.

Look example from OPCUA-TEST-NODES.json

Adito5393, 2025-01-10:
One solution to achiving that tree sctructure is to setFolder to the root Objects before adding other folders (e.g. "Folder 2"):

```bash
// Set Objects folder as the parent
var setObjectsFolder = {
    payload: { "opcuaCommand": "setFolder" },
    topic: "ns=0;i=85"
};
...
// Set to root before adding othe Folders
messages.push(setObjectsFolder)
messages.push(addFolder2)
```