mikakaraila/node-red-contrib-opcua #250: OpcUa-Method with only one input argument send array of size 2

HansDamph, 2021-01-21:
Hi friends. I am trying to get Data from a machine via opc ua. It works with UAexpert so far and i do not know why i do not get the output.
What i think might be the problem, that even i select only one input argument the log shows array of size 2.

Here is my configuration:
[image]
Output from node-red:
[image]

From UAexpert. I am not quite sure if is selected the correct object and method ID:
[image]

Any help would be very much appreciated!!

mikakaraila, 2021-01-21:
OK, I will check.

mikakaraila, 2021-01-21:
This was my earlier test flow:

[image]

mikakaraila, 2021-01-21:
Now outputArguments[] will be copied into msg.payload[]

[image]

mikakaraila, 2021-01-21:
Please test with 0.2.96 I added there check for inputArguments

HansDamph, 2021-01-25:
Thank you for your very fast response and solution. It does work now with 0.2.96

closed 2021-01-25 by HansDamph