mikakaraila/node-red-contrib-opcua #253: Problem after upgrading

sandumarius, 2021-01-22:
After I upgraded the node to last version i cannot connect to OPC-UA server because of self certificatates. I have the following settings:
[image]

I generated the certificates using: openssl req -newkey rsa:1024 -nodes -keyout private_key.pem -x509 -out client_selfsigned_cert_1024.pem

sandumarius, 2021-01-22:
solved by moving the generated certificates to ~/.config/NodeOPCUA-Default-nodejs/certificates

sandumarius, 2021-01-22:
:38    [NODE-OPCUA-W04] The public/private key pair uses a key length which is equal or lower than 1024 bits.
 ...                                                                 OPCUA version 1.04 requires that security key length are greater or equal to 2048 bits.
 ...                                                                 The client is operating at risk.
20:33:12.039Z :verify                        :78    [NODE-OPCUA-W06] The certificate subjectAltName does not match the server applicationUri
 ...                                                                 Please regenerate a specific certificate that matches your server applicationUri

mikakaraila, 2021-01-23:
Use 2048 as key size

https://www.digicert.com/kb/ssl-support/openssl-quick-reference-guide.htm

closed 2021-01-25 by mikakaraila