mikakaraila/node-red-contrib-opcua #255: OPCUA WRITE BYTE ARRAY

cazzin88, 2021-01-26:
[image]
[image]
[image]

Goodmorning,
i must write a byte array with OpcUa (see images).

I created an variabile array, i inserted into variable the values and I moved this variabile into "msg.payload", but this method perhaps in not correct.

What is the correct method?

Thanks

mikakaraila, 2021-01-26:
OPC UA Array must be defined as OPC UA DataValue. It is not yet implemented. It contains mush more properties than just javascript array. And it must used typed array. I will look some day... I have implemented this to other code all 1-3 dimension arrays.

cazzin88, 2021-01-26:
ok Thanks. I'll try another solution.

bye

mikakaraila, 2021-01-26:
Sorry, it is implemented, but need some fixes... let´s see. It should work with 1-dimension array.

mikakaraila, 2021-01-26:
Seems to work now.

Prosys UA Simulation server, started:
[image]

Node-red flow:
[image]

Byte and ByteArray written:
[image]

Result correct: Byte value 8 and ByteArray [5,4,3,2,1]
[image]

mikakaraila, 2021-01-26:
I have to cleanup code and add missing types etc.

mikakaraila, 2021-01-26:
Took bit more time:
[image]

Implemented & tested Arrays:
[image]

cazzin88, 2021-01-27:
Goodmorning Mr mikakaraila, thanks a lot!
can you notify when new release of "node-red-contrib-opcua" is available ?

thanks

mikakaraila, 2021-01-27:
Last night v0.2.98

cazzin88, 2021-01-27:
[image]
[image]
[image]
[image]

I must pass the value to write with msg.payload because is a value that i elaboration from a sql function. I don't must write a constant value.

Thanks

cazzin88, 2021-01-27:
If insert a constant values into Opcua-item node, the flow is ok!

[image]

cazzin88, 2021-01-27:
[image]

mikakaraila, 2021-01-27:
OK, I didn´t implement that for Array. Have to fix it...

cazzin88, 2021-01-27:
Ok , Thanks a lot.

cazzin88, 2021-01-27:
When you will have time, notify on next updated.

Thanks a lot e good work!

mikakaraila, 2021-01-27:
Fixed, I will test bit more, but now with empty payload (string) you can initialize with Item node value.
Then payload can be used to update value dynamically:

[image]

mikakaraila, 2021-01-27:
OK test with 0.2.99
Inject empty (string) payload will use value at item node value.
Inject string with payload with 0,0,48,0,0 will be used as value, connected as input to item node.

cazzin88, 2021-01-27:
Thanks a lot. I'll try this versione!

cazzin88, 2021-01-27:
Thanks a lot. I'll try this version!

closed 2021-01-28 by mikakaraila