mikakaraila/node-red-contrib-opcua #256: Invalid Endpoint

NicolaiWrt, 2021-01-26:
Hello Mika,

it's me again. The configuration you sent me worked perfectly.
Today I started the programm again, but now the OPCUA Client says:
[image]
The Endpoint should be correct. But I don't know how to fix the certificate.
It should be self-created by Node Red.
[image]
How can i reset this?

Best regards

Nicolai

mikakaraila, 2021-01-26:
Do you have latest 0.2.97 and node-opcua v2.31.0? 

There were some fixes on node-opcua and there is more discussion about certificates.

NicolaiWrt, 2021-01-26:
Yes I have the latest 0.2.97 OPC UA Node Version. 
What do you mean with v2.31.0?

mikakaraila, 2021-01-26:
As I wrote. Node-opcua is then also latest one

NicolaiWrt, 2021-01-26:
Sorry maybe I misunderstand something. 
I run Node Red locally on Windows. And I installed your Node with the integrated Palette Manager.
So I also installed your node-red-contrib-opcua node.
Should I also have to install the Node-opcua?
Or is it automatically installed?
How can I check if it's installed and when yes, which version is installed?
Thanks in advance.

mikakaraila, 2021-01-26:
No need to install anything manually. Have you changed IP or hostname?
What is Linux distro on ARM device?

You can try this: 
"Try to remove the "/Users/srogers/Library/Preferences/NodeOPCUA-Default-nodejs" folder so that default certificate get recreated .."
https://github.com/node-opcua/node-opcua/issues/930

mikakaraila, 2021-01-26:
Do you have resource path on server? Try to connect first with UaExpert!

NicolaiWrt, 2021-01-26:
Yes I have changed the Name of the Endpoint of the Server.
I will try to remove the folder.
I think a new certificate will help. 

Yes, I have a resource path on the server. I can connect to the Server with OPC UA. 
Only the Client Node doesn’t work.

I will test it tomorrow.
Thank you very much for your help.

NicolaiWrt, 2021-01-28:
Hey Mika,

I tried it... I removed the folder and the error is away, but it still does not work.
Sometimes it shows this error:
[image]

The connection with UA expert works fine.
I already reinstalled the whole opc-ua-node, also does not help.

Can I try something else?

This is the information I got about the ARM processor:
Linux 3.12.10+ arm LE

Best regards

Nicolai

mikakaraila, 2021-01-28:
Run with node-red -v flowfile.json
What is exact endpoint that you arre using with UaExpert?

Console output will tell us more, attach it her.

NicolaiWrt, 2021-01-28:
Hello Mika

thanks for fast reply.
I connect to this Endpoint in UA expert:

[image]

The Server Settings look like this:

[image]

How I can run "node-red -v flowfile.json"?

Here is the Log:
[Node-RED.log](https://github.com/mikakaraila/node-red-contrib-opcua/files/5887095/Node-RED.log)

mikakaraila, 2021-01-28:
From the command line...

mikakaraila, 2021-01-28:
At my computer it works:

[image]

[image]

mikakaraila, 2021-01-29:
Found it, my server code was not using given resource path, fix is coming.

Now correct resource path can be seen correctly:
[image]

NicolaiWrt, 2021-01-29:
Hello Mika,

I can not run it in a command line, because it's running on an arm linux device.
But not a "real" Linux, more like an embedded version.
So I have to compile a Node for NodeRed before I can load it on my device.
I will contact the manufacturer, how it is possible to control node-red via the console.

As I sad on my computer it also perfectly works.

The bug with the resource path: Could that be also the problem, why it's not working at my project?

Best regards

Nicolai

mikakaraila, 2021-01-29:
Yes, server code didn´t  add it. This caused endpoint didn´t match, wait for v0.2.100

I have full day now at the office.

NicolaiWrt, 2021-01-29:
Hello Mika,

thanks for your fast help.
I appreciate it.

There is no hurry. Take your time.

Best regards

Nicolai

mikakaraila, 2021-01-29:
OK, v0.22.100 published

closed 2021-01-29 by mikakaraila