mikakaraila/node-red-contrib-opcua #257: Issue with the monitoring action

halcyon0606, 2021-01-27:
Hello Mika,

I was running through some tests again yesterday and I noticed that the monitor node does not work like before. When I am actively monitoring a node in the addressSpace, whenever I change the value greater than the deadband I expect the client to return the changed value, but in this case I do not get any returned value. This was working in the previous versions so I hope that it can be fixed soon, thank you!

mikakaraila, 2021-01-27:
OK, have to check. No known modifications for this part of code...

mikakaraila, 2021-01-27:
Quick test shows it works:
[image]

halcyon0606, 2021-01-27:
That is strange, it does not report back the values to me no matter how large the increment, I will do another test on my end then and get back to you.

halcyon0606, 2021-01-27:
Just updated to 0.2.98, now there is a problem with using the write action, even UaExpert has a problem with handling the write request.

Here is the error message shown:
[image]

mikakaraila, 2021-01-27:
I need some example, I made Array support yesterday and it can have some side effects.
I tested with my own test flow, but it does not cover everything...

halcyon0606, 2021-01-27:
I'll provide my server and client flows so you can look into it:

### Server Flow
[{"id":"8c852dbd.73c4a","type":"tab","label":"OPC UA Server","disabled":false,"info":""},{"id":"fc5f9de1.5c10b","type":"split","z":"8c852dbd.73c4a","name":"","splt":"\\n","spltType":"str","arraySplt":1,"arraySpltType":"len","stream":false,"addname":"","x":470,"y":280,"wires":[["1c017d80.08dc23"]]},{"id":"ad524f5.1484cb","type":"inject","z":"8c852dbd.73c4a","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":true,"onceDelay":"1","topic":"","payload":"","payloadType":"date","x":110,"y":260,"wires":[["a94666bf.f00368"]]},{"id":"a94666bf.f00368","type":"function","z":"8c852dbd.73c4a","name":"Structuring server","func":"//Listing out all commands to be added to message\ncommand1 = [{\"opcuaCommand\":\"addFolder\"},\"ns=1;s=Conveyor\"];\ncommand2 = [{\"opcuaCommand\":\"addVariable\"},\"ns=1;s=IO_Speed;datatype=Float\"];\ncommand3 = [{\"opcuaCommand\":\"setFolder\"},\"ns=1;i=85\"];\ncommand4 = [{\"opcuaCommand\":\"addFolder\"},\"ns=1;s=Sensor1\"];\ncommand5 = [{\"opcuaCommand\":\"addVariable\"},\"ns=1;i=12900;datatype=Boolean\"];\ncommand6 = [{\"opcuaCommand\":\"setFolder\"},\"ns=1;i=85\"];\ncommand7 = [{\"opcuaCommand\":\"addFolder\"},\"ns=1;s=Sensor2\"];\ncommand8 = [{\"opcuaCommand\":\"addVariable\"},\"ns=1;s=IO;datatype=Boolean\"];\nmsg.payload = [command1,command2,command3,command4,command5,command6,command7,command8];\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","x":290,"y":260,"wires":[["fc5f9de1.5c10b"]]},{"id":"1c017d80.08dc23","type":"delay","z":"8c852dbd.73c4a","name":"","pauseType":"rate","timeout":"100","timeoutUnits":"milliseconds","rate":"1","nbRateUnits":"0.1","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":false,"x":610,"y":300,"wires":[["4e2cfdf7.894c24"]]},{"id":"4e2cfdf7.894c24","type":"function","z":"8c852dbd.73c4a","name":"Splitting message into topic and payload","func":"msg.topic = msg.payload[1];\nmsg.payload = msg.payload[0];\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","x":900,"y":300,"wires":[["fc45075c.d51898","ecaf97a.37eab68","ed4af43f.9d9bb8"]]},{"id":"fc45075c.d51898","type":"debug","z":"8c852dbd.73c4a","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"payload","targetType":"msg","statusVal":"","statusType":"auto","x":1210,"y":200,"wires":[]},{"id":"ecaf97a.37eab68","type":"debug","z":"8c852dbd.73c4a","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"topic","targetType":"msg","statusVal":"","statusType":"auto","x":1200,"y":280,"wires":[]},{"id":"ed4af43f.9d9bb8","type":"OpcUa-Server","z":"8c852dbd.73c4a","port":"54880","name":"","endpoint":"","autoAcceptUnknownCertificate":false,"registerToDiscovery":false,"maxNodesPerBrowse":"","maxNodesPerHistoryReadData":"","maxNodesPerHistoryReadEvents":"","maxNodesPerHistoryUpdateData":"","maxNodesPerRead":"","maxNodesPerWrite":"","maxNodesPerMethodCall":"","maxNodesPerRegisterNodes":"","maxNodesPerNodeManagement":"","maxMonitoredItemsPerCall":"","maxNodesPerHistoryUpdateEvents":"","maxNodesPerTranslateBrowsePathsToNodeIds":"","x":1220,"y":380,"wires":[["89c8a4d5.07e828"]]},{"id":"89c8a4d5.07e828","type":"debug","z":"8c852dbd.73c4a","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":1390,"y":380,"wires":[]},{"id":"fd4ebf33.9c6cb","type":"inject","z":"8c852dbd.73c4a","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"{\"opcuaCommand\":\"deleteNode\",\"nodeId\":\"ns=1;i=12900\"}","payloadType":"json","x":910,"y":360,"wires":[["ed4af43f.9d9bb8"]]},{"id":"5ce5a61f.fbde18","type":"status","z":"8c852dbd.73c4a","name":"","scope":["ed4af43f.9d9bb8"],"x":640,"y":580,"wires":[["34022c45.63c5a4"]]},{"id":"34022c45.63c5a4","type":"debug","z":"8c852dbd.73c4a","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"status.text","targetType":"msg","statusVal":"","statusType":"auto","x":880,"y":580,"wires":[]},{"id":"e4a3f896.8fc7e8","type":"inject","z":"8c852dbd.73c4a","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"{\"opcuaCommand\":\"restartOPCUAServer\"}","payloadType":"json","x":910,"y":420,"wires":[["ed4af43f.9d9bb8"]]}]

### Clients Flow
[{"id":"2d630062.17aca","type":"tab","label":"Client Actions Demo","disabled":false,"info":""},{"id":"a6c8675d.2df678","type":"OpcUa-Client","z":"2d630062.17aca","endpoint":"c06892f0.58fd5","action":"info","deadbandtype":"a","deadbandvalue":1,"time":10,"timeUnit":"s","certificate":"n","localfile":"","localkeyfile":"","securitymode":"None","securitypolicy":"None","name":"","x":720,"y":340,"wires":[["17e6787d.f71c08"]]},{"id":"ccb6bc91.b0237","type":"inject","z":"2d630062.17aca","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"","payloadType":"date","x":330,"y":340,"wires":[["3bda4d17.686962"]]},{"id":"3bda4d17.686962","type":"OpcUa-Item","z":"2d630062.17aca","item":"ns=1;s=IO_Speed","datatype":"Float","value":"","name":"","x":520,"y":340,"wires":[["a6c8675d.2df678"]]},{"id":"17e6787d.f71c08","type":"debug","z":"2d630062.17aca","name":"","active":true," [... cut ...] me":"","x":870,"y":1340,"wires":[["43f97d0a.969394"]]},{"id":"588d357b.c2af2c","type":"function","z":"2d630062.17aca","name":"","func":"var myCount = flow.get(\"count\")||0;\nmsg.payload = myCount + 5;\n\nflow.set(\"count\",msg.payload)\n\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","x":480,"y":140,"wires":[["e0419b1f.59bc48"]]},{"id":"c06892f0.58fd5","type":"OpcUa-Endpoint","endpoint":"opc.tcp://MSI:54880","secpol":"Basic256Sha256","secmode":"SignAndEncrypt","login":false}]

mikakaraila, 2021-01-27:
JSON is corrupted, can you attach flows as txt files?

halcyon0606, 2021-01-27:
Sorry my bad, I do not know how to attach it as a text file, but I've double checked this time to ensure the following JSON code works:

### Server Flows:

`[
    {
        "id": "8c852dbd.73c4a",
        "type": "tab",
        "label": "OPC UA Server",
        "disabled": false,
        "info": ""
    },
    {
        "id": "fc5f9de1.5c10b",
        "type": "split",
        "z": "8c852dbd.73c4a",
        "name": "",
        "splt": "\\n",
        "spltType": "str",
        "arraySplt": 1,
        "arraySpltType": "len",
        "stream": false,
        "addname": "",
        "x": 550,
        "y": 280,
        "wires": [
            [
                "1c017d80.08dc23"
            ]
        ]
    },
    {
        "id": "ad524f5.1484cb",
        "type": "inject",
        "z": "8c852dbd.73c4a",
        "name": "",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": true,
        "onceDelay": "1",
        "topic": "",
        "payload": "",
        "payloadType": "date",
        "x": 190,
        "y": 260,
        "wires": [
            [
                "a94666bf.f00368"
            ]
        ]
    },
    {
        "id": "a94666bf.f00368",
        "type": "function",
        "z": "8c852dbd.73c4a",
        "name": "Structuring server",
        "func": "//Listing out all commands to be added to message\ncommand1 = [{\"opcuaCommand\":\"addFolder\"},\"ns=1;s=Conveyor\"];\ncommand2 = [{\"opcuaCommand\":\"addVariable\"},\"ns=1;s=IO_Speed;datatype=Float\"];\ncommand3 = [{\"opcuaCommand\":\"setFolder\"},\"ns=1;i=85\"];\ncommand4 = [{\"opcuaCommand\":\"addFolder\"},\"ns=1;s=Sensor1\"];\ncommand5 = [{\"opcuaCommand\":\"addVariable\"},\"ns=1;i=12900;datatype=Boolean\"];\ncommand6 = [{\"opcuaCommand\":\"setFolder\"},\"ns=1;i=85\"];\ncommand7 = [{\"opcuaCommand\":\"addFolder\"},\"ns=1;s=Sensor2\"];\ncommand8 = [{\"opcuaCommand\":\"addVariable\"},\"ns=1;s=IO;datatype=Boolean\"];\nmsg.payload = [command1,command2,command3,command4,command5,command6,command7,command8];\nreturn msg;",
        "outputs": 1,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
[... 1165 lines cut ...]
        "type": "function",
        "z": "2d630062.17aca",
        "name": "",
        "func": "var myCount = flow.get(\"count\")||0;\nmsg.payload = myCount + 5;\n\nflow.set(\"count\",msg.payload)\n\nreturn msg;",
        "outputs": 1,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "x": 480,
        "y": 140,
        "wires": [
            [
                "e0419b1f.59bc48"
            ]
        ]
    },
    {
        "id": "c06892f0.58fd5",
        "type": "OpcUa-Endpoint",
        "endpoint": "opc.tcp://MSI:54880",
        "secpol": "Basic256Sha256",
        "secmode": "SignAndEncrypt",
        "login": false
    }
]`

mikakaraila, 2021-01-27:
OK, found it. Server code calls one function with a different parameter..

mikakaraila, 2021-01-27:
OK test with 0.2.99

halcyon0606, 2021-01-28:
Thank you Mika, the write function now works without any issues. However, I still have issues using the monitoring action of the client, it does not return the monitored item's value when the change is greater than the deadband set.

mikakaraila, 2021-01-28:
Attach here screenshot how you are using parameters.

I made clean layout for your flow:
[image]

mikakaraila, 2021-01-28:
[CLEANUP.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/5885208/CLEANUP.txt)

Rename cleanup.txt to cleanup.json

mikakaraila, 2021-01-28:
I am using these parameters and they are working:
[image]

halcyon0606, 2021-01-28:
Thank you for the cleaned up version, I have tested using that but I am still unable to get the output from the client's debug node after the initial output, it will go into a keepAlive state and no values will get reported back even though the value is constantly changing. But it could be a computer issue, will check back once more the moment I finish testing it on a separate hardware.

mikakaraila, 2021-01-28:
If you have large interval and percentage deadband it could be just out-of-range so it will not detect change.
What is actual value that you are monitoring? What are real values?

mikakaraila, 2021-01-28:
I think I found it. If you inject timestamp in monitoring mode it will use it as "interval".

Inject empty string and it works:

[image]
[image]

closed 2021-01-28 by mikakaraila