mikakaraila/node-red-contrib-opcua #258: Question about how to implement CA Certificates

halcyon0606, 2021-01-29:
Hello Mika, 

This is not an issue but something that I am trying out. I am just wondering how would a certificate issued by a certificate authority be implemented here? I would like to test out more advanced security features so would it be possible for you to provide me an example of where the certificate of the CA would go and also the issued cert and key? Thank you!

mikakaraila, 2021-01-29:
Here are some links:

https://reference.opcfoundation.org/v104/Core/docs/Part2/8/

https://stackoverflow.com/questions/21297139/how-do-you-sign-a-certificate-signing-request-with-your-certification-authority

1. Using the x509 module: openssl x509
2 Using the ca module: openssl ca

halcyon0606, 2021-01-29:
Ah yes I'm familiar with creating a CA Certificate with OpenSSL, I am wondering how to use it in the case of this package? Like which certificate folder do I put it into, like what goes into the issuer folder, etc.

halcyon0606, 2021-02-08:
Hello Mika,

In the folder shown in the image below, where would I insert the certificate issued by the Certificate Authority? Do I also need to include the CA's own certificate as well?
[image]