mikakaraila/node-red-contrib-opcua #259: Unsubscribe all items at once.

ramonneves, 2021-02-05:
Hello Mika!

Is there a way to unsubscribe all subscribed items in one shot?
I've tried "action" = "deletesubscription" but the items continue to be read.

Thanks in advance.

mikakaraila, 2021-02-05:
Hmm, I expect then node-opcua does remove listener. Have to inspect this more with Etienne

mikakaraila, 2021-02-06:
There was error in verbose, v0.2.101 will include this to it.

Deletesubscription will terminate and as I tested it terminates subscription and none of the subscribed item will not update value anymore.

[image]

mikakaraila, 2021-02-06:
GIT repo contains v0.2.101 but npm publish reports network error. I will publish version later.

mikakaraila, 2021-02-07:
npm contains now latest version

closed 2021-02-07 by mikakaraila

ramonneves, 2021-02-07:
Hello Mika, 

Just tried with the following straight to the client and still reading the tags:
[image]

mikakaraila, 2021-02-07:
Is it connected to same client node that is making subscription?

ramonneves, 2021-02-07:
Yes, it is. 
I read a bunch of tags from a csv file and then subscribe them. 
If I click the "Unsubscribe" and do it one by one, it is ok. But not with the Unsubscribe all:

[image]

mikakaraila, 2021-02-07:
Hmm, it can be the server doesn't work correctly.

Can you see server console or log?

ramonneves, 2021-02-08:
Yes, I can. What should I look for?

ramonneves, 2021-02-08:
The server is working fine. 
One weird behavior that is happening is something with the logging. I configured the settings to log "debug" level, but none of the  OPC UA messages are being sent neither to the console or node-red debug. I see some other messages, but not the ones generated by actions:

[image]

mikakaraila, 2021-02-08:
Did you start with command:
node-red -v flow.json

ramonneves, 2021-02-08:
ok, got the messages:
[image]

ramonneves, 2021-02-08:
Ok, according to this piece of code, I should have a topic set. 
[image]
What should I put in the topic, since I want to delete everything?

ramonneves, 2021-02-08:
It works if I write anything in the "topic". Maybe a bug.
[image]

mikakaraila, 2021-02-08:
My inject node is this way:
[image]

[image]

ramonneves, 2021-02-08:
Ok, got it. Thanks Mika!