mikakaraila/node-red-contrib-opcua #260: Browse: Returns the correct amount of Objekts but all have the same information

HansDamph, 2021-02-08:
I want to browse through OPC-UA Server from Injection Moulding Machine. It did work a few days before but now i receive the same message over and over again. It is the same problem with different Node-ID (Subfolders).

I takes 110 seconds to browse this folder:
[image]
[image]
[image]

Browsing with UA-Expert does work.

mikakaraila, 2021-02-08:
Ack, something wrong. Of course if you browse whole address space it will take time.

HansDamph, 2021-02-08:
Thank you very much!!! Version 0.2.102 is working! 

One more questions. When i browse, i get "browseName". Just like here:
[image]

With UA-Expert i get additional information "Description". Would it be possible to read that too?
[image]

closed 2021-02-08 by HansDamph

mikakaraila, 2021-02-08:
node crawler does not return it by default.

If you look node-red console you can see what it reads:
[image]

HansDamph, 2021-02-09:
Thank you for explaining! Here is what i got:

│  │  ├─ browseName: T803I

│  │  ├─ nodeId: ns=2;i=207570

│  │  ├─ typeDefinition: BaseArbParameterType

│  │  └─ hasComponent

In other words: There is no way to access that data with nodeopcua at the time.

mikakaraila, 2021-02-09:
For each or selected nodeId you can use action info.

HansDamph, 2021-02-09:
I get an error message when i want INFO from NodeClass "Object" (T801)

[image]
[image]

It works with underlying NodeClass "Variable" (T801-Value) without error, but it does not have any more info. 

[image]

mikakaraila, 2021-02-09:
Yes I noticed same. Please make new issue. Info seems to support only basic types. I have to extend it's functionality.

mikakaraila, 2021-02-09:
I added description into this one first, easier to use it here, I will update INFO action next.

[image]

mikakaraila, 2021-02-09:
Will be in v0.2.103