mikakaraila/node-red-contrib-opcua #261: Invalid endpoint error, causing due to failed openssl command.

nikhilkinkar, 2021-02-08:
we have few test devices on which node-red application is running, on one of those devices we are constantly getting the error

> invalid endpoint opc.tcp://OPCUA_SERVER_IP:4840 Error: Command failed: "openssl" req -new -sha256  -text  -extensions v3_selfsigned  -config "/home/edge/.config/node-opcua-default-nodejs/PKI/own/openssl.cnf.tmp" -key "/home/edge/.config/node-opcua-default-nodejs/PKI/own/private/private_key.pem" -out "/home/edge/.node-red/node_modules/node-opcua-client/certificates/client_selfsigned_cert_2048.pem.csr" -subj "/C=FR/L=Orleans/O=Sterfive/CN=NodeOPCUA-Client@epc1522/DC=epc1522"
> Can't open /home/edge/.node-red/node_modules/node-opcua-client/certificates/client_selfsigned_cert_2048.pem.csr for writing, No such file or directory
> 139981074134272:error:02001002:system library:fopen:No such file or directory:../openssl-1.1.1g/crypto/bio/bss_file.c:69:fopen('/home/edge/.node-red/node_modules/node-opcua-client/certificates/client_selfsigned_cert_2048.pem.csr','w')
> 139981074134272:error:2006D080:BIO routines:BIO_new_file:no such file:../openssl-1.1.1g/crypto/bio/bss_file.c:76:

this is visible under the opcua client node,
[image]

I have checked the path `/home/edge/.node-red/node_modules/node-opcua-client/` but the folder structure seems to be different in this case, inside the given path I can see 

> $ ls
> LICENSE  README.md  dist  node_modules  package.json  source  test_helpers  tsconfig.json  typedoc.js

the **_certificates directory_** (as mentioned in error) is not present here. Same build is working on another device smoothly. Just to check the folder structure I checked it with older build with node-red-contrib-opcua version 0.2.89 there the `/home/edge/.node-red/node_modules/node-opcua-client/` contains the _certificates folder_, that build is also working. I am not sure what is causing this error. Is it device specific issue or some config is going wrong, could anybody please suggest how to get rid of this issue.

mikakaraila, 2021-02-08:
Do you have internet connection? At least you have when you install node-opcua and other npm packages.

Node-opcua tries to download and install openssl first. It is missing or it is wrong version. It causes next step fails as
it tries to generate certificate files.

openssl" req -new -sha256 -text -extensions v3_selfsigned -config "/home/edge/.config/node-opcua-default-nodejs/PKI/own/openssl.cnf.tmp" -key "/home/edge/.config/node-opcua-default-nodejs/PKI/own/private/private_key.pem" -out "/home/edge/.node-red/node_modules/node-opcua-client/certificates/client_selfsigned_cert_2048.pem.csr" -subj "/C=FR/L=Orleans/O=Sterfive/CN=NodeOPCUA-Client@epc1522/DC=epc1522"

mikakaraila, 2021-02-09:
Another case can be access rights, which user is running node-red vs. installation done by admin/root?

nikhilkinkar, 2021-02-10:
@mikakaraila Thanks for you inputs. I found that after restarting the device, the issue was gone, is there any caching mechanism that may have kept the path `"/home/edge/.node-red/node_modules/node-opcua-client/certificates/client_selfsigned_cert_2048.pem.csr"`, because we had initially node-red-contrib-opcua v0.2.89 running on this system, later we changed to v0.2.100, then it started giving error. Now the device was restarted it is working for latest version as well.

closed 2021-02-10 by mikakaraila