mikakaraila/node-red-contrib-opcua #262: Error message on INFO from NodeClass "Object". It seems like only basic types are supported yet

HansDamph, 2021-02-09:
I get an error message when i want INFO from NodeClass "Object" (T801)

[image]
[image]

It works with underlying NodeClass "Variable" (T801-Value) without error, but it does not have any more info. 

[image]

_Originally posted by @HansDamph in https://github.com/mikakaraila/node-red-contrib-opcua/issues/260#issuecomment-775769023_

mikakaraila, 2021-02-09:
Ack as I tested this myself it needs some extended functionality.

mikakaraila, 2021-02-09:
Needs more testing, but I re-implemented this in simpler way, return msg more similar way as browse does:

[image]

mikakaraila, 2021-02-09:
Just published v0.2.103 with this feature

closed 2021-02-09 by mikakaraila