mikakaraila/node-red-contrib-opcua #263: same gauge

koburg, 2021-02-09:
hi im using ops ua with dashboard. 

and I don't know why but no matter what I do. im not able to show multiple different values to gauge.

when I make the first one and add another value then the dashboard keep showing the same value in all gauges

i have made separate flows for each also tried one with switch not working either.

mikakaraila, 2021-02-09:
Can you provide your flow to understand actual problem better?

koburg, 2021-02-09:
[image]

Yes I got it working now.

see the picture.

1. I did subscribe to the items(float) instead of read and then all gauge was showing same value from first subscribed item.
2. so I guess you can only read and not subscribe to use gauge's ?

mikakaraila, 2021-02-09:
Subscription is updating value as it will change (automatically, server will publish new value). 
Read is based on inject (invoke) as you use it now with 1000ms interval.

It should work with subscription. I will take a look later.

mikakaraila, 2021-02-09:
For me gauge works with subscription:

[image]

koburg, 2021-02-09:
https://user-images.githubusercontent.com/3473374/107406327-9c3bb080-6b08-11eb-807b-04d859cbf0e3.mov

I hope you can play this small video.

this is the gauge subscribed now it mix 2 different values this is not happen when using read.

mikakaraila, 2021-02-09:
Do you have example how it works with read? Video?

mikakaraila, 2021-02-25:
If no news I expect subscription will update both values at the "same time" as thus overwrite gauge.
In case of read values are updated separate time and gauge works.
=> property of gauge UI component.

closed 2021-02-25 by mikakaraila