mikakaraila/node-red-contrib-opcua #266: Browse: Return result as one array instead of multiple messages

HansDamph, 2021-02-12:
### Feature request

When I am browsing, I check browse status to get an idea of what is happening.

Usually I get:
1. active browsing 
2. browse done
-> Every result is sent by one message which takes a huge amount of time to process. In my setup (Azure VM Ubuntu with 2x CPU) it takes around 15 ms per message. If i browse 1300 Items, it takes additional 20 seconds until all information is available after "browse done"
3. There is no indication when all messages are sent. My current solution is to "debounce" message to detect after a timeout that there will not come another message - which works but I do not like it ;-)

Could you make an option, to get data as array instead of single message? 

Thank you very much!

mikakaraila, 2021-02-12:
OK, I will take a look. Good idea / feature.

mikakaraila, 2021-02-12:
Implemented during lunch break, I will test & publish.

mikakaraila, 2021-02-12:
During testing I refactored code a bit and now AllInOne will get whole address space in 3-4 seconds.

[image]

mikakaraila, 2021-02-12:
Just add msg.collect = true into the input message then it will collect all items into one message.
This is now breaking change, but I expect everybody will like as this is faster and contains more information.

[image]

mikakaraila, 2021-02-12:
OK, I just published v0.2.106

closed 2021-02-12 by mikakaraila

HansDamph, 2021-02-15:
Works very fast now! Thank you again!

mikakaraila, 2021-02-15:
I started to think should I make item to contain topic with nodeId and datatype, then it can be used directly to read/subscribe item.
Not the array but individual messages.