mikakaraila/node-red-contrib-opcua #267: Maximum Array Size

NicolaiWrt, 2021-02-15:
Hello Mika,

my OPC UA Server is running pretty well.
Today I tried to send an Int16- and an Int32-Array with 233 Values in it to the OPC UA Server.
But it doesn't work. I tried it before with an Array which had 71 values and it worked.
Nothing changed, only the size of the array.

Now my question: Is there a maximum array size?

I looked at it with UAExpert and it showed me those errors:
[image]
[image]

Best regards and thanks in advance.

Nicolai

mikakaraila, 2021-02-15:
I have been using 60000 value array:
[image]

What is actual server? Embedded/nano? BadInternalError points to server...

mikakaraila, 2021-02-15:
Also 3 dimensional are working:
[image]

NicolaiWrt, 2021-02-17:
Hello Mika,

hm ok.

As I said it worked with an Array[70]
But not with an array with a size of 230.
It is an one dimensional array.

mikakaraila, 2021-02-17:
What is your server? Can you access Arrray[230] with UaExpert?
Remember that client cannot increase Array size, it must be pre-allocated by server.
Otherwise if server is not protected from buffer overflow you can crash it.

NicolaiWrt, 2021-02-17:
My Server is running with your OPC UA-Node in Node-RED.
I can also not access it with UA Expert. With the Array[70] it was possible.
I'm not writing on the server with a client. I write directly on the Server in Node-RED.

mikakaraila, 2021-02-18:
OK, I will check this.

mikakaraila, 2021-02-18:
OK, I added Array size into the input msg and now it will be used as variable is added to server.
I will test a bit more, but now it should work:

[image]

mikakaraila, 2021-02-18:
Supported Arrays at the moment:
[image]

Syntax to add Array: datatype=TypeArray[size]
[image]

mikakaraila, 2021-02-18:
v0.2.108 published

closed 2021-02-18 by mikakaraila

NicolaiWrt, 2021-02-19:
> OK, I will check this.
> 
> How did you create Array[70]?? In my code there is no support for server Arrays or it is not complete...
> It could explain it works partially.

I just created an Int32 Variable in the Server.
And then send an Int32 Array to this variable.
Maybe it was just luck, that it worked. ^^

Thanks for your help and the new Version.