mikakaraila/node-red-contrib-opcua #268: OPCUA Events

guicamael, 2021-02-24:
Hi,
Since version OPCUA 0.2.104 and newer, events are no longer uploaded from the Prosys 5.0.4-284 simulator via the OPCUA Events node, version OPCUA 0.2.103 works without any problems.  Do you think it would be possible to correct this anomaly in the next version? Thank you

mikakaraila, 2021-02-24:
I expect merge request broke this feature. Events are in the payload.
I will study this later what was changed...

guicamael, 2021-02-24:
The event message is not received  !
[image]

mikakaraila, 2021-02-24:
It seems to be so that client gets event, but does not send it forward.

24 Feb 17:25:52 - [info] [OpcUa-Client:A & C] Client status: active event

guicamael, 2021-02-24:
yes I confirme we see on prosys simulator the  event sent to OPCUA node.

mikakaraila, 2021-02-24:
I got it working, but I will check with the latest simulator one more time:

[image]

mikakaraila, 2021-02-24:
And also with the latest Prosys UA Simulation server:

[image]

[image]

guicamael, 2021-02-24:
Could you send your test Event flow ?

mikakaraila, 2021-02-24:
Yes, I will copy it to own separated from the test flow.

This could be needed, root node with namespace index:
[image]

mikakaraila, 2021-02-24:
Here is the flow file, just rename it from .txt back to .json
[event_flow.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/6037790/event_flow.txt)

mikakaraila, 2021-02-24:
I modified EventId to hex string, easier to read as UaExpert shows it also:
[image]

mikakaraila, 2021-02-24:
I changed (added) Quality to StatusText that is easier to read and use for example at UI:
[image]

guicamael, 2021-02-25:
Thank you for this details, but with your flow I have the same problem. no event see on node-red opcua with version 0.2.104 or newers. we have several environement with the same issue and to be sur I just reinstall from scratch node-red docker !

mikakaraila, 2021-02-25:
Wait until I release new version, even those changes done should not have broken this functionality.
I will have Siemens RFID device with OPC UA Server (coming today) and I will test events with it one more time.

mikakaraila, 2021-02-28:
I have test this with Prosys UA Simulation Server and Siemens RF360R and should work now: v0.2.109

Open / comment more if not working.

closed 2021-02-28 by mikakaraila

guicamael, 2021-03-01:
Thanks for this update, all work fine 👍