mikakaraila/node-red-contrib-opcua #269: OPC UA Server -  Read some indexes of array

jpsantos-motofil, 2021-02-25:
Hello!
First of all, thank you for your work.

One question: I can't we read only some indexes of an array? For example, imagine that my server has a variable which is a Int16 Array[5] and I only want to read indexes 2, 3 and 4. Is this possible?

Thank you!

mikakaraila, 2021-02-25:
I expect indexRange would do the work, but I have not integrated it or used it. I will take a look when I have some extra time...

mikakaraila, 2021-02-25:
I checked code and it requires extra parameters to read/write. I will study how I could add this into the current implementation.

mikakaraila, 2021-02-25:
Was easy to extend:
[image]

mikakaraila, 2021-02-25:
I will add it to write also then it will be symmetric.

jpsantos-motofil, 2021-02-25:
Awesome! Thank you for being so quick!

mikakaraila, 2021-02-25:
Partial write (range) works now with Prosys Server:

[image]

mikakaraila, 2021-02-25:
Example of read (range):
[image]

Example of write (range):
[image]

mikakaraila, 2021-02-28:
Update: I changed range from msg.payload.range to msg.range
I will add this to description and readme and publish new version.

closed 2021-02-28 by mikakaraila

jpsantos-motofil, 2021-03-04:
Hi! Once again, thank you for your work. Just as side note, on v0.2.109 the working way is to use msg.payload.range and NOT msg.range...

mikakaraila, 2021-03-04:
OK, I will fix description and code to match.

binflo13, 2024-01-12:
I have encountered a problem when trying to write to specific ranges of an array. 
I am able to read specific positions of an array using msg.payload.range, but I am having difficulties doing the same with writing.

When I pass the following flow to a write node:

```
const newMsg = {};
newMsg.topic = "ns=1;i=302";
newMsg.range = "104:105";
newMsg.payload = [true, false];
return newMsg;
```
I receive the following error:

```

12 Jan 12:06:16 - [red] Uncaught Exception:
12 Jan 12:06:16 - [error] TypeError: Cannot read properties of undefined (reading 'indexOf')
    at module.exports.build_new_dataValue (C:\Users\User\.node-red\node_modules\node-red-contrib-opcua\opcua\opcua-basics.js:943:22)
    at write_action_input (C:\Users\User\.node-red\node_modules\node-red-contrib-opcua\opcua\102-opcuaclient.js:1735:23)
    at OpcUaClientNode.processInputMsg [as _inputCallback] (C:\Users\User\.node-red\node_modules\node-red-contrib-opcua\opcua\102-opcuaclient.js:825:11)
    at C:\Users\User\AppData\Roaming\npm\node_modules\node-red\node_modules\@node-red\runtime\lib\nodes\Node.js:210:26
    at Object.trigger (C:\Users\User\AppData\Roaming\npm\node_modules\node-red\node_modules\@node-red\util\lib\hooks.js:166:13)
    at Node._emitInput (C:\Users\User\AppData\Roaming\npm\node_modules\node-red\node_modules\@node-red\runtime\lib\nodes\Node.js:202:11)
    at Node.emit (C:\Users\User\AppData\Roaming\npm\node_modules\node-red\node_modules\@node-red\runtime\lib\nodes\Node.js:186:25)
    at Node.receive (C:\Users\User\AppData\Roaming\npm\node_modules\node-red\node_modules\@node-red\runtime\lib\nodes\Node.js:494:10)
    at deliverMessageToDestination (C:\Users\User\AppData\Roaming\npm\node_modules\node-red\node_modules\@node-red\runtime\lib\flows\Flow.js:799:40)
    at Immediate.<anonymous> (C:\Users\User\AppData\Roaming\npm\node_modules\node-red\node_modules\@node-red\runtime\lib\flows\Flow.js:815:21)
```

Could anyone help me understand and resolve this issue?

Thank you in advance.

mikakaraila, 2024-01-12:
Did you look my examples? I expect you should have at least 105 items on array...

binflo13, 2024-01-15:
Which examples? The ones on this post or the ones on the examples folder?

mikakaraila, 2024-01-16:
Examples folder contains OPCUA-TEST-NODES.json look this part:

[image]

binflo13, 2024-01-25:
Thank you, @mikakaraila! I think I had the datatype missing. I tried the following and it worked:

```
msg.range = "104:105"
msg.payload = "true,false"
msg.topic = "ns=1;i=302"
msg.datatype="BooleanArray";
```