mikakaraila/node-red-contrib-opcua #270: AutoID support

mikakaraila, 2021-02-25:
AutoID for RFID Readers.

OPC UA nodeset added:

[image]

mikakaraila, 2021-02-26:
Device online and working:
[image]

mikakaraila, 2021-02-26:
Subscribe works:
[image]

mikakaraila, 2021-02-26:
Events also working:
[image]

mikakaraila, 2021-02-26:
Initial layout for example flow:
[image]

UI:
[image]

danmarsic, 2021-03-18:
Here again my issue mentioned in Configuration Error certificate and private key doesn't match #246.

I am using RFID module with OPC UA Server supporting AutoID Companion specification to read and write RFID HF Tags.
https://www.turck.de/en/product/00000040000381bb0003003a

The method ReadTag has six input arguments, where the first and the last one are not basic data types but structs specific to AutoID Companion specification. According to @mikakaraila the support for extension objects must be added in order to call this method in NodeRED.

@mikakaraila I would like at least read the events from RFID module. Where can I find this Root node parameter for events from my RFID device?  
[image]

mikakaraila, 2021-03-18:
Normally I use UaExpert or just standard server node as root for events (ns=0;i=2253)

[image]

danmarsic, 2021-03-24:
Thank you for the answer, but unfortunately it does not work.
I am getting the events from OPC UA of the RFID reader in UaExpert and there you can see that my server also uses ns=0;i=2253 as root for events (I also tried with id of ident0 which is the RFID head where events are generated, but did not work).
[image]

In NodeRED is basically modified your Events.json example to my server but still it does not work. Furthermore, you can notice that the normal OPC UA variabes with READ action from the same server are working fine. Do you have any suggestions what should I try to make it work?
[image]

And a final question. Since methods with AutoID structs are currently not supported, how did you managed to execute the method ScanStart from NodeRED for the Siemens RFID? ScanStart must be executed in order for readar to starts to looking for the events. We have triggered this method now manually from UaExpert.

danmarsic, 2021-04-01:
@mikakaraila could you give us some hint on how to solve the issues with OPC UA events from Turck RFID reader as described in the previous post?

Could you at least send us your flow.json which you successfully tested with Siemens RFID?

mikakaraila, 2021-04-01:
Sorry I have been on vacation. Siemens RFID detects tag itself and updates value to variable.

Here is the flow I have used. Still working with ExtensionObject.

Rename file back to .json

[RF360R.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/6242432/RF360R.txt)

danmarsic, 2021-04-01:
@mikakaraila thank you for the example. :) 
The only difference I notice in your flow for reading RFID events in that you are using Progress Event Type instead of Base Event (all). We will try with this setting and let you know if it works.

Our reader also has LastScanData variable but with data type LastScanData. Did you made any additional setting on the Siemens device in order to use it as a **string** in OpcUA-Item Node in NodeRED?

mikakaraila, 2021-04-01:
Device worked immediately as I put power on and connected Ethernet. I don´t know if they adjusted something for me before sending it to me. Do you have any devices available for couple of weeks? I could test it.

danmarsic, 2021-04-01:
We have our own Turck RFID OPC UA AutoId device. However, we are a private company so I should check it out internally, if we can send it to a private person. 
Do you have any e-mail where we can reach you? So we are not spamming this public Issue with any private details.

mikakaraila, 2021-04-06:
OK let´s discuss by email mika.karaila@valmet.com
We have some contacts to Turck and we use some of their devices connected to our DCS.

mikakaraila, 2021-04-06:
OK now it is possible to write ExtensionObject. Tested with ScanSettings:

[image]

Leo1963, 2021-11-16:
Hello,
in the AutoId specification is the ScanData method. A structure is passed as a parameter for this method. For this you need a Structur variable type ScanSettings in the Node Method like this :

const scanSettingsParams = {
                        duration : 0,
                        cycles : 0,
                        dataAvailable : true,
                        lacationType: 0 }

Do you see a way to do this with your Node Method?
ExtensionObjects would have to be included :-)
Mfg.
Leo

mikakaraila, 2021-11-16:
OK, it will need some changes, the current implementation is more for the basic data types.

Leo1963, 2021-11-17:
How can I understand that now, would you do something about it? 
That would be great

mikakaraila, 2021-11-17:
Yes I will extend method call parameters, just busy at work. I will try to do it tomorrow / weekend.

mikakaraila, 2021-11-18:
Done, but I forgot RFID Reader to office:

[image]

mikakaraila, 2021-11-18:
Code will need some extra parameters, but I will need to test this with real server:

18 Nov 21:28:22 - [info] [OpcUa-Method:Scan settings] Call request: { /*CallMethodRequest*/
 objectId                      /* NodeId              */: ns=5;s=MyDevice
 methodId                      /* NodeId              */: ns=5;s=MyMethod
 inputArguments                /* Variant          [] */: [ /* length =1*/

   Variant(Scalar<ExtensionObject>, value: Variant(Scalar<ExtensionObject>, value: { /*ScanSettings*/
    duration                      /* Double              */: 1000
    cycles                        /* Int32               */: 5
    dataAvailable                 /* UABoolean           */: true
    locationType                  /* LocationTypeEnume   */: LocationTypeEnumeration.NMEA ( 0)
   };))
 ]
};
18 Nov 21:28:22 - [info] [OpcUa-Method:Scan settings] Calling: { /*CallMethodRequest*/
 objectId                      /* NodeId              */: ns=5;s=MyDevice
 methodId                      /* NodeId              */: ns=5;s=MyMethod
 inputArguments                /* Variant          [] */: [ /* length =1*/

   Variant(Scalar<ExtensionObject>, value: Variant(Scalar<ExtensionObject>, value: { /*ScanSettings*/
    duration                      /* Double              */: 1000
    cycles                        /* Int32               */: 5
    dataAvailable                 /* UABoolean           */: true
    locationType                  /* LocationTypeEnume   */: LocationTypeEnumeration.NMEA ( 0)
   };))
 ]
};

mikakaraila, 2021-11-21:
Testing, but something wrong with the Scan result, node-opcua complain scandata enumeration:

```Javascript
21 Nov 14:45:04 - [info] [OpcUa-Client:BUILD EXTENSION OBJ.] Client status: session active
21 Nov 14:45:12 - [info] [OpcUa-Method:Scan  with settings] Call method: {"objectId":"ns=4;i=5002","methodId":"ns=4;i=7010","inputArguments":[{"dataType":"ExtensionObject","value":{"duration":1000,"cycles":5,"dataAvailable":true,"locationType":0}}],"outputArguments":null}
21 Nov 14:45:12 - [info] [OpcUa-Method:Scan  with settings] Calling method: "ns=4;i=7010"
21 Nov 14:45:12 - [info] [OpcUa-Method:Scan  with settings] InputArguments: [{"dataType":"ExtensionObject","value":{"duration":1000,"cycles":5,"dataAvailable":true,"locationType":0}}]
Error: Cannot find enumeration with type ScanData
    at Object.getEnumeration (C:\Users\karaimi\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-factory\dist\factories_enumerations.js:72:15)
    at figureOutSchema (C:\Users\karaimi\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-factory\dist\factories_structuredTypeSchema.js:67:48)
    at buildField (C:\Users\karaimi\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-factory\dist\factories_structuredTypeSchema.js:82:20)
    at Array.map (<anonymous>)
    at new StructuredTypeSchema (C:\Users\karaimi\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-factory\dist\factories_structuredTypeSchema.js:113:38)
    at Object.buildStructuredType (C:\Users\karaimi\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-factory\dist\factories_structuredTypeSchema.js:266:12)
    at _getOrCreateStructuredTypeSchema (C:\Users\karaimi\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-schemas\dist\source\tools.js:136:45)
    at Object.getOrCreateStructuredTypeSchema (C:\Users\karaimi\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-schemas\dist\source\tools.js:158:12)
    at C:\Users\karaimi\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-schemas\dist\source\parse_binary_xsd.js:310:21
    at SaxLtx.<anonymous> (C:\Users\karaimi\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-xml2json\dist\source\xml2json.js:283:17)
node-id    : ns=3;i=6044
nodeClass  : Variable
browseName : { /* DataValue */
   value: Variant(Scalar<QualifiedName>, value: 3:ScanSettings)
   statusCode:      Good (0x00000)
   serverTimestamp: 2021-10-21T16:02:13.168Z $ 339.400.000
   sourceTimestamp: null
}
{ /*BrowseResult*/
 statusCode                    /* StatusCode          */: Good (0x00000)
 continuationPoint             /* ByteString          */: null
 references                    /* ReferenceDescript[] */: [ /* empty */ ]
};
(node:1256) UnhandledPromiseRejectionWarning: Error: getDataTypeDefinition invalid HasEncoding reference dataTypeNodeId must be NodeClass.DataType but was Variable
    at C:\Users\karaimi\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-client-dynamic-extension-object\dist\client_dynamic_extension_object.js:650:19
    at Generator.next (<anonymous>)
    at fulfilled (C:\Users\karaimi\.node-red\node_modules\node-red-contrib-opcua\node_modules\node-opcua-client-dynamic-extension-object\dist\client_dynamic_extension_object.js:5:58)
    at processTicksAndRejections (internal/process/task_queues.js:97:5)
(node:1256) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 2)
(node:1256) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

mikakaraila, 2021-11-21:
UaExpert, Scan parameters:

[image]

Result:
[image]