mikakaraila/node-red-contrib-opcua #271: Error on Server trying to read from Client

ramonneves, 2021-02-25:
Hello Mika,

I'm trying to add a variable to the server (images below) : 
Flow:
[image]

Add Folder:
[image]

Add Variable:
[image]

but as soon as I try to read it from any opc client, I get the following error:

(node:19872) UnhandledPromiseRejectionWarning: Error: expecting a number NaN
    at coerceVariantType (C:\Users\ramon\.node-red\node_modules\node-opcua-variant\dist\variant.js:612:23)
    at convertTo (C:\Users\ramon\.node-red\node_modules\node-opcua-variant\dist\variant.js:430:21)
    at coerceVariantArray (C:\Users\ramon\.node-red\node_modules\node-opcua-variant\dist\variant.js:450:23)
    at constructHook (C:\Users\ramon\.node-red\node_modules\node-opcua-variant\dist\variant.js:336:28)
    at new Variant (C:\Users\ramon\.node-red\node_modules\node-opcua-variant\dist\variant.js:59:26)
    at new DataValue (C:\Users\ramon\.node-red\node_modules\node-opcua-data-value\dist\datavalue.js:237:42)
    at UAVariable._readArrayDimensions (C:\Users\ramon\.node-red\node_modules\node-opcua-address-space\dist\src\ua_variable.js:1299:16)
    at UAVariable.readAttribute (C:\Users\ramon\.node-red\node_modules\node-opcua-address-space\dist\src\ua_variable.js:409:29)
    at ServerEngine._readSingleNode (C:\Users\ramon\.node-red\node_modules\node-opcua-server\dist\server_engine.js:1481:29)
    at ServerEngine.read (C:\Users\ramon\.node-red\node_modules\node-opcua-server\dist\server_engine.js:927:36)
(node:19872) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 10)

mikakaraila, 2021-02-25:
Yes, I can confirm. Broken with some other change.

mikakaraila, 2021-02-26:
Caused by Array support. I initialized Scalar variable information as undefined.
Fixed, will come soon to next release.

ramonneves, 2021-02-26:
Great!

Could you send me the lines change so I can fix in my local files and keep working on my project?

Thanks!

mikakaraila, 2021-02-26:
104-opcuaserver.js
Lines 526 & 527:
                        var dim1 = 0;           // Fix for the scalars
                        var valueRank = null;   // Fix for the scalars

mikakaraila, 2021-02-28:
Published just v0.2.109

closed 2021-02-28 by mikakaraila

ramonneves, 2021-02-28:
Hello Mika, 

Same thing happening on adding a Boolean var.
Function Node code:
[image]

Error after adding the variable to any OPC UA Client:
(node:8580) UnhandledPromiseRejectionWarning: Error: expecting a number NaN
    at coerceVariantType (C:\Users\ramon\.node-red\node_modules\node-opcua-variant\source\variant.ts:770:23)
    at convertTo (C:\Users\ramon\.node-red\node_modules\node-opcua-variant\source\variant.ts:549:21)
    at coerceVariantArray (C:\Users\ramon\.node-red\node_modules\node-opcua-variant\source\variant.ts:584:23)
    at constructHook (C:\Users\ramon\.node-red\node_modules\node-opcua-variant\source\variant.ts:424:28)
    at new Variant (C:\Users\ramon\.node-red\node_modules\node-opcua-variant\source\variant.ts:111:26)
    at new DataValue (C:\Users\ramon\.node-red\node_modules\node-opcua-data-value\source\datavalue.ts:292:42)
    at UAVariable._readArrayDimensions (C:\Users\ramon\.node-red\node_modules\node-opcua-address-space\src\ua_variable.ts:1654:16)
    at UAVariable.readAttribute (C:\Users\ramon\.node-red\node_modules\node-opcua-address-space\src\ua_variable.ts:562:29)
    at ServerEngine._readSingleNode (C:\Users\ramon\.node-red\node_modules\node-opcua-server\source\server_engine.ts:1949:29)
    at ServerEngine.read (C:\Users\ramon\.node-red\node_modules\node-opcua-server\source\server_engine.ts:1221:36)

mikakaraila, 2021-03-01:
Did you update latest v0.2.109? I tested and everything worked with the latest version.

ramonneves, 2021-03-01:
Yes I did:
[image]

ramonneves, 2021-03-01:
Please try the following flow:

[{"id":"69bbcd87.959054","type":"tab","label":"OPC Server","disabled":false,"info":""},{"id":"19f355c3.a4fa6a","type":"OpcUa-Server","z":"69bbcd87.959054","port":"53880","name":"","endpoint":"UA/BumperOPC","autoAcceptUnknownCertificate":true,"registerToDiscovery":false,"constructDefaultAddressSpace":true,"maxNodesPerBrowse":0,"maxNodesPerHistoryReadData":0,"maxNodesPerHistoryReadEvents":0,"maxNodesPerHistoryUpdateData":0,"maxNodesPerRead":0,"maxNodesPerWrite":0,"maxNodesPerMethodCall":0,"maxNodesPerRegisterNodes":0,"maxNodesPerNodeManagement":0,"maxMonitoredItemsPerCall":0,"maxNodesPerHistoryUpdateEvents":0,"maxNodesPerTranslateBrowsePathsToNodeIds":0,"x":1020,"y":280,"wires":[["80d20145.66f7d"]]},{"id":"80d20145.66f7d","type":"debug","z":"69bbcd87.959054","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":1210,"y":280,"wires":[]},{"id":"84101968.b666e8","type":"status","z":"69bbcd87.959054","name":"","scope":["19f355c3.a4fa6a"],"x":400,"y":280,"wires":[["5c461708.4b30f8"]]},{"id":"5c461708.4b30f8","type":"function","z":"69bbcd87.959054","name":"If Server Running","func":"if(msg.status.text == \"running\"){\n    //Create App Folder\n    msg = {\n        \"payload\" : { \"opcuaCommand\": \"addFolder\"}, \n        \"topic\" : \"ns=1;s=DataBumper\"\n    };\n    node.send(msg)\n    //Create EquipmentName Var\n    msg = {\n        \"payload\" : { \"opcuaCommand\": \"addVariable\"}, \n        \"topic\" : \"ns=1;s=EquipmentName;datatype=StringArray\"\n    };\n    //Create EquipmentType Var\n    node.send(msg)\n     msg = {\n        \"payload\" : { \"opcuaCommand\": \"addVariable\"}, \n        \"topic\" : \"ns=1;s=EquipmenType;datatype=Int32Array\"\n    };\n    //Create Status Var\n    node.send(msg)\n     msg = {\n        \"payload\" : { \"opcuaCommand\": \"addVariable\"}, \n        \"topic\" : \"ns=1;s=Status;datatype=Int32Array\"\n    };\n     //Create ActivityTime Var\n    node.send(msg)\n     msg = {\n        \"payload\" : { \"opcuaCommand\": \"addVariable\"}, \n        \"topic\" : \"ns=1;s=ActivityTime;datatype=StringArray\"\n    };\n    node.send(msg)\n    \n    //Create DataReady Var\n    //msg = {\n    //    \"payload\" : { \"opcuaCommand\": \"addVariable\"}, \n    //    \"topic\" : \"ns=1;s=DataReady;datatype=Boolean\"\n    //};\n    \n    node.send(msg)\n}\n","outputs":1,"noerr":0,"initialize":"","finalize":"","x":590,"y":280,"wires":[["519418d9.f73678"]]},{"id":"519418d9.f73678","type":"delay","z":"69bbcd87.959054","name":"","pauseType":"delay","timeout":"100","timeoutUnits":"milliseconds","rate":"1","nbRateUnits":"1","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":false,"x":810,"y":280,"wires":[["19f355c3.a4fa6a"]]}]

mikakaraila, 2021-03-01:
C:\Users\karaimi\.node-red>notepad test_flow.json

C:\Users\karaimi\.node-red>node-red -v test_flow.json
1 Mar 14:43:36 - [info]

Welcome to Node-RED
===================

1 Mar 14:43:36 - [info] Node-RED version: v1.0.6
1 Mar 14:43:36 - [info] Node.js  version: v12.16.2
1 Mar 14:43:36 - [info] Windows_NT 10.0.17763 x64 LE
1 Mar 14:43:36 - [info] Loading palette nodes
1 Mar 14:43:37 - [info] Worldmap version 2.8.1
1 Mar 14:43:38 - [info] Dashboard version 2.27.0 started at /ui
1 Mar 14:43:38 - [info] Settings file  : C:\Users\karaimi\.node-red\settings.js
1 Mar 14:43:38 - [info] Context store  : 'default' [module=memory]
1 Mar 14:43:38 - [info] User directory : C:\Users\karaimi\.node-red
1 Mar 14:43:38 - [warn] Projects disabled : editorTheme.projects.enabled=false
1 Mar 14:43:38 - [info] Flows file     : C:\Users\karaimi\.node-red\test_flow.json
1 Mar 14:43:38 - [warn] Existing flow file is not valid json
1 Mar 14:43:38 - [warn]

ramonneves, 2021-03-01:
Strange.
I exported from a running flow... Let's try again:
```
[{"id":"69bbcd87.959054","type":"tab","label":"OPC Server","disabled":false,"info":""},{"id":"19f355c3.a4fa6a","type":"OpcUa-Server","z":"69bbcd87.959054","port":"53880","name":"","endpoint":"UA/BumperOPC","autoAcceptUnknownCertificate":true,"registerToDiscovery":false,"constructDefaultAddressSpace":true,"maxNodesPerBrowse":0,"maxNodesPerHistoryReadData":0,"maxNodesPerHistoryReadEvents":0,"maxNodesPerHistoryUpdateData":0,"maxNodesPerRead":0,"maxNodesPerWrite":0,"maxNodesPerMethodCall":0,"maxNodesPerRegisterNodes":0,"maxNodesPerNodeManagement":0,"maxMonitoredItemsPerCall":0,"maxNodesPerHistoryUpdateEvents":0,"maxNodesPerTranslateBrowsePathsToNodeIds":0,"x":980,"y":280,"wires":[["80d20145.66f7d"]]},{"id":"80d20145.66f7d","type":"debug","z":"69bbcd87.959054","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":1210,"y":280,"wires":[]},{"id":"84101968.b666e8","type":"status","z":"69bbcd87.959054","name":"","scope":["19f355c3.a4fa6a"],"x":400,"y":280,"wires":[["5c461708.4b30f8"]]},{"id":"5c461708.4b30f8","type":"function","z":"69bbcd87.959054","name":"If Server Running","func":"if(msg.status.text == \"running\"){\n    //Create App Folder\n    msg = {\n        \"payload\" : { \"opcuaCommand\": \"addFolder\"}, \n        \"topic\" : \"ns=1;s=DataBumper\"\n    };\n    node.send(msg)\n    //Create EquipmentName Var\n    msg = {\n        \"payload\" : { \"opcuaCommand\": \"addVariable\"}, \n        \"topic\" : \"ns=1;s=EquipmentName;datatype=Int32Array\"\n    };\n    node.send(msg)\n    \n    //Create EquipmentType Var\n    \n     msg = {\n        \"payload\" : { \"opcuaCommand\": \"addVariable\"}, \n        \"topic\" : \"ns=1;s=EquipmenType;datatype=Int32Array\"\n    };\n    \n    node.send(msg)\n    \n    //Create Status Var\n    \n     msg = {\n        \"payload\" : { \"opcuaCommand\": \"addVariable\"}, \n        \"topic\" : \"ns=1;s=Status;datatype=Int32Array\"\n    };\n    node.send(msg)\n    \n     //Create ActivityTime Var\n     msg = {\n        \"payload\" : { \"opcuaCommand\": \"addVariable\"}, \n        \"topic\" : \"ns=1;s=ActivityTime;datatype=Int32Array\"\n    };\n    node.send(msg)\n    \n    //Create DataReady Var\n    //msg = {\n    //    \"payload\" : { \"opcuaCommand\": \"addVariable\"}, \n    //    \"topic\" : \"ns=1;s=DataReady;datatype=Boolean\"\n    //};\n    \n    //node.send(msg)\n}\n","outputs":1,"noerr":0,"initialize":"","finalize":"","x":590,"y":280,"wires":[["519418d9.f73678"]]},{"id":"519418d9.f73678","type":"delay","z":"69bbcd87.959054","name":"","pauseType":"delay","timeout":"100","timeoutUnits":"milliseconds","rate":"1","nbRateUnits":"1","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":false,"x":790,"y":280,"wires":[["19f355c3.a4fa6a"]]}]
```
Also, I've tested changing the code a little bit, and now the client works:
[image]

ramonneves, 2021-03-01:
The problem is actually happening with arrays, not the Boolean var.

mikakaraila, 2021-03-01:
Then you miss Array size from your payload:

ns=1;s=Int32Array;datatype=Int32Array[100]

ramonneves, 2021-03-01:
Ok, got it. 
Thanks for your time Mika.

mikakaraila, 2021-03-01:
I have to check node description and readme for this use case.