mikakaraila/node-red-contrib-opcua #272: Uncaught Exception when clients browse or advise items

jhottell, 2021-02-26:
Node-RED crashes with an Uncaught Exception when browsing or advising items via UA Expert or opcua-commander.

**Environment**
Windows 10, NodeJS 15.8.0, Node-RED 1.2.9, node-red-contrib-opcua  0.2.108

**Error**
`26 Feb 11:30:33 - [red] Uncaught Exception:
26 Feb 11:30:33 - Error: expecting a number NaN
    at coerceVariantType (C:\Users\FILEA\.node-red\node_modules\node-opcua-variant\source\variant.ts:770:23)
    at convertTo (C:\Users\FILEA\.node-red\node_modules\node-opcua-variant\source\variant.ts:549:21)
    at coerceVariantArray (C:\Users\FILEA\.node-red\node_modules\node-opcua-variant\source\variant.ts:584:23)
    at constructHook (C:\Users\FILEA\.node-red\node_modules\node-opcua-variant\source\variant.ts:424:28)
    at new Variant (C:\Users\FILEA\.node-red\node_modules\node-opcua-variant\source\variant.ts:111:26)
    at new DataValue (C:\Users\FILEA\.node-red\node_modules\node-opcua-data-value\source\datavalue.ts:292:42)
    at UAVariable._readArrayDimensions (C:\Users\FILEA\.node-red\node_modules\node-opcua-address-space\src\ua_variable.ts:1654:16)
    at UAVariable.readAttribute (C:\Users\FILEA\.node-red\node_modules\node-opcua-address-space\src\ua_variable.ts:562:29)
    at ServerEngine._readSingleNode (C:\Users\FILEA\.node-red\node_modules\node-opcua-server\source\server_engine.ts:1949:29)
    at ServerEngine.read (C:\Users\FILEA\.node-red\node_modules\node-opcua-server\source\server_engine.ts:1221:36)`

Here is a simple test flow to recreate my test.
`[{"id":"1c972871.1481f8","type":"tab","label":"Flow 3","disabled":false,"info":""},{"id":"7fd82ae9.3bdbc4","type":"debug","z":"1c972871.1481f8","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","x":910,"y":180,"wires":[]},{"id":"d8afdb60.f186c8","type":"OpcUa-Server","z":"1c972871.1481f8","port":"48015","name":"","endpoint":"","autoAcceptUnknownCertificate":true,"registerToDiscovery":false,"constructDefaultAddressSpace":false,"maxNodesPerBrowse":"","maxNodesPerHistoryReadData":"","maxNodesPerHistoryReadEvents":"","maxNodesPerHistoryUpdateData":"","maxNodesPerRead":"","maxNodesPerWrite":"","maxNodesPerMethodCall":"","maxNodesPerRegisterNodes":"","maxNodesPerNodeManagement":"","maxMonitoredItemsPerCall":"","maxNodesPerHistoryUpdateEvents":"","maxNodesPerTranslateBrowsePathsToNodeIds":"","x":720,"y":180,"wires":[["7fd82ae9.3bdbc4"]]},{"id":"f4789aa6.29f8a8","type":"function","z":"1c972871.1481f8","name":"Build Well Data","func":"var msg01 = { payload: { \"opcuaCommand\": \"setFolder\" }, topic: \"ns=1;s=Data\" };\nvar msg02 = { payload: { \"opcuaCommand\": \"addVariable\" }, topic: \"ns=1;s=tubing_pressure;datatype=Float\" }; \nvar msg03 = { payload: { \"opcuaCommand\": \"addVariable\" }, topic: \"ns=1;s=casing_pressure;datatype=Float\" }; \nvar msg04 = { payload: { \"opcuaCommand\": \"addVariable\" }, topic: \"ns=1;s=gas_flowrate;datatype=Float\" }; \n\n\nreturn [[msg01, msg02, msg03, msg04]]; \n","outputs":1,"noerr":0,"initialize":"","finalize":"","x":340,"y":160,"wires":[["d8afdb60.f186c8","18ea7c5c.e3c554"]]},{"id":"b8fa0328.a1af7","type":"function","z":"1c972871.1481f8","name":"Update Well Data","func":"var msg01 = { payload: { \"messageType\" : \"Variable\", \"variableName\": \"tubing_pressure\", \"variableValue\": 12.22 }, topic: \"tubing_pressure\"};\nvar msg02 = { payload: { \"messageType\" : \"Variable\", \"variableName\": \"casing_pressure\", \"variableValue\": 13.33 }, topic: \"casing_pressure\"};\nvar msg03 = { payload: { \"messageType\" : \"Variable\", \"variableName\": \"gas_flowrate\", \"variableValue\": 14.44 }, topic: \"gas_flowrate\"};\n\nreturn [[msg01, msg02, msg03]]; \n","outputs":1,"noerr":0,"initialize":"","finalize":"","x":350,"y":260,"wires":[["d8afdb60.f186c8","15593f91.35de1"]]},{"id":"a7f7ebd7.42b078","type":"inject","z":"1c972871.1481f8","name":"publish","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":true,"onceDelay":"10","topic":"","payload":"","payloadType":"date","x":140,"y":260,"wires":[["b8fa0328.a1af7"]]},{"id":"e9c1d869.8ae5c8","type":"function","z":"1c972871.1481f8","name":"Build Folders","func":"id = '001';\n//id = global.get(\"selected_simulator\");\n\nvar msg01 = { payload: { \"opcuaCommand\": \"addFolder\" }, topic: \"ns=1;s=Well\" };\nvar msg02 = { payload: { \"opcuaCommand\": \"setFolder\" }, topic: \"ns=1;s=Well\" };\nvar msg03 = { payload: { \"opcuaCommand\": \"addFolder\" }, topic: \"ns=1;s=Data\" };\n\nreturn [[msg01, msg02, msg03]]; \n","outputs":1,"noerr":0,"initialize":"","finalize":"","x":330,"y":120,"wires":[["d8afdb60.f186c8","18ea7c5c.e3c554"]]},{"id":"1d4058ff.86fdf7","type":"delay","z":"1c972871.1481f8","name":"","pauseType":"delay","timeout":"1","timeoutUnits":"seconds","rate":"1","nbRateUnits":"1","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":false,"x":140,"y":160,"wires":[["f4789aa6.29f8a8"]]},{"id":"7e9c0fe4.746ab","type":"comment","z":"1c972871.1481f8","name":"Startup","info":"","x":150,"y":80,"wires":[]},{"id":"5ab6bb62.ceb604","type":"comment","z":"1c972871.1481f8","name":"Publish","info":"","x":150,"y":220,"wires":[]},{"id":"158be464.2cf9dc","type":"inject","z":"1c972871.1481f8","name":"startup","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":true,"onceDelay":"8","topic":"","payload":"001","payloadType":"str","x":140,"y":120,"wires":[["e9c1d869.8ae5c8","1d4058ff.86fdf7"]]},{"id":"15593f91.35de1","type":"debug","z":"1c972871.1481f8","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","x":590,"y":260,"wires":[]},{"id":"18ea7c5c.e3c554","type":"debug","z":"1c972871.1481f8","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","x":590,"y":100,"wires":[]}]`

I decided to try on Ubuntu to see if it was limited to windows but I get the error a bit differently. UA expert hangs up indefinitely until I kill Node-RED.

**Environment**
Ubuntu 18.04, NodeJS 14.16.0, Node-RED 1.2.9, node-red-contrib-opcua  0.2.108

**Error**
`cli.html#cli_unhandled_rejections_mode). (rejection id: 5)
(node:14455) UnhandledPromiseRejectionWarning: Error: expecting a number NaN
    at coerceVariantType (/home/autoserv/.node-red/node_modules/node-opcua-variant/dist/variant.js:612:23)
    at convertTo (/home/autoserv/.node-red/node_modules/node-opcua-variant/dist/variant.js:430:21)
    at coerceVariantArray (/home/autoserv/.node-red/node_modules/node-opcua-variant/dist/variant.js:450:23)
    at constructHook (/home/autoserv/.node-red/node_modules/node-opcua-variant/dist/variant.js:336:28)
    at new Variant (/home/autoserv/.node-red/node_modules/node-opcua-variant/dist/variant.js:59:26)
    at new DataValue (/home/autoserv/.node-red/node_modules/node-opcua-data-value/dist/datavalue.js:237:42)
    at UAVariable._readArrayDimensions (/home/autoserv/.node-red/node_modules/node-opcua-address-space/dist/src/ua_variable.js:1299:16)
    at UAVariable.readAttribute (/home/autoserv/.node-red/node_modules/node-opcua-address-space/dist/src/ua_variable.js:409:29)
    at ServerEngine._readSingleNode (/home/autoserv/.node-red/node_modules/node-opcua-server/dist/server_engine.js:1481:29)
    at ServerEngine.read (/home/autoserv/.node-red/node_modules/node-opcua-server/dist/server_engine.js:927:36)
(node:14455) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 6)
`

mikakaraila, 2021-02-28:
I expect this is duplicate to #271 caused as I added Array support to server.
I will publish new version today.

Under development version works with your test flow:
[image]

mikakaraila, 2021-02-28:
Fixed in v0.2.109

closed 2021-02-28 by mikakaraila