mikakaraila/node-red-contrib-opcua #273: Fix For Scalars

ramonneves, 2021-03-01:
Mika,

I have a little suggestion:
Some clients are having issues on identifying Scalars if valueRank is null. They only work if the valueRank is -1.
So, in order to fix that, i had to change the code in 2 places:
[image]

[image]

I've found this inside the conformance check:
[image]

In UA expert, the scalar variables have the valueRank = -1
[image]

If I don´t do that, my other client software identify it as an array **(without the code change)**:
[image]
[image]

mikakaraila, 2021-03-01:
OK, thanks. I will make a change. This seems to be interesting to see. I have seen both in use null and -1.

mikakaraila, 2021-03-04:
Fixed to v0.2.110

closed 2021-03-04 by mikakaraila

ramonneves, 2021-03-04:
Great. 
I've also seen that you added the multidimension array support. That is awesome.

Thanks.