mikakaraila/node-red-contrib-opcua #275: Can´t start OpcUa Server (ubuntu)

johannisbaer, 2021-03-04:
Hey!

I couldn´t start the OpcUa Server on Linux.  But it worked on Windows. The error message was:

```
[NODE-OPCUA-E02] generateAddressSpace : cannot find nodeset2 xml file at /data/node_modules/node-red-contrib-opcua/opcua/public/vendor/opc-foundation/xml/Opc.Ua.AutoID.NodeSet2.xml
Error: TypeError: Cannot read property 'objects' of null
```

**Solution:**

I found the file `Opc.Ua.AutoID.nodeset2.xml` in `/data/node_modules/node-red-contrib-opcua/opcua/public/vendor/opc-foundation/xml/` and renamed it to `Opc.Ua.AutoID.NodeSet2.xml`. This was the only file with lower-case **n**ode**s**et.

mikakaraila, 2021-03-04:
Thanks, I will fix filename to next version. Linux is more sensitive.

mikakaraila, 2021-03-04:
Fixed to v0.2.110

closed 2021-03-04 by mikakaraila