mikakaraila/node-red-contrib-opcua #276: Problems with DateTime datatype in OPC UA Client and OPC UA Server

rugee-f, 2021-03-05:
Hi mika
I have a problem when I try to write to a DateTime variable in an OPC UA Server running on a Siemens PLC 1510.
I can connect to the server and read the item but a write to the item is failing. 
I installed the new 0.2.110 version from yesterday.

The read on my item delivers this message:
{"_msgid":"fab0ffc6.8711c","payload":"2021-01-01T01:00:00.000Z","topic":"ns=3;s=\"gtyp_Interface_Dashboard\".\"Publish\".\"ldt_AcknowledgeButton\"","datatype":"DateTime","browseName":"","statusCode":{"value":0}}

The write  produces the following two messages:
OpcUaClientNode: Second set value by Input 2021-03-05T10:33:00.229Z
and
OpcUaClientNode: Setting value to [{"_msgid":"1","payload":"2","topic":"3","datatype":"4","browseName":"5"},"59e0ab93.c90774","2021-03-05T10:33:00.229Z","ns=3;s=\"gtyp_Interface_Dashboard\".\"Publish\".\"ldt_AcknowledgeButton\"","DateTime",""]

The value on the PLC is not changing. 

To simulate the read/write process I took your OPC UA Server node and try to create an item with the datatype 'DateTime' but this isn't working. 
There are 2 messages on adding an item of the DateTime datatype:  
OpcUaServerNode: adding Node ns=1;s=TestTime;datatype=DateTime  
and 
TypeError: Cannot read property 'toString' of null

Is it possible that there is a general problem with the datetime datatype items in your OPC UA nodes?

mikakaraila, 2021-03-05:
Ack, there was missing one Variant and datatype was internally UtcTime -> DateTime.
Fixed to next version.

mikakaraila, 2021-03-05:
v0.2.111 published

closed 2021-03-05 by mikakaraila

rugee-f, 2021-03-05:
You are lightning fast - thank you very much!