mikakaraila/node-red-contrib-opcua #278: Question: reconnecting to subscription tags on a rebooted server.

youngkzy, 2021-03-08:
QUESTION:  Subscribed to arrays on Kepware OPC Server 6 and monitoring tags.  During the weekend the OPC Server on a Windows Server was restarted and the connection was lost.  This morning, I entered a new subscription to the data.  What is the reccommended method to handle OPC Server disconnections and re-connections?  The active subscription makes the connection,  Is there a means to re-connect to an existing subscription?  Otherwise I get duplicates as There is a subscription to the same tags multiple times.  An Example of a means would be great if it already exists, I cannot locate it in the documentation. Thanks,

mikakaraila, 2021-03-08:
If it is just lost connection, then server "keeps" subscription and nodeIds in it´s memory.
But as server boots then everything is lost, client will try to create session, subscription etc.
I cannot remember how long it will retry & how many times, depends on client parameters.
Plus some servers can behave different way after reboot.  You should test different cases: network break, server start, server boot.

youngkzy, 2021-03-08:
@mikakaraila,
  Thanks for your quick reply.  OK I can test this, but I'm also looking to "fix" by means of some flow logic and nodes to address it if it's not a "node" problem.  It probably isn't your node.  Kepware server reboots every night. I was using the system on my work PC to monitor some tags in a system for troubleshooting.  It failed as the server rebooted...  What recommendation do you have for this kind of problem using what is already there.  The examples do not go into this kind of thing.   Can I monitor NODE STATUS?  (Node-RED Status Node) Or can I monitor a clock on the OPC Server system tags?  Or can I monitor something from OPCUA node which tells me that the connection is broke in order to force a new connection?   Now,  I need to understand... 
1. when my subscription is no longer any good.. how to tell?
2. ping or query the OPC Server to see if it is present..  ?? how to check OPC server is up? (after the file server is running). 
3. Determine if the subscription is old or still good?  How to reconnect? or force a new reconnect when the server is back up..  It does look like it was trying a reconnect but the remote server was rebooting.

This kind of thing?   No need to do the work for me.  Good explanation should suffice as long as I can get the hows related to the nodes.

Thanks

mikakaraila, 2021-03-08:
You can use Catch node to monitor node status, this way:
[image]

It can be that computer is running, but OPC UA Server (KepServer) will crash. Then ping will not report it.
If computer will crash then ping will detect it.

Or use READ Current time from the server, if current time is not updated.

youngkzy, 2021-03-08:
@mikakaraila ,
   Thanks for that advice.  OK,  So subscribing to an already set subscription is going to get me multiple values of the same tags when I add the same list to the subscription.  How would I connect if I can sense that the subscription is still valid?   And perhaps understand that it is still valid?  Can I ask the system how many items are in the subscription?  or get that from some reporting value in the node?

mikakaraila, 2021-03-09:
You can see this so client will try to re-connect to server:
[image]

youngkzy, 2021-03-11:
@mikakaraila ,
   I do not know if I asked the question in the right way. Or maybe I'm a bit dense...  If it is obvious, I'm overlooking it.
Let's say... 
1.  The remote OPC Server has been rebooted.  I understand that I will need to SUBSCRIBE again to these tags and old subscription will not need to be deleted or UNSUBSCRIBED?  I can ping and see if the server has been rebooted if lost ping maybe.  I can see if the OPC Server is lost by checking date-time data OK.  
2.  The remote OPC Server HAS NOT BEEN Rebooted.  We lost the OPC Server or there was a connection glitch and it's not connected for some reason...  I do not know how to force a reconnect to the same subscription(that should connect).  Is that possible?  How?  Can I poll the connection? to see what my current Subscription is? or is possible?  So I can A. UNSUBSCRIBE, or DELETE SUBSCRIPTION, Then Re-SUBSCRIBE...

mikakaraila, 2021-03-11:
1) I suppose client will try to connect again, create session and if server doesn´t return old subscription then library will create a new one. No need to do anything.
2) Same story client should first re-connect on network level and then do as in case 1)

Node-opcua makes life easy, don´t try to do it too complicated on the application level as library already handles it.

closed 2021-03-11 by mikakaraila

youngkzy, 2021-03-11:
@mikakaraila ,
  Again Thanks for your help.  My problem is one that does not behave for some reason and it appears that it is stubborn.  I came to work today to find my tags I am monitoring for debugging and writing to a log file are disconnected.  And the status is RED under the SUBSCRIBE nodes.  Thus the basis for these questions and my persistence.  Sorry!  The server may have been re-booted again.  I was told that this should only happen on weekends.  So I'm unsure of what transpired and working to find out.   However, my logging did not work for the time which it found this fault.  I applied my manual Inject nodes to initiate the SUBSCRIBE to get it back to working, as it was just sitting there.
Kind Regards,
@youngkzy

mikakaraila, 2021-03-12:
It depends also on server how it behaves.

But as primary root cause is server boot, it would be better to solve it.
Memory/driver problem or power supply...

youngkzy, 2021-03-12:
@mikakaraila ,
  The remote server in this case is supposed to re-boot on some weekly basis. That is for Windows updates etc... So, it's not THE problem, but it is a thing I need to deal with.  I guess we are not quite on the same page.  I am OK with it behaving this way on Kepware if that isn't an issue for anyone else.  I require no extra attention except to get a clean way to get back to monitoring.  I can use the method I described earlier IF that is the best way to approach it.  

One question before I leave you...   or assumption I want to verify.  If the system for some reason has a subscription, but it disconnects,  Is the only way to re-connect,  to dump/delete the current subscription and Subscribe with a new one?   assumption: there is no way to connect manually to a subscription.   The only way I know to know how many items is from the subscription report in the messages.

Kind Regards,

mikakaraila, 2021-03-13:
As server does not have any subscriptions after reboot, I suppose you can just create it again after ping works.
Some delay most probably needed before KepServer is really running and accepting connections.