mikakaraila/node-red-contrib-opcua #279: Question to multiple read/ multiple write

rugee-f, 2021-03-08:
Is it possible to get the result of a multiple read in a single object with the values e.g. stored in an array?
The normal multiple read is delivering single messages for every Item.
In the "documentation issue" "OpcUa-Client: WRITE MULTIPLE #198" I found a picture (unfortunately no code example) that looks almost like what I need (it is for writing - I want to read).  
I was not able to create a single multiple read Object with the "nodeId"/ "datatype" notation.
Are there any examples for the ua item declaration in functionblocks?

mikakaraila, 2021-03-08:
There is one technical issue, how to return values "usable".
Return value is array of DataValues which can be different types. I expect you would like to use this as "raw".

                  // Use nodeId in topic, arrays are same length
                  node.send({
                    topic: multipleItems[i],
                    payload: dataValue.value.value
                  });

I don´t won´t to break old functionality... any proposals?

Some new setting?
msg.return_read_as_array = true;

rugee-f, 2021-03-09:
Hi Mika, I have almost no knowledge of the "old functionality". I try to build an OPC UA to MQTT gateway in node-red.
Whenever a timestamp on my plc is changing I grab some values on OPC ua side and push them to several MQTT topics.
Or in the other direction: when a MQTT topic is changing I took some values and push them to my OPC UA Items. My Idea was to do this in a few function nodes. The #198 case pointed me in the direction that this should be possible but I could not find an example flow. Of course a new option to the existing node could solve this, and yes a raw format (maybe as string) could help me to build my mqtt message. But maybe it could be done with the existing functionality. To clarify what I want to do I added a flow - It is not working - but you can see what I like to have...
[desired_function_flow.json.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/6106727/desired_function_flow.json.txt)

mikakaraila, 2021-03-09:
Hmm, bit clumsy. Needs a lot manual coding inside function node to map each array item.
Another case will be in some point mqtt message size is limited to ~128kb.

rugee-f, 2021-03-09:
Any better suggestions? - I am open for everything, maybe my poor knowledge about your nodes are my limiting factor.
I thought waiting for my trigger event - grabbing 5 to 10 OPC values and transfering them to MQTT was very straightforward. And the single flows could be bound clearly to the desired functionality. So it is not too complicated if you regard one flow from input to output. I did not knew of the 128 kb MQTT message size limitation. I have to keep this in mind...
Thank you for your really fast replies.

mikakaraila, 2021-03-09:
Working day at the office, I will make proposal evening...

mikakaraila, 2021-03-09:
Here is proposal, it will need small change as you wanted to get all in one msg v0.2.113

This won´t break existing functionality:
[image]

[
    {
        "id": "cab0b952.67b2d8",
        "type": "tab",
        "label": "Flow 1",
        "disabled": false,
        "info": ""
    },
    {
        "id": "722696f3.ebb558",
        "type": "function",
        "z": "cab0b952.67b2d8",
        "name": "Double",
        "func": "//Add one by one\nmsg.topic = \"ns=5;s=Double;datatype=Double\";\n\nreturn msg;",
        "outputs": 1,
        "noerr": 0,
        "x": 280,
        "y": 500,
        "wires": [
            [
                "4edf2e86.20446"
            ]
        ]
    },
    {
        "id": "1bf7523b.d531de",
        "type": "inject",
        "z": "cab0b952.67b2d8",
        "name": "ADD",
        "topic": "",
        "payload": "",
        "payloadType": "str",
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "x": 90,
        "y": 500,
        "wires": [
            [
                "722696f3.ebb558"
            ]
        ]
    },
    {
        "id": "4edf2e86.20446",
        "type": "OpcUa-Client",
        "z": "cab0b952.67b2d8",
        "endpoint": "bce9b84b.b1c1b8",
        "action": "readmultiple",
        "deadbandtype": "a",
        "deadbandvalue": 1,
        "time": 10,
        "timeUnit": "s",
        "certificate": "n",
        "localfile": "",
        "localkeyfile": "",
        "securitymode": "None",
        "securitypolicy": "None",
        "name": "Prosys read multiple",
        "x": 620,
        "y": 500,
        "wires": [
            [
                "d5b687ac.8a8998"
[... 163 lines cut ...]
    {
        "id": "498e6800.e5bc48",
        "type": "debug",
        "z": "cab0b952.67b2d8",
        "name": "",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "targetType": "full",
        "x": 990,
        "y": 540,
        "wires": []
    },
    {
        "id": "bce9b84b.b1c1b8",
        "type": "OpcUa-Endpoint",
        "z": "",
        "endpoint": "opc.tcp://H7Q8Q13.mshome.net:53530/OPCUA/SimulationServer",
        "secpol": "None",
        "secmode": "None",
        "login": false
    }
]

closed 2021-03-09 by mikakaraila