mikakaraila/node-red-contrib-opcua #280: Using different namespaces

johannisbaer, 2021-03-09:
Good morning Mika!

How can I add a folder/variable in another namespace (e.g. ns=2)?
No errors for namespace 1 (ns=1).
I am using the nodered/node-red docker container and your newest version (0.2.112).
Again: I have no problems under Windows, but in the container...

[image]

johannisbaer, 2021-03-09:
OK. Thank You!

mikakaraila, 2021-03-09:
This will come to v0.2.113
[image]

mikakaraila, 2021-03-09:
Tested with addVariable:
[image]

closed 2021-03-09 by mikakaraila