mikakaraila/node-red-contrib-opcua #281: Server is not running

corte, 2021-03-09:
Hi,
I've been trying to create a OPC UA Server on my node-red installation, but I'm geting error: 
```
"Server node error on: error: "Server is not running""
```

I've just tried to use the example flow in this repository named `OPCUAServer_Flow.json`.

Can you help me understand what I'm doing wrong? Is there something else I need to install in order for this to work?

mikakaraila, 2021-03-09:
Which version you are using?
Start with command: node-red -v OPCUAServer_Flow.json
Look what is printed to console.

mikakaraila, 2021-03-09:
Works with the latest v0.2.112

[image]

raulus127, 2024-08-06:
Sorry, I have the same problem. I have version v4.0.2 of node-red. What should I do?

raulus127, 2024-08-06:
These warnings also appear:

6 Aug 14:33:56 - [warn] [OpcUa-Server:11682102.b5b9af] Mandatory parameter 'namespace' is missing
6 Aug 14:33:56 - [warn] [OpcUa-Server:11682102.b5b9af] warning: properties like messageType, namespace, variableName or VariableValue is missing.

raulus127, 2024-08-06:
Your flow credentials file is encrypted using a system-generated key.

If the system-generated key is lost for any reason, your credentials
file will not be recoverable, you will have to delete it and re-enter
your credentials.

You should set your own key using the 'credentialSecret' option in
your settings file. Node-RED will then re-encrypt your credentials
file using your chosen key the next time you deploy a change.