mikakaraila/node-red-contrib-opcua #282: Endpoint can not Connect to open62541 Server

klementuel, 2021-03-10:
Hello,
I would like to connect to my open62541 server via your Node Red Endpoint.
Unfortunately this does not work. But the log file from the server shows me the same when I try to connect as when I open a connection with UaExpert.
[image]

I had already tried it with another server via Python opcua. Unfortunately also without success

The endpoint then at some point outputs the error in the image
[image]

Does anyone know what this could be?

regards Klemens

klementuel, 2021-03-10:
Maybe it helps
I start node red with `node-red -v flowfile.json` 

And get this output:
```
10 Mar 15:50:17 - [info] [OpcUa-Client:Read] 	Local 'own' certificate is NOT used.
Connection options:[{"securityPolicy":"1","securityMode":1,"requestedSessionTimeout":300000,"clientName":"2","endpointMustExist":false,"defaultSecureTokenLifetime":200000,"connectionStrategy":"3","keepSessionAlive":true},"http://opcfoundation.org/UA/SecurityPolicy#None","Read",{"maxRetry":10,"initialDelay":5000,"maxDelay":30000}]
10 Mar 15:50:17 - [info] [OpcUa-Client:Read] Connection options:[{"securityPolicy":"1","securityMode":1,"requestedSessionTimeout":300000,"clientName":"2","endpointMustExist":false,"defaultSecureTokenLifetime":200000,"connectionStrategy":"3","keepSessionAlive":true},"http://opcfoundation.org/UA/SecurityPolicy#None","Read",{"maxRetry":10,"initialDelay":5000,"maxDelay":30000}]
EndPoint: [{"id":"1","type":"2","_closeCallbacks":"3","_inputCallback":null,"_inputCallbacks":null,"wires":"4","_wireCount":0,"credentials":"5","endpoint":"6","securityPolicy":"7","securityMode":"7","login":false,"user":null,"password":null},"9ef114b7.5f8958","OpcUa-Endpoint",[],[],{},"opc.tcp://debian-gnu-linux-10-vm:4840/","None"]
10 Mar 15:50:17 - [info] [OpcUa-Client:Read] EndPoint: [{"id":"1","type":"2","_closeCallbacks":"3","_inputCallback":null,"_inputCallbacks":null,"wires":"4","_wireCount":0,"credentials":"5","endpoint":"6","securityPolicy":"7","securityMode":"7","login":false,"user":null,"password":null},"9ef114b7.5f8958","OpcUa-Endpoint",[],[],{},"opc.tcp://debian-gnu-linux-10-vm:4840/","None"]
UserIdentity: [{"type":0}]
10 Mar 15:50:17 - [info] [OpcUa-Client:Read] UserIdentity: [{"type":0}]
Read: Create Client: [{"securityPolicy":"1","securityMode":1,"requestedSessionTimeout":300000,"clientName":"2","endpointMustExist":false,"defaultSecureTokenLifetime":200000,"connectionStrategy":"3","keepSessionAlive":true},"http://opcfoundation.org/UA/SecurityPolicy#None","Read",{"maxRetry":10,"initialDelay":5000,"maxDelay":30000}]
10 Mar 15:50:17 - [warn] [OpcUa-Client:Read] Read: Create Client: [{"securityPolicy":"1","securityMode":1,"requestedSessionTimeout":300000,"clientName":"2","endpointMustExist":false,"defaultSecureTokenLifetime":200000,"connectionStrategy":"3","keepSessionAlive":true},"http://opcfoundation.org/UA/SecurityPolicy#None","Read",{"maxRetry":10,"initialDelay":5000,"maxDelay":30000}]
Client status: create client
10 Mar 15:50:17 - [info] [OpcUa-Client:Read] Client status: create client
Connecting to opc.tcp://debian-gnu-linux-10-vm:4840/
10 Mar 15:50:17 - [info] [OpcUa-Client:Read] Connecting to opc.tcp://debian-gnu-linux-10-vm:4840/
Client status: connecting
10 Mar 15:50:17 - [info] [OpcUa-Client:Read] Client status: connecting
Exact endpointUrl: opc.tcp://debian-gnu-linux-10-vm:4840/ hostname: debian-gnu-linux-10-vm
10 Mar 15:50:17 - [info] [OpcUa-Client:Read] Exact endpointUrl: opc.tcp://debian-gnu-linux-10-vm:4840/ hostname: debian-gnu-linux-10-vm
10 Mar 15:50:17 - [info] Started flows
Read: Case A: Endpoint does not contain, 1==None 2==Sign 3==Sign&Encrypt securityMode:[1] securityPolicy:["http://opcfoundation.org/UA/SecurityPolicy#None"]
10 Mar 15:51:17 - [warn] [OpcUa-Client:Read] Read: Case A: Endpoint does not contain, 1==None 2==Sign 3==Sign&Encrypt securityMode:[1] securityPolicy:["http://opcfoundation.org/UA/SecurityPolicy#None"]
Read: Case B: UserName & password does not match to server (needed by Sign): undefined undefined
10 Mar 15:51:17 - [warn] [OpcUa-Client:Read] Read: Case B: UserName & password does not match to server (needed by Sign): undefined undefined
Client status: invalid endpoint opc.tcp://debian-gnu-linux-10-vm:4840/
10 Mar 15:51:17 - [info] [OpcUa-Client:Read] Client status: invalid endpoint opc.tcp://debian-gnu-linux-10-vm:4840/
```

mikakaraila, 2021-03-10:
I will install and test a bit more

mikakaraila, 2021-03-10:
I´m running open62541 server on AWS EC2 docker and works.
Tested version 0.2.113

[image]

klementuel, 2021-03-10:
Hello Mika,
I just test it also on the same machine with the opcua-commander.
I think that should be the same base like node-red-contrib-opcua?

Unfortunately it worked there and I have to look for the error here again :-(

mikakaraila, 2021-03-10:
What error you mean?  
I cannot see any specific error in your screenshot. Bit more detailed description would help me to understand. 
Error in connecting server & client, variant read error or method call error or something with nodeset?
node-opcua supports only opc binary transport not http

mikakaraila, 2021-03-10:
One stupid case: inject empty string, I expect if you inject timestamp it will be used as a parameter for communication...

mikakaraila, 2021-03-11:
Code snippet that makes connection is here:
```javascript
 verbose_log("Exact endpointUrl: " + opcuaEndpoint.endpoint + " hostname: " + os.hostname());
        await node.client.connect(opcuaEndpoint.endpoint);
      } catch (err) {
        verbose_warn("Case A: Endpoint does not contain, 1==None 2==Sign 3==Sign&Encrypt securityMode:" + stringify(connectionOption.securityMode) + " securityPolicy:" + stringify(connectionOption.securityPolicy));
        verbose_warn("Case B: UserName & password does not match to server (needed by Sign): " + userIdentity.userName + " " + userIdentity.password);
        set_node_errorstatus_to("invalid endpoint " + opcuaEndpoint.endpoint, err);
```

mikakaraila, 2021-03-11:
To find out more do this:
set NODEOPCUADEBUG=CLIENT{TRACE}
node-red -v flowfile.json

mikakaraila, 2021-03-11:
This will give all request/response messages:
set NODEOPCUADEBUG=CLIENT{TRACE-REQUEST-RESPONSE-PERF-STAT}

mikakaraila, 2021-03-11:
This part will be interesting:

```javascript
06:18:01:256    >>>>>>                      2 GetEndpointsRequest            35
------------------------------------- Client Sending a request   GetEndpointsRequest h= 2  channel id  35  securityToken= 432
{ /*GetEndpointsRequest*/
 requestHeader                 /* RequestHeader       */: {
   authenticationToken         /* NodeId              */: ns=0;i=0
   timestamp                   /* DateTime            */: 2021-03-11T06:18:01.256Z
   requestHandle               /* UInt32              */: 2               0x2
   returnDiagnostics           /* UInt32              */: 0               0x0
   auditEntryId                /* UAString            */: null
   timeoutHint                 /* UInt32              */: 60000               0xea60
   additionalHeader            /* ExtensionObject     */: null
 }
 endpointUrl                   /* UAString            */: opc.tcp://XXXX.eu-west-1.compute.amazonaws.com:4840/
 localeIds                     /* UAString         [] */: [ /* empty*/ ]
 profileUris                   /* UAString         [] */: [ /* empty*/ ]
};
06:18:01:320   <<<<<< _on_message_received  2           GetEndpointsResponse Good (0x00000) 35
{ /*GetEndpointsResponse*/
 responseHeader                /* ResponseHeader      */: {
   timestamp                   /* DateTime            */: 2021-03-11T06:18:01.286Z
   requestHandle               /* UInt32              */: 2               0x2
   serviceResult               /* StatusCode          */: Good (0x00000)
   serviceDiagnostics          /* DiagnosticInfo      */: { /*DiagnosticInfo*/
 namespaceUri                  /* Int32               */: null
 symbolicId                    /* Int32               */: -1
 locale                        /* Int32               */: -1
 localizedText                 /* Int32               */: -1
 additionalInfo                /* String              */: null
 innerStatusCode               /* StatusCode          */: Good (0x00000)
 innerDiagnosticInfo           /* DiagnosticInfo      */: null
};
   stringTable                 /* UAString         [] */: null []
   additionalHeader            /* ExtensionObject     */: null
 }
 endpoints                     /* EndpointDescripti[] */: [
   { /*0*/
     endpointUrl               /* UAString            */: opc.tcp://XXXX.eu-west-1.compute.amazonaws.com:4840/
     server                    /* ApplicationDescri   */: {
       applicationUri          /* UAString            */: urn:open62541.server.application
       productUri              /* UAString            */: http://open62541.org
       applicationName         /* LocalizedText       */: locale=en text=open62541-based OPC UA Application
       applicationType         /* ApplicationType     */: ApplicationType.Server ( 0)
       gatewayServerUri        /* UAString            */: null
       discoveryProfileUri     /* UAString            */: null
       discoveryUrls           /* UAString         [] */: null []
     }
     serverCertificate         /* ByteString          */
BUFFER
     securityMode              /* MessageSecurityMo   */: MessageSecurityMode.None ( 1)
     securityPolicyUri         /* UAString            */: http://opcfoundation.org/UA/SecurityPolicy#None
     userIdentityTokens        /* UserTokenPolicy  [] */: [
       { /*0*/
         policyId              /* UAString            */: open62541-anonymous-policy
         tokenType             /* UserTokenType       */: UserTokenType.Anonymous ( 0)
         issuedTokenType       /* UAString            */: null
         issuerEndpointUrl     /* UAString            */: null
         securityPolicyUri     /* UAString            */: null
       },
       { /*1*/
         policyId              /* UAString            */: open62541-username-policy
         tokenType             /* UserTokenType       */: UserTokenType.UserName ( 1)
         issuedTokenType       /* UAString            */: null
         issuerEndpointUrl     /* UAString            */: null
         securityPolicyUri     /* UAString            */: http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
       }
     ]
     transportProfileUri       /* UAString            */: http://opcfoundation.org/UA-Profile/Transport/uatcp-uasc-uabinary
[... 49 lines cut ...]
     securityMode              /* MessageSecurityMo   */: MessageSecurityMode.Sign ( 2)
     securityPolicyUri         /* UAString            */: http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
     userIdentityTokens        /* UserTokenPolicy  [] */: [
       { /*0*/
         policyId              /* UAString            */: open62541-anonymous-policy
         tokenType             /* UserTokenType       */: UserTokenType.Anonymous ( 0)
         issuedTokenType       /* UAString            */: null
         issuerEndpointUrl     /* UAString            */: null
         securityPolicyUri     /* UAString            */: null
       },
       { /*1*/
         policyId              /* UAString            */: open62541-username-policy
         tokenType             /* UserTokenType       */: UserTokenType.UserName ( 1)
         issuedTokenType       /* UAString            */: null
         issuerEndpointUrl     /* UAString            */: null
         securityPolicyUri     /* UAString            */: http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
       }
     ]
     transportProfileUri       /* UAString            */: http://opcfoundation.org/UA-Profile/Transport/uatcp-uasc-uabinary
     securityLevel             /* Byte                */: 2
   }
 ]
};
Connected to opc.tcp://XXXX.eu-west-1.compute.amazonaws.com:4840/
```

mikakaraila, 2021-03-11:
This seems a bit similar to your problem:
https://github.com/open62541/open62541/issues/2010

klementuel, 2021-03-11:
Hello Mika
Thanks for your response:
I don't use a user or password, I don't need that at the moment.

unfortunately when I try this
set NODEOPCUADEBUG=CLIENT{TRACE-REQUEST-RESPONSE-PERF-STAT}
node-red -v flowfile.json

I get less information in the terminal than with set NODEOPCUADEBUG=CLIENT{TRACE}.
So not what you had shown, mmh do I need to open some special log file

to make sure it's not the server I tried the test with a python server.
You can also find it in the log, but the same error message comes back.
I can connect to both servers with UA Expert, with a Python client and opcua-commander.

Here the log with the set of set NODEOPCUADEBUG=CLIENT{TRACE}

```
11 Mar 12:14:21 - [info] Starting flows
11 Mar 12:14:21 - [info] [OpcUa-Client:Read] 	Local 'own' certificate is NOT used.
Connection options:[{"securityPolicy":"1","securityMode":1,"requestedSessionTimeout":300000,"clientName":"2","endpointMustExist":false,"defaultSecureTokenLifetime":200000,"connectionStrategy":"3","keepSessionAlive":true},"http://opcfoundation.org/UA/SecurityPolicy#None","Read",{"maxRetry":10,"initialDelay":5000,"maxDelay":30000}]
11 Mar 12:14:21 - [info] [OpcUa-Client:Read] Connection options:[{"securityPolicy":"1","securityMode":1,"requestedSessionTimeout":300000,"clientName":"2","endpointMustExist":false,"defaultSecureTokenLifetime":200000,"connectionStrategy":"3","keepSessionAlive":true},"http://opcfoundation.org/UA/SecurityPolicy#None","Read",{"maxRetry":10,"initialDelay":5000,"maxDelay":30000}]
EndPoint: [{"id":"1","type":"2","_closeCallbacks":"3","_inputCallback":null,"_inputCallbacks":null,"wires":"4","_wireCount":0,"credentials":"5","endpoint":"6","securityPolicy":"7","securityMode":"7","login":false,"user":null,"password":null},"9ef114b7.5f8958","OpcUa-Endpoint",[],[],{},"opc.tcp://debian-gnu-linux-10-vm:4840/","None"]
11 Mar 12:14:21 - [info] [OpcUa-Client:Read] EndPoint: [{"id":"1","type":"2","_closeCallbacks":"3","_inputCallback":null,"_inputCallbacks":null,"wires":"4","_wireCount":0,"credentials":"5","endpoint":"6","securityPolicy":"7","securityMode":"7","login":false,"user":null,"password":null},"9ef114b7.5f8958","OpcUa-Endpoint",[],[],{},"opc.tcp://debian-gnu-linux-10-vm:4840/","None"]
UserIdentity: [{"type":0}]
11 Mar 12:14:21 - [info] [OpcUa-Client:Read] UserIdentity: [{"type":0}]
Read: Create Client: [{"securityPolicy":"1","securityMode":1,"requestedSessionTimeout":300000,"clientName":"2","endpointMustExist":false,"defaultSecureTokenLifetime":200000,"connectionStrategy":"3","keepSessionAlive":true},"http://opcfoundation.org/UA/SecurityPolicy#None","Read",{"maxRetry":10,"initialDelay":5000,"maxDelay":30000}]
11 Mar 12:14:21 - [warn] [OpcUa-Client:Read] Read: Create Client: [{"securityPolicy":"1","securityMode":1,"requestedSessionTimeout":300000,"clientName":"2","endpointMustExist":false,"defaultSecureTokenLifetime":200000,"connectionStrategy":"3","keepSessionAlive":true},"http://opcfoundation.org/UA/SecurityPolicy#None","Read",{"maxRetry":10,"initialDelay":5000,"maxDelay":30000}]
Client status: create client
11 Mar 12:14:21 - [info] [OpcUa-Client:Read] Client status: create client
Connecting to opc.tcp://debian-gnu-linux-10-vm:4840/
11 Mar 12:14:21 - [info] [OpcUa-Client:Read] Connecting to opc.tcp://debian-gnu-linux-10-vm:4840/
Client status: connecting
11 Mar 12:14:21 - [info] [OpcUa-Client:Read] Client status: connecting
Exact endpointUrl: opc.tcp://debian-gnu-linux-10-vm:4840/ hostname: debian-gnu-linux-10-vm
11 Mar 12:14:21 - [info] [OpcUa-Client:Read] Exact endpointUrl: opc.tcp://debian-gnu-linux-10-vm:4840/ hostname: debian-gnu-linux-10-vm
11 Mar 12:14:21 - [info] Started flows
Read: Case A: Endpoint does not contain, 1==None 2==Sign 3==Sign&Encrypt securityMode:[1] securityPolicy:["http://opcfoundation.org/UA/SecurityPolicy#None"]
11 Mar 12:15:21 - [warn] [OpcUa-Client:Read] Read: Case A: Endpoint does not contain, 1==None 2==Sign 3==Sign&Encrypt securityMode:[1] securityPolicy:["http://opcfoundation.org/UA/SecurityPolicy#None"]
Read: Case B: UserName & password does not match to server (needed by Sign): undefined undefined
11 Mar 12:15:21 - [warn] [OpcUa-Client:Read] Read: Case B: UserName & password does not match to server (needed by Sign): undefined undefined
Client status: invalid endpoint opc.tcp://debian-gnu-linux-10-vm:4840/
11 Mar 12:15:21 - [info] [OpcUa-Client:Read] Client status: invalid endpoint opc.tcp://debian-gnu-linux-10-vm:4840/
11 Mar 12:26:45 - [info] Stopping flows
Client status: closed
11 Mar 12:26:45 - [info] [OpcUa-Client:Read] Client status: closed
Client disconnected!
11 Mar 12:26:45 - [info] [OpcUa-Client:Read] Client disconnected!
11 Mar 12:26:45 - [info] Stopped flows
11 Mar 12:26:45 - [info] Starting flows
11 Mar 12:26:45 - [info] [OpcUa-Client:open Read] 	Local 'own' certificate is NOT used.
Connection options:[{"securityPolicy":"1","securityMode":1,"requestedSessionTimeout":300000,"clientName":"2","endpointMustExist":false,"defaultSecureTokenLifetime":200000,"connectionStrategy":"3","keepSessionAlive":true},"http://opcfoundation.org/UA/SecurityPolicy#None","open Read",{"maxRetry":10,"initialDelay":5000,"maxDelay":30000}]
11 Mar 12:26:45 - [info] [OpcUa-Client:open Read] Connection options:[{"securityPolicy":"1","securityMode":1,"requestedSessionTimeout":300000,"clientName":"2","endpointMustExist":false,"defaultSecureTokenLifetime":200000,"connectionStrategy":"3","keepSessionAlive":true},"http://opcfoundation.org/UA/SecurityPolicy#None","open Read",{"maxRetry" [... cut ...]
11 Mar 12:27:45 - [info] [OpcUa-Client:Bahn Read] Client status: invalid endpoint opc.tcp://192.168.178.27:4840/bahn
```
Since I still had to deal with the login, I used the Compact Server from node-red-contrib-opcua-server for the test with username and password. The error remains the same with the Node Red client, but with the extension that the username and password are not correct (really true, they are output in plain text in the log file) and I can connect to the UA Expert client despite the user name and password like the other server before

klementuel, 2021-03-11:
> This seems a bit similar to your problem:
> [open62541/open62541#2010](https://github.com/open62541/open62541/issues/2010)

I don't think this is the same, because my server says that a client has connected during the attempt, so the host or IP or port must be correct, otherwise I wouldn't get this far.

klementuel, 2021-03-11:
> One stupid case: inject empty string, I expect if you inject timestamp it will be used as a parameter for communication...

The error already occurs before the inject case, but I tried it anyway and all errors remain the same.

klementuel, 2021-03-11:
How to connect a callback function to a method created with the Model Compiler?

mikakaraila, 2021-03-11:
Are you running node-red on Linux?

Then use this command for environment variable:
export NODEOPCUADEBUG=CLIENT{TRACE-REQUEST-RESPONSE-PERF-STAT}

klementuel, 2021-03-11:
Yes, and I try it but it has still the same effect!?

mikakaraila, 2021-03-11:
Then node-opcua cannot see environment variable, you should get these messages:

[image]

mikakaraila, 2021-03-11:
This actually depends on shell you are using with tcsh:
setenv ABC "123"

klementuel, 2021-03-11:
root@debian-gnu-linux-10-vm:~# export NODEOPCUADEBUG=CLIENT{TRACE-REQUEST-RESPONSE-PERF-STAT}
root@debian-gnu-linux-10-vm:~# env
SHELL=/bin/bash
COLORTERM=truecolor
SUDO_GID=1000
SUDO_COMMAND=/bin/bash
LC_MONETARY=en_US.UTF-8
SUDO_USER=parallels
PWD=/root
LOGNAME=root
XAUTHORITY=/run/user/1000/gdm/Xauthority
HOME=/root
LANG=en_US.UTF-8
LC_PAPER=en_US.UTF-8
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:
NODEOPCUADEBUG=CLIENT{TRACE-REQUEST-RESPONSE-PERF-STAT}
TERM=xterm-256color
USER=root
DISPLAY=:0
SHLVL=1
LC_MEASUREMENT=en_US.UTF-8
LC_TIME=en_US.UTF-8
PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
SUDO_UID=1000
MAIL=/var/mail/root
LC_NUMERIC=en_US.UTF-8
OLDPWD=/root/.node-red
_=/usr/bin/env

I can set it with export

klementuel, 2021-03-11:
After I write

mikakaraila, 2021-03-11:
Not to the settings.js
Here is link to guide (your shell is bash):
https://unix.stackexchange.com/questions/56444/how-do-i-set-an-environment-variable-on-the-command-line-and-have-it-appear-in-c

Try to do it correctly.

klementuel, 2021-03-11:
Yes I was before on this site,
I try all the stuff but nothing work

Just to be sure, I should see this output in the terminal, or is there an extra log file for this?

It's probably my OS that Debian can't handle Node Red and your OPC. If it's not possible, then I'll use my Python scripts to read something.

mikakaraila, 2021-03-11:
Node-red console is the only output that environment variable will produce output.

Should be very simple to set environment variable and then start node-red.

I expect it could be related to network settings as was with Matlab (hosts file on Windows) or IPv4/IPv6 NIC interface.
But if we cannot get debug log working, bad luck / no basic Linux skills. Cannot help any more remotely.

mikakaraila, 2021-03-11:
I installed debian vmware image, put there docker and open62541 plus node-red and everything works:

[image]

mikakaraila, 2021-03-11:
You can check your IP addresses with command: sudo ifconfig
[image]

I used this IP inside node-red flow:
[image]

closed 2021-03-11 by mikakaraila

klementuel, 2021-03-11:
I don´t know,
I try it on my raspberry and the VM, it ends every time  with the same problem.
I try it on the client side with the ip address, the local host or if I run the server on the same machine also with local host.

Can you run the debug log on debian?

If this the case I think my computers hates me ;-)

klementuel, 2021-03-11:
how to you install node red?
I follow this guide... https://nodered.org/docs/getting-started/raspberrypi
Maybe there is the problem?

mikakaraila, 2021-03-11:
I am running it as snap and it´s using running node-red as service.

I don´t know how you run node-red on your instance, but there are more instructions:

Running as a service
The install script for the Pi also sets it up to run as a service. This means it can run in the background and be enabled to automatically start on boot.

The following commands are provided to work with the service:

node-red-start - this starts the Node-RED service and displays its log output. Pressing Ctrl-C or closing the window does not stop the service; it keeps running in the background
node-red-stop - this stops the Node-RED service
node-red-restart - this stops and restarts the Node-RED service
node-red-log - this displays the log output of the service
You can also start the Node-RED service on the Raspbian Desktop by selecting the Menu -> Programming -> Node-RED menu option.

klementuel, 2021-03-11:
Hey Mika,
I just run node red now as a docker container, and the client run...
So I think there musst be a problem with the node red version from the raspberry

I run it in all before as service or in the terminal thats doesn´t help...
I think a install the normal version and test it again

mikakaraila, 2021-03-12:
OK, good as everything is working in my setups it was in your environment something that caused it.

klementuel, 2021-03-12:
Hello Mika,
Yes, thanks for your effort, I had now completely deleted Node Red again with

```
sudo npm remove -g node-red node-red-admin
rm -R ~/.node-red
```

I had also removed the services in the /usr/bin folder with
`rm node-red*`

and for safety also node js with
`sudo apt remove nodejs`

now simply reinstalled everything with the script from node red
`bash <(curl -sL https://raw.githubusercontent.com/node-red/linux-installers/master/deb/update-nodejs-and-nodered)`

and lo and behold, your debug output also goes with it:
```
export NODEOPCUADEBUG=CLIENT{TRACE-REQUEST-RESPONSE-PERF-STAT}
node-red -v
```
[image]

I tried different scenarios to recreate the error, but unfortunately it always works now ;-)

On the Raspberry it works now too, so I guess I got a bugy NodeRed version, or whatever. I set up both systems in the same week.

So thanks again for your patience and all the testing and helping you did.