mikakaraila/node-red-contrib-opcua #284: Invalid Endpoint: Certificate and private key do not match!

Jamil100, 2021-03-17:
Hello Mika,

I've been using your node for a while now and quite recently I've been experiencing an error:
[image]

I've seen the solution of deleting the NodeOPCUA-Default-nodejs folder so that default certificate gets recreated. However, I am running my node red instance on a docker container and I can't find the path to this folder.

Any suggestions?

Jamil100, 2021-03-17:
I have the latest version of node-red-contrib-opcua: 0.2.113

Jamil100, 2021-03-17:
Update: I was able to delete the folder but now I am getting the error of "Too Many Sessions"
[image]

mikakaraila, 2021-03-17:
If you are testing against some nano/embedded opc ua server. It´s limited by server to 5 parallel sessions.
Remove some client nodes from your flow if you have multiple. The current implementation opens session and keeps it open.
I am considering to change this behaviour but I want to keep it compatible. Needs some pre-thinking...

matteofranceschini, 2021-03-23:
@Jamil100  which folder did you delete? I have the same problem when connecting to a Siemens S7 :(

prmdhost, 2021-03-24:
@matteofranceschini The folder should be in your home directory: `~/.config/node-opcua-default-nodejs`

prmdhost, 2021-03-24:
I'm also having the problem with 'Invalid Endpoint' in a docker container. When the container is created and the home directory is empty, the client still sometimes fails with the certificate error. I have to recreate the container until every client (three in my case) can proceed without error.

matteofranceschini, 2021-03-24:
I'm also in using this in a container, but I have this random behavior in which deleting and creating the container sometimes fixes the error but others... it looks like it's impossible to have it fixed.

Sadly deleting that folder does not fix the issue for me, the behavior remains random. I might look into the issue you have @prmdhost , it's probably the same.

matteofranceschini, 2021-03-24:
If useful, I get this error when creating the container:
```
24 Mar 09:54:40 - [info] Starting flows
24 Mar 09:54:40 - [info] [OpcUa-Client:8db0f352.90082] 	Local 'own' certificate is NOT used.
24 Mar 09:54:40 - [info] [OpcUa-Client:da0f27f7.10284] 	Local 'own' certificate is NOT used.
24 Mar 09:54:40 - [info] [OpcUa-Client:c44f7c5f.1ffed] 	Local 'own' certificate is NOT used.
24 Mar 09:54:40 - [info] [OpcUa-Client:aa8877d2.623098] 	Local 'own' certificate is NOT used.
24 Mar 09:54:40 - [info] [OpcUa-Client:8448f616.5245b8] 	Local 'own' certificate is NOT used.
24 Mar 09:54:40 - [info] [OpcUa-Client:8d4b7e35.36dd3] 	Local 'own' certificate is NOT used.
24 Mar 09:54:40 - [info] Started flows
########################################### OPENSSL ERROR ###########################################
CWD = /opt
Command failed: "openssl"  x509 -req  -days 3650 -extensions v3_selfsigned  -extfile "/root/.config/node-opcua-default-nodejs/PKI/own/openssl.cnf.tmp" -in "/root/.config/node-opcua-default-nodejs/PKI/own/certs/client_certificate.pem.csr" -signkey "/root/.config/node-opcua-default-nodejs/PKI/own/private/private_key.pem" -text  -out "/root/.config/node-opcua-default-nodejs/PKI/own/certs/client_certificate.pem" -text 
Can't open /root/.config/node-opcua-default-nodejs/PKI/own/certs/client_certificate.pem.csr for reading, No such file or directory
140180028660552:error:02001002:system library:fopen:No such file or directory:crypto/bio/bss_file.c:69:fopen('/root/.config/node-opcua-default-nodejs/PKI/own/certs/client_certificate.pem.csr','r')
140180028660552:error:2006D080:BIO routines:BIO_new_file:no such file:crypto/bio/bss_file.c:76:
########################################### OPENSSL ERROR ###########################################
```

I'm using the "homeassistant" version of the node-red docker, in which the home folder (as far as I know) is mapped to /root .

prmdhost, 2021-03-24:
Ah, now I'm thinking it may have to do with multiple clients trying to create a certificate with the same name `own` at the same time

matteofranceschini, 2021-03-24:
Update: I had 6 clients active, I have moved down to three, and now they work. It gets me the same error, but strangly, it works.

prmdhost, 2021-03-24:
I've also seen the error with all the clients running but sometimes only one or two of the three work.

prmdhost, 2021-03-24:
This is the error when for example one of them does not connect:
```
24 Mar 10:32:49 - [info] Starting flows
24 Mar 10:32:49 - [info] [OpcUa-Client:96c8f327.77367]  Local 'own' certificate is NOT used.
24 Mar 10:32:49 - [info] [OpcUa-Client:108109d6.bac476]         Local 'own' certificate is NOT used.
24 Mar 10:32:49 - [info] [OpcUa-Client:2b42fbff.355344]         Local 'own' certificate is NOT used.
24 Mar 10:32:49 - [info] Started flows
09:32:49.705Z :verify                        :110   [NODE-OPCUA-E01] Configuration error : the certificate and the private key do not match !
09:32:49.705Z :verify                        :111                     please check the configuration of the OPCUA Server
lt-nodejs/PKI/own/private/private_key.pem
09:32:49.706Z :verify                        :113    certificateManager.privateKey=  /usr/src/node-red/.config/node-opcua-default-nodejs/PKI/own/private/private_key.pem
09:32:49.706Z :verify                        :114                  certificateFile=  /usr/src/node-red/.config/node-opcua-default-nodejs/PKI/own/certs/client_certificate.pem
```

prmdhost, 2021-03-25:
After some tests, I think it boils down to a race condition while creating the default certificate.

node-opcua/node-opcua#971

prmdhost, 2021-03-25:
I integrated the `node-opcua-default-nodejs` in a custom Dockerfile extending from the default node-red image and it works except for some warnings regarding the subject of the certficate 

```
08:52:36.088Z :verify                        :58    [NODE-OPCUA-W06] The certificate subjectAltName does not match the client applicationUri
 ...                                                Please regenerate a specific certificate that matches your client applicationUri
 ...                                                certificate subjectAltName  = urn:c2fa3ef22845:NodeOPCUA-Client
 ...                                                client applicationUri  = urn:9f690b72dcd3:NodeOPCUA-Client
 ...                                                certificateFile         = /usr/src/node-red/.config/node-opcua-default-nodejs/PKI/own/certs/client_certificate.pem
 ...
```

mikakaraila, 2021-03-25:
Simple cause is that node-red inside docker cannot access host file system and create needed certificates there.
Solutions:
a) add certificates inside docker image as you have done
b) allow docker to access host file system (https://www.digitalocean.com/community/tutorials/how-to-share-data-between-the-docker-container-and-the-host)

matteofranceschini, 2021-03-25:
Certificates are not my strongest knowledge, have I understood correctly if I say that you are suggesting to share the folder `node-opcua-default-nodejs` to the host, to have them "permanently" generated the first time?

prmdhost, 2021-03-25:
You could share the folder from your local filesystem to the docker container as a volume, then the client in the container won't have to create it

matteofranceschini, 2021-05-03:
It works for me, too! Thanks!!!

mikakaraila, 2021-06-09:
Closed as it works and new certificate manager takes care of certificates.

closed 2021-06-09 by mikakaraila