mikakaraila/node-red-contrib-opcua #285: Multiple read with different polling times

scelle, 2021-03-23:
Hi, 

Suppose I have 100 variables to read.  These are process data and will continously change, so, I want to poll these and store values unconditionally.
But poll rates for these variables can all be different. varying between 1 and 60 seconds, and its possible that for some reason the poll rates will need to be modified dynamically.
How could I setup  my opc flow without having to create an inject node for each individual variable?  Is this possible with readmultiple?

mikakaraila, 2021-03-23:
I would use readMultiple and then use different injects (timers) to read variables.

scelle, 2021-03-23:
ok, so, at start of flow,  I add all my nodeId's (var_1 to var_100)  once by sending 100 messages to my opclient with topic `  ns = 5;s = Var_**x** ;datatype = Double `.
Then I could group all nodeId's by polling times and combine them all in one inject (timer).
But what do I put in that inject message for polling a specific nodeId and not all?
As far as I know I can only use **readmultiple** as a topic?

scelle, 2021-03-25:
I compared performance between normal read  en multiple read, and as expected the readmultiple is a lot more efficient if I look at the cpu usage for the node.exe process.  but I am still struggling on how to use this readmultiple if I have nodeId's which have to be polled at different intervals. 
The solution I created for this:
I created an "inject" (function node) with a base injection time of 1s  that sends a series of messages in this order:
- msg0: topic = "clearitems"
- msg1-xx: add nodeId's to multipleItems array depending on polling time and time since last read value
- msg xx+1: topic = "readmultiple"

This works fine most of the time, but unfortunately sometimes things get out of sync and messages are no longer read.

Any other suggestions?
Is my only other option to use an opc client for each group of equal interval times or is it better to stick to only one opc client?

closed 2021-06-10 by mikakaraila