mikakaraila/node-red-contrib-opcua #286: OPCUA Client connecting loop

L1ch-K1ng, 2021-04-04:
Hello,
I have some problems with the OPCUA Client.
The client keeps trying to connect to my OPCUA server.

[image]

`TypeError: fs.rmSync is not a function
    at smartRemove (/home/pi/.node-red/node_modules/@ster5/global-mutex/dist/source/index.js:43:12)
    at removeIfTooOld (/home/pi/.node-red/node_modules/@ster5/global-mutex/dist/source/index.js:60:13)
    at /home/pi/.node-red/node_modules/@ster5/global-mutex/dist/source/index.js:117:30
    at Generator.next (<anonymous>)
    at /home/pi/.node-red/node_modules/@ster5/global-mutex/dist/source/index.js:27:71
    at new Promise (<anonymous>)
    at __awaiter (/home/pi/.node-red/node_modules/@ster5/global-mutex/dist/source/index.js:23:12)
    at lock (/home/pi/.node-red/node_modules/@ster5/global-mutex/dist/source/index.js:106:12)
    at /home/pi/.node-red/node_modules/@ster5/global-mutex/dist/source/index.js:128:57
TypeError: fs.rmSync is not a function`

This log snippet is repeating in a loop.

I tried,
... installing everything new from scretch
... reinstalling the node-red-contrib-opcua node
... reinstalling node-red
... drinking 2l coffee :/

Didn't get the OPCUA Client to work.

-----------------------------

System: Raspberry Pi 4
OS: Raspbian (latest image + updates)
Node-Red: v1.2.9
Node-Red-Contrib-OPCUA: v.0.2.113

L1ch-K1ng, 2021-04-04:
Problem solved.

Actually after the fifth reinstallation it worked :/ I don't know why.

closed 2021-04-04 by L1ch-K1ng

mikakaraila, 2021-04-06:
It has been depending on node-opcua somehow... Etienne added some mutex handler for the certificate manager.