mikakaraila/node-red-contrib-opcua #287: Cannot create client, check connection options

jonathanschneider, 2021-04-16:
Hello,
I installed the package on an Ubuntu 18.04 machine and have trouble connecting the client to a server. The client remains in the "connecting" state and the following is output in the debug pane:

> Client node error on:  error: ["Cannot create client, check connection options: [{\"securityPolicy\":\"1\",\"securityMode\":1,\"requestedSessionTimeout\":300000,\"clientName\":\"2\",\"endpointMustExist\":false,\"defaultSecureTokenLifetime\":200000,\"connectionStrategy\":\"3\",\"keepSessionAlive\":true},\"http://opcfoundation.org/UA/SecurityPolicy#None\",\"\",{\"maxRetry\":10,\"initialDelay\":5000,\"maxDelay\":30000}]"]

I can't figure out what the problem is as the node is configured exactly is I want it: no security, no credentials. The error message also shows more options than I can configure in the node.

I mentioned that I am using Ubuntu 18.04 as it is working correctly on my Raspberry Pi with the exact same configuration. I also tried three different OPC UA servers so I can rule out it's the server.

Any ideas?

mikakaraila, 2021-04-16:
What is the server? Which profile version 1.02 or 1.04? Can be some strange interoperability issue...

jonathanschneider, 2021-04-16:
I tried three different servers:
- SIMATIC S7-1500
- SITOP PSU8600
- ANSI C demo server by Unified Automation

I believe they're all supporting v1.04.

I don't think it has to do with the servers as 
- the client node can connect to these servers when run on my Raspberry Pi but not on Ubuntu
- a Wireshark trace shows no OPC UA communcation between the Ubuntu machine and the server - not even a connection attempt

mikakaraila, 2021-04-16:
I have just similar case with Siemens S7-1500 with None/None, got just all logs... 

That was good point that error happens only on Ubuntu. Perhaps something to do with openssl.

jonathanschneider, 2021-04-16:
On the Raspberry Pi OpenSSL 1.1.0l is installed, on the Ubuntu machine OpenSSL 1.1.1 is installed.

mikakaraila, 2021-04-16:
What is path to openssl?

jonathanschneider, 2021-04-16:
`/usr/bin/openssl`

Do you think the node might be missing the selfsigned certificate? Where would I find it if it was created? When would it be created? During the installation of the package or on initial usage?

mikakaraila, 2021-04-16:
Can you check node-opcua version?
There's certificate manager that will take care of certificates

jonathanschneider, 2021-04-17:
v2.33.0

mikakaraila, 2021-04-17:
Latest is v2.40.0 but I started to expect node-opcua-crypto I have to check it with Etienne

jonathanschneider, 2021-04-17:
node-red-contrib-opcua depends on node-opcua v2.33.0

jonathanschneider, 2021-04-17:
I now manually updated to v2.40.0 but that didn’t make a difference.

jonathanschneider, 2021-04-17:
I just discovered something that might be helpful. I added a server node to my flow to see if the client could connect to a server on the localhost. The server could not start and in the terminal I saw this error message:

`Error: Error: EACCESS: permission denied: mkdir '/home/<user>/.config/node-red-opcua-nodejs/PKI'`

The directory the node is trying to create does not exist on my Ubuntu machine but exists and is accessible without root priviliges on my Raspberry Pi.

Might that be clue?

jonathanschneider, 2021-04-17:
Yes, I changed the permission of ".config" and the client started working. I guess on Ubuntu ".config" is owned by root and other users have no permission to read or write that directory. Maybe there is a better place to store the certificates?

mikakaraila, 2021-04-17:
It's certificate manager default place. I have to check what could be used instead

jonathanschneider, 2021-04-17:
> Yes, I changed the permission of ".config" and the client started working. I guess on Ubuntu ".config" is owned by root and other users have no permission to read or write that directory. Maybe there is a better place to store the certificates?

After doing some research on that directory it seems like a pretty common place to put user configuration files and it should be owned by the user and not root. I have no idea why the directory on my machine was owned by root and thus not accessible for the node.
However, I'd say that's pretty good place those certificates.

If, for whatever reason, the node can't access the certificates that should be output in the console and the debug pane instead of something about the connection options. Any chance you can add that error message?

mikakaraila, 2021-04-19:
Hmm, error seems to be inside node-opcua, I have to ask Etienne if this could be improved. 
As error message is coming from the system command or os package:
Error: Error: EACCESS: permission denied: mkdir '/home/<user>/.config/node-red-opcua-nodejs/PKI'

But as you wrote place is ok.

jonathanschneider, 2021-04-19:
That error message came only when I tried the server node. The client node had the same problem accessing the certificate but didn’t output anything about permissions or certificates.

mikakaraila, 2021-04-19:
I already asked this from Etienne, we just have to wait until he wakes up :-)

erossignon, 2021-04-19:
@jonathanschneider  is right. This is an issue on his Ubuntu installation the  '/home/<user>/.config` must be R/W by the `<user>`. Something has happened on the machine to restrict the write/read access on the .config folder.

There is little we can do here.

jonathanschneider, 2021-04-19:
@erossignon could you add an error message if the client can't access the directory? Only the server threw an error.

mikakaraila, 2021-04-21:
Bug but it is actually not in the node-red-contrib-opcua but inside node-opcua.
Let´s wait if Etienne will add error message...

erossignon, 2021-04-21:
I don't think there anything we can do here. We have no intention to check the permission on every file we access in node-opcua.
Let's close.

mikakaraila, 2021-04-22:
Hmm, question was to add error message, is it impossible?

erossignon, 2021-04-22:
>Hmm, question was to add error message, is it impossible?
I've added this PR that make sure that the certificate manager is used by all instances of clients and shared with servers as well. I also fixed a problem that was causing the error message sent by node-opcua not to be displayed by node-red-contrib-opcua due to a try/catch block that was hiding it.

mikakaraila, 2021-04-22:
Etienne refactored code and now error should be visible if it still comes out after Etienne added certificate manager to client also.

Please test with new version.

closed 2021-04-26 by mikakaraila