mikakaraila/node-red-contrib-opcua #289: TypeError: cannot read property 'position' of undefined.

paulsalibe, 2021-04-16:
Hello,

I am trying to set up an OPCUA Client node with a certificate. Everything matches up and I have created and exported the certificate file from my Siemens program. This part works on the computer I set up the certificates on with the Siemens program because I can access it through UA Expert. But when I export the certificate .pem file (/home/pi/.config/node-opcua-default-nodejs/PKI/trusted/certs/BG300-0001192.168.0.84840_X.509 Certificate_3 (1).pem) and put it in that location on my raspberry pi and paste the path into the Client node, it spits out the error "TypeError: cannot read property 'position' of undefined.

As far as I know I am doing this correctly and I can access the server through UA Expert. I am using Basic256Sha256 Sign & Encrypt

mikakaraila, 2021-04-17:
I expect it´s coming from the node-opcua-crypto. Can you check certificate file with openssl?
I mean just use command like: "openssl x509 -in certfile.pem -text" that it is valid.

erossignon, 2021-04-19:
By default, Node-opcua-pki will copy the certificate file of the server to the rejected folder.
 (/home/pi/.config/node-opcua-default-nodejs/PKI/rejected). This could happen if you try to connect to the server before the certificate was setup.

You can do either way:

- move the selected certificate from `rejected` to `trusted/certs`

or

- copy the PEM certificate in advance to the  `trusted/certs` and make sure it doesn't exist already (under an othername, name be) in the `rejected` folder.

the rejected` folder` takes precedence over the    `trusted/certs`

mikakaraila, 2021-04-19:
Have you checked trusted/certs folder? 

I expect certificate manager uses thumbnail (fingerprint of certificate) as default name for certificate file.
You filename seems to be exported name: BG300-0001192.168.0.84840_X.509 Certificate_3 (1).pem

Etienne: any comments to this?

paulsalibe, 2021-04-19:
> I expect it´s coming from the node-opcua-crypto. Can you check certificate file with openssl?
> I mean just use command like: "openssl x509 -in certfile.pem -text" that it is valid.
> 
> I suppose code expects to find these fields and size must be 2048 or greater:
> 
> [image]

I am not entirely sure the nature of this command or if this is what you meant.. but I ssh'ed into the Pi and typed that command..
[image]

paulsalibe, 2021-04-19:
Also I do not have anything in the rejected folder. In the trusted folder I have certs and crl. Crl is empty and certs contains just the file I pasted in there which is just under its exported name. How can I determine the thumbnail of the certification? Sorry, I am pretty new to all of this.

mikakaraila, 2021-04-19:
Use openssl to check your certificate.
It can contain something that is not expected

paulsalibe, 2021-04-19:
> Use openssl to check your certificate.
> It can contain something that is not expected

These are what that produces ( I renamed the .pem file as it had a space in it and I dont think the command liked that) but I am not quite sure what to make of this. Let me know if you see any problems. I appreciate your help 

[image]
[image]

paulsalibe, 2021-04-20:
Any help would be greatly appreciated, think I hit a dead end and am not even sure what else to try.

mikakaraila, 2021-04-20:
@erossignon should take a look. For me it seems perfect.

paulsalibe, 2021-04-20:
Sorry did not mean to close that, accidental button hit. But would this have anything to do with how I export the certificate file from Siemens TIA portal? And/or emailing it to myself to then load onto the Raspberry Pi? Also, is it common that certificate files and private keys would live in the same .pem file?

erossignon, 2021-04-20:
This certificate doesn't look like the PLC certificate ... It seems to be the certificate of the client itself( generated by NodeOPCUA)
@paulsalibe, please dump the certificate you're trying to import from Siemens plc instead

paulsalibe, 2021-04-20:
> This certificate doesn't look like the PLC certificate ... It seems to be the certificate of the client itself( generated by NodeOPCUA)
> @paulsalibe, please dump the certificate you're trying to import from Siemens plc instead

Ok thank you - when you say dump, just delete from the trusted > certs file and try to connect with self signed certificate?

erossignon, 2021-04-20:
I mean :  run the following command from the trusted\cert folder where you have copied the certificate imported from the Siemens Plc,

   $ openssl x509  -in "BG300-0001192.168.0.84840_X.509 Certificate_3 (1).pem"  -text

paulsalibe, 2021-04-20:
> I mean : run the following command from the trusted\cert folder where you have copied the certificate imported from the Siemens Plc,
> 
> ```
>     openssl x509  -in "BG300-0001192.168.0.84840_X.509 Certificate_3 (1).pem"  -text
> ```

I had deleted it so I replaced it with the same file... here are the screenshots
[image]
[image]

paulsalibe, 2021-04-20:
And here are screenshots of my configuration in the node. When I set it up as the screenshots show it hangs on "Connecting" under the node. 

When I paste that path into both certificate and private key it shoots the error that this Issue is titled "TypeError: cannot read property 'position' of undefined" 

When I just try to use self signed certificates it also hangs on "Connecting"
[image]
[image]

mikakaraila, 2021-04-20:
Hmm, can you test with credentials checked x ?

Then second test without giving certificate, because Etienne has implemented certificate manager that will make Local certificate useless.

Certificate manager automates everything. I didn´t know that you are using "old" option (and without actually local private key file).

So test without anything like this:
[image]

mikakaraila, 2021-04-20:
And can you show UaExpert endpoint?
Siemens have normally resource path after endpoint port number like:
opc.tcp://hostname:4841/OpcUaServer_904

paulsalibe, 2021-04-20:
> Hmm, can you test with credentials checked x ?
> 
> Then second test without giving certificate, because Etienne has implemented certificate manager that will make Local certificate useless.
> 
> Certificate manager automates everything. I didn´t know that you are using "old" option (and without actually local private key file).
> 
> So test without anything like this:
> [image]
> 
> And endpoint:
> 
> [image]

I have tried it with just the credentials that way and it hangs on "Connecting" as well. Those credentials are specified within the Siemens TIA Portal correct?

Will send UAExpert endpoint shortly just have to go grab computer I have that on.

paulsalibe, 2021-04-20:
When I say hangs on "Connecting" this screenshot is what I mean.
Also the UAExpert endpoint is just opc.tcp://192.168.0.8:4840.... its configuration name is SIMATIC.S7-1200.OPC-UA.Application:BG300_0001 but not sure if that means anything. That endpoint opc.tcp://192.168.0.8:4840 has worked in the past with the "None" security option but now trying to secure it doesn't work.

[image]

paulsalibe, 2021-04-20:
With that configuration you suggested this is my node-red-log output if this means anything/helps...
[image]

mikakaraila, 2021-04-20:
Is client certificate trusted on server? It is the first step you must do.

What is node-red-contrib-opcua version?
And node-opcua version?

If you run node-red -v flow.json you will get more verbose, warning and error logs to console.

paulsalibe, 2021-04-20:
> Is client certificate trusted on server? It is the first step you must do.
> 
> What is node-red-contrib-opcua version?
> And node-opcua version?
> 
> If you run node-red -v flow.json you will get more verbose, warning and error logs to console.

0.2.113 version for node-red-contrib-opcua
0.2.33 version for node-opcua

The following is what I get when I run that command... my node red address is 192.168.0.110:1880 so I am unsure what that is telling me. Client certificate should be trusted on server if I can access the OPCUA Server through UAExpert correct?

[image]

mikakaraila, 2021-04-21:
I think I should update node-opcua to latest version and update certificate manager into the client.
Busy week, but perhaps during weekend.

mikakaraila, 2021-04-22:
I got help from Etienne and I published new version, please test.

paulsalibe, 2021-04-23:
Thank you for your help! Although, when I am going to update to the latest version of node-red-contrib-opcua it isn't working for some reason. From the command line it hangs in the spot shown on the screenshot. From the Palette Manager it just doesn't seem to work. I had never had a problem installing nodes before...

[image]

paulsalibe, 2021-04-23:
> Thank you for your help! Although, when I am going to update to the latest version of node-red-contrib-opcua it isn't working for some reason. From the command line it hangs in the spot shown on the [...]

And this seems to be exclusively to the node-red-contrib-opcua node, as I can upgrade others.

mikakaraila, 2021-04-24:
Hmm, can you try with exact version like:

D:\temp\NODERED-TEST>npm install node-red-contrib-opcua
npm WARN saveError ENOENT: no such file or directory, open 'D:\temp\package.json'
npm WARN enoent ENOENT: no such file or directory, open 'D:\temp\package.json'
npm WARN marked-terminal@4.1.1 requires a peer of marked@^1.0.0 || ^2.0.0 but none is installed. You must install peer dependencies yourself.
npm WARN temp No description
npm WARN temp No repository field.
npm WARN temp No README data
npm WARN temp No license field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@2.3.2 (node_modules\fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.3.2: wanted {"os":"darwin","arch":"any"} (current: {"os":"win32","arch":"x64"})

+ node-red-contrib-opcua@0.2.114
added 148 packages from 47 contributors, updated 1 package and audited 1392 packages in 26.68s

52 packages are looking for funding
  run `npm fund` for details

found 297 vulnerabilities (220 low, 22 moderate, 55 high)
  run `npm audit fix` to fix them, or `npm audit` for details

mikakaraila, 2021-04-24:
It could be that your npm is not "latest" or not in sync with node, some discussion about this @latest here:
https://github.com/npm/npm/issues/14570

mikakaraila, 2021-04-26:
Hmm, bad news for you. Node-opcua does not support node.js v10 anymore.

See: https://github.com/node-opcua/node-opcua/releases

It was dropped away in v2.40.0

paulsalibe, 2021-04-26:
So does that mean I won't be able to run the latest node-red-contrib-opcua off our Raspberry Pi?

mikakaraila, 2021-04-26:
No, you just need to update from node v10 to node v12. It should be enough.

https://medium.com/@thedyslexiccoder/how-to-update-nodejs-npm-on-a-raspberry-pi-4-da75cad4148c

erossignon, 2021-04-26:
The problem is that the latest version of npm doesn't support nodejs 10.0 either, as said in the first line of the `npm warn` message.  Nodes 10 has reached the it-s end of life and shall be updated to a newer version ( 14 is currently the recommended stable version)

paulsalibe, 2021-05-04:
> No, you just need to update from node v10 to node v12. It should be enough.
> 
> https://medium.com/@thedyslexiccoder/how-to-update-nodejs-npm-on-a-raspberry-pi-4-da75cad4148c

This walkthrough is great but I believe there are some steps missing.. When I follow this exactly it just reinstalls the version I had before and no npm. Is there a step missing where I tell it which version to upgrade to?

paulsalibe, 2021-05-05:
[image]

So I updated NodeJS and npm to the version shown in the screenshot, but now am seeing an error with updating through npm itself, also as shown in the screenshot.

paulsalibe, 2021-05-05:
[image]

And here is a screenshot from that log...

paulsalibe, 2021-05-05:
[image]

Then tried a few things I saw online and now its hanging where it was before...

paulsalibe, 2021-05-05:
npm WARN registry Unexpected warning for https://registry.npmjs.org/: Miscellaneous Warning EAI_AGAIN: request to https://registry.npmjs.org/node-red-contrib-opcua failed, reason: getaddrinfo EAI_AGAIN registry.npmjs.org 
npm WARN registry Using stale data from https://registry.npmjs.org/ due to a request error during revalidation.

got this above warning but then it installed. Using self-signed certificate and it is now working! Thanks a lot

mikakaraila, 2021-05-06:
Ok, your environment nodejs / npm were outdated and somehow npm registry was corrupted, but good it is working now.

closed 2021-05-06 by mikakaraila