mikakaraila/node-red-contrib-opcua #290: mysql node with opc-ua server

usman2k, 2021-04-19:
Hi, I am using "mysql" node and get data in node-red.
[image]

How can I transfer the "mysql" data to "OPC-UA" server.
[image]

mikakaraila, 2021-04-19:
As you have built address space for server, then you need to use function node:
1) Topic for the nodeId, map your SQL query variable to topic that is then server´s nodeId
2) Payload from the SQL server query as value into the msg

usman2k, 2021-04-19:
Thanks for your reply, Do you have some simple examples regarding this. just to add even one variable.
Regards,

mikakaraila, 2021-04-19:
Do you have msg that you have above as json? I can use it to make an example for you.

usman2k, 2021-04-19:
Thanks again.
[image]
and the values changing continuously.

mikakaraila, 2021-04-19:
Here example:

[image]

Rename from .txt back to .json
[SQL2OPCUA.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/6335819/SQL2OPCUA.txt)

usman2k, 2021-04-19:
Thanks for your kindness. Actually, in my sanario, the data comming from "mysql" and want to publish it in opc-ua.
But, what you made, it is reversed.

mikakaraila, 2021-04-19:
Use your mysql on lower left node.
Now it's injecting your mysql data as json payload.

usman2k, 2021-04-19:
Thank you for you kindness. It is working, Actually, in my "data format query" i set it as it shows only the last row of the "mysql". And now, I enter data in next row then it is only showing me the "mysql" msg.payload. While I am getting nothing from the opc-ua server.
[image]

mikakaraila, 2021-04-19:
Foreach or other loop for the array and done.

Here is screnshot from update variable "hinge"  with new value:

[image]

usman2k, 2021-04-19:
Yes, variable "hinge" is there but in new row i have new variable "latch" as shown in below picture.
[image]
[image]

mikakaraila, 2021-04-20:
This way, try to do something by yourself...
[image]

UaExpert with variables:
[image]

UaExpert with new values:
[image]

Inside function node build needed message inside while loop:
[SQL2OPCUA.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/6341072/SQL2OPCUA.txt)

usman2k, 2021-04-20:
Thanks again for your kind support, I will try it...

mikakaraila, 2021-04-26:
Hope you got everything working.

closed 2021-04-26 by mikakaraila

stesquality, 2021-06-17:
Thanks again, actually, I have tried, when I am adding a new variable in mysql then I need to reconnect the UaExpert client. Otherwise, the new variable not shows there. is there any possibility that it will automatically restart.

usman2k, 2021-06-17:
Thanks again, It is working, but when I am adding new variable then I need to reconnect the UaExpert client. Is it possible that the client will restart automatically.

mikakaraila, 2021-06-17:
UaExpert doesn't refresh address space.
Rebrowse should be enough.

usman2k, 2021-06-18:
Actually, I am using Tia Portal as a client and I need to import manually, the new added variable from node-red. Is it any possible that it automatically update like node value.

mikakaraila, 2021-06-18:
You should look Tia Portal manual or ask from Siemens.