mikakaraila/node-red-contrib-opcua #291: Security mode and security Policies

miguel-politecnico, 2021-04-21:
Is there a way to add some security to the server? or maybe is not implemented yet

I can manage to create a server, the address space and connect with the client. 

The problem is that I'm using the none endpoint without user and without security (no certificates, no sign, no encrypted). I'm wondering if these options are avaliable in some way or are not implemented?

[image]

mikakaraila, 2021-04-21:
Server uses certificate manager so you can use those endpoints. Server supports them.

miguel-politecnico, 2021-04-22:
I didn't ask the question properly. My mistake.

I already know that. But the point is you need to disable the unsecure endpoints in order to get a well configured server with security.

If the none endpoint is avaliable there is no security in the server, it doesn't matter if you have the other endpoints. Any client could connect to the server through the none endpoint

So the question is, are there any way to disable some endpoints sending a msg to the server node?

Thanks in advance for your time

mikakaraila, 2021-04-22:
OK then I should add option to allow only secure connections. I will take a look during wekend.

miguel-politecnico, 2021-04-22:
Maybe it's possible to choose the endpoints in the configuration tab of the node. I've been looking in the node-opcua api, and I think it's possible to choose the endpoints in an array property when you create the server. 
If I can help in any way, I don't have knowledge of building nodes in nodered from the scratch, but I think I can manage to make some modifications if I know the right files where to look at.

As I said before, thanks for your time.

mikakaraila, 2021-04-24:
It will be this way, still some code to write:
[image]

mikakaraila, 2021-04-24:
New version available, v0.2.115, please test.

miguel-politecnico, 2021-04-24:
I've installed the new version and made a few tests.

It seems to disable the endpoints propertly. This a screen capture with only Basic256Sha256 sign & Encript police enable.

[image]

The problem is that I'm still able to connect without installing the properly certificates in the server.  The node has a configuration checkbox to accept all the certificates, but with the checkbox disable is still accepting the connection.

I don't know exactly the folder where the certificates are or where to put the client cerficate. I've been looking at ~.config/node-red-opcua-nodejs/PKI/trusted/certs

I found the own certificate there. I tried to copy the client certificate in the rejected folder but in all of my test the client is always connecting with the server.

Another issue it's the anonymous autentication of the authentication that is enabled and it's not possible to add an username and password for the connection.

In any case it's a step on the right direction. 
Thanks for your time.

mikakaraila, 2021-04-25:
You must move or delete certificates from the trusted folder.
Allow anonymous I forgot to add checkbox. I will do for next version.

I will add also console/verbose output to show certificate manager certificate location path it will help people to find certificates.

mikakaraila, 2021-04-25:
New version 0.2.116, please test again.

AllowAnonymous:
[image]

Use verbose:
node-red -v EMPTY.json
[image]

miguel-politecnico, 2021-04-26:
I've updated to the new version. Several tests made.
First thing a typing bug in the message of rejected folder, it said Tejected folder. It's not important.

Relating to Allow anonymous user checkbox. It seems to work, if I uncheck the option and deploy my flow, the connection is rejected. 

[image]

The problem now is how to add a new user with privileges to make a connection. When you stablish the new connection the client asks for the credentials. This credentials need to be defined earlier in the server. Right now I didn't found a place where to do it in the server.

[image]

Regarding to certificates. I deleted all the files in the folders, and disable the checkbox Auto Accept Unknown Certificates. This way I think the server should reject the connection, but I can still connect to the server. I also tried to copy the certificate in the rejected folder, but I can also connect to the server.

[image]

Another issue is that when I enable Accept Unknown Certificates, the system doesn´t copy the certificate in the trusted folder. Other servers, when you enable that option, copy automatically every certificate of the client in the trusted folder.

Right now the configuration of the server node is a bit longer, I don't know if it is possible to sort in tabs or something like that, because all the options are in the tab properties. It is not a problem, and I don't know the implications about that in terms of work and time. 

[image]

Some other nodes that I tested have sorted in this way:

[image]

As I said before, first thing is to achieve that the node works properly. Then the cosmetic changes could be made.

In any case, thanks for your effort and time.

mikakaraila, 2021-04-26:
OK, I was expecting that you would like to define user credentials on server side.
It will need again more work. I will make new feature request issue for it otherwise one issue will be kilometers long and containing multiple items and will be complicated to track actual features.
So please consider to close this one and report each bug/feature as own issue.

Certificate manager is totally depending node-opcua implementation. I expect certificate should be in rejected or trusted folder.
NOT in both, I cannot say how it will prioritize it so DON´T COPY certificate, MOVE IT!

mikakaraila, 2021-04-26:
Next version will have users:
https://github.com/mikakaraila/node-red-contrib-opcua/issues/293

mikakaraila, 2021-04-26:
I will close this one as v0.2.117 contains now #293 

Please test and report findings as new issue

closed 2021-04-26 by mikakaraila