mikakaraila/node-red-contrib-opcua #294: Browse node produces "Encode Error" on certain nodeIDs

LeFish1, 2021-05-07:
** Short desciption of the issue **
When using the browse function I get a "Encode Error" on NodeIDs with the dataType "ExtensionObject".

** Additional desciption of the issue **
Here is my setup:
* Raspberry Pi (kernel version 5.10)
* CodeSys SoftSPS (version 4.0.1.0)
* NodeRED (version 1.3.4)

I am using the following flow for developing:
`[{"id":"a669c8af.193628","type":"inject","z":"dc6a0ab1.b2e238","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"","payloadType":"date","x":160,"y":220,"wires":[["c85a9001.f962f"]]},{"id":"6ed3e22a.31664c","type":"OpcUa-Browser","z":"dc6a0ab1.b2e238","endpoint":"8d25595.d7fb7a8","item":"","datatype":"","topic":"ns=4;s=|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG._fb_machine.OPCComm.Actors","items":[],"name":"","x":770,"y":220,"wires":[["35202209.8da09e"]]},{"id":"c85a9001.f962f","type":"change","z":"dc6a0ab1.b2e238","name":"def global.OPCComm","rules":[{"t":"set","p":"OPCComm","pt":"global","to":"{}","tot":"json"}],"action":"","property":"","from":"","to":"","reg":false,"x":340,"y":220,"wires":[["3a4b273a.49c198"]]},{"id":"3a4b273a.49c198","type":"change","z":"dc6a0ab1.b2e238","name":"Set OPC Browse Root","rules":[{"t":"set","p":"topic","pt":"msg","to":"ns=4;s=|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG._fb_machine.OPCComm.Actors","tot":"str"},{"t":"set","p":"OPCScope","pt":"msg","to":"OPCComm.Actors","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":560,"y":220,"wires":[["6ed3e22a.31664c"]]},{"id":"b0a7df3.2c5d12","type":"delay","z":"dc6a0ab1.b2e238","name":"","pauseType":"delay","timeout":"1","timeoutUnits":"seconds","rate":"1","nbRateUnits":"1","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":false,"x":1000,"y":220,"wires":[["ead9efa9.fc84a"]]},{"id":"35202209.8da09e","type":"switch","z":"dc6a0ab1.b2e238","name":"Block Init Start","property":"OPCScope","propertyType":"msg","rules":[{"t":"cont","v":"OPCComm","vt":"str"}],"checkall":"true","repair":false,"outputs":1,"x":900,"y":160,"wires":[["b0a7df3.2c5d12"]]},{"id":"ead9efa9.fc84a","type":"debug","z":"dc6a0ab1.b2e238","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":1310,"y":140,"wires":[]},{"id":"8a6c3a39.ca63b8","type":"split","z":"dc6a0ab1.b2e238","name":"","splt":"\\n","spltType":"str","arraySplt":1,"arraySpltType":"len","stream":false,"addname":"","x":1190,"y":220,"wires":[[]]},{"id":"8d25595.d7fb7a8","type":"OpcUa-Endpoint","endpoint":"opc.tcp://localhost:4840","secpol":"None","secmode":"None","login":false}]
`
When running the output of the browser node into the split node I get the following Error:
`TypeError: encodeObject Error: [Cannot read property 'schema' of undefined] Value: {
  item: {
    referenceTypeId: NodeId { identifierType: 1, value: 4004, namespace: 3 },
    isForward: true,
    nodeId: ExpandedNodeId {
      identifierType: 2,
      value: '|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG._fb_machine.OPCComm.Actors.PLC_to_OPC',
      namespace: 4,
      namespaceUri: null,
      serverIndex: 0
    },
    browseName: QualifiedName { namespaceIndex: 4, name: 'PLC_to_OPC' },
    displayName: LocalizedText { locale: 'en-Us', text: 'PLC_to_OPC' },
    nodeClass: 2,
    typeDefinition: ExpandedNodeId {
      identifierType: 1,
      value: 62,
      namespace: 0,
      namespaceUri: null,
      serverIndex: 0
    },
    value: Application.ST_PLC_to_OPC_Actors { testVar_PLC_to_OPC: false },
    dataType: 'ExtensionObject'
  }
}`

Within the scope ***._fb_machine.OPCComm.Actors I have two objects (PLC_to_OPC and OPC_to_PLC). Both have the same content (1 boolean variable)

Therefore I expect two messages after the split node.

The first message contains the proper object:
`{"item":{"referenceTypeId":"ns=3;i=4004","isForward":true,"nodeId":"ns=4;s=|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG._fb_machine.OPCComm.Actors.OPC_to_PLC","browseName":{"namespaceIndex":4,"name":"OPC_to_PLC"},"displayName":{"locale":"en-Us","text":"OPC_to_PLC"},"nodeClass":2,"typeDefinition":"ns=0;i=62","value":null,"dataType":"Null"}}
`
The second message contains the above error message.

The only visible difference is the dataType (null vs. ExtensionObject).

mikakaraila, 2021-05-07:
I expect node-opcua is supporting ExtensionObjects but as you have faced problem, it is not fully tested.

mikakaraila, 2021-05-09:
I mean node-opcua browse cannot encode ExtensionObjects. You should ask from node-opcua and @erossignon

mikakaraila, 2023-02-28:
I expect this one is not anymore hot topic. There are other open extension object issues open.