mikakaraila/node-red-contrib-opcua #296: Maximum Call Stack Size Exceeded

paulsalibe, 2021-05-17:
Hello, I have been receiving this error below that crashed node-red and needs a complete restart of the service. I have looked online and while there are similar issues, I have yet to find a real solution and none of the others seem to call out something with the certificate manager. It seems to mainly happen after a shutdown and when we turn it back on. Any help would be greatly appreciated!

`RangeError: Maximum call stack size exceeded
Exception in PromiseRejectCallback:
/home/pi/.node-red/node_modules/node-opcua-pki/dist/pki/certificate_manager.js:27
    return new (P || (P = Promise))(function (resolve, reject) {
           ^
RangeError: Maximum call stack size exceeded
Exception in PromiseRejectCallback:
/home/pi/.node-red/node_modules/node-opcua-certificate-manager/dist/certificate_manager.js:7
        function step(result) { result.done ? resolve(result.value) : adopt(result.value).then(fulfilled, rejected); }
                                                                                          ^
RangeError: Maximum call stack size exceeded
nodered.service: Main process exited, code=exited, status=1/FAILURE
nodered.service: Failed with result 'exit-code'.
`

mikakaraila, 2021-05-17:
It´s inside node-opcua, I think @erossignon should look this one. Some error handler loops until it causes stack overflow.

paulsalibe, 2021-05-18:
I have two READ clients and one WRITE client in my flows. They are typically somewhat reliable, but when I download and write to the PLC, it stops the OPC database. I believe it is during this time that Node-Red kind of "faults" out and does not know what to do. Maybe the client nodes are still trying to send, causing these handler loops? Because Node-Red crashes, but is still shown as "active" for about 4 minutes and then crashes and restarts. Is there a way to stop these Read/Write nodes before a download? or have them auto refresh? I have attached a screenshot for reference. This is what I see before the Error i posted above...
[image]

mikakaraila, 2021-05-18:
Cannot say more, but as it is running as service there can be other error messages.

Can you try to run it command line and repeat problem? Then there could be a better hint for the root cause.

jelmerthijsen, 2021-06-09:
I experience the same problem. Every night when the OPC server shuts down, node-red crashes. This problem appeared when i started to use a local certificate file with my client nodes. I use 11 client nodes. The error: "(node:10204) UnhandledPromiseRejectionWarning: Unhandled promise rejection" repeats itself and fills up the entire node-red console. This is what the console shows before node-red crashes:

[image]

I use version 0.2.221

mikakaraila, 2021-06-09:
I have to re-factor code as node-red is now parallel (older version runs nodes in order).

Node-opcua supports now re-connect by itself so I don´t have to make this kind of hack anymore.
I expect it loops until stack overflows and causes extra load to server as it creates new client & session.

mikakaraila, 2021-06-17:
OK, I will test a bit more and publish new version for testing.

mikakaraila, 2021-06-18:
Re-coding network error solves this one also, look #319

mikakaraila, 2021-06-18:
v0.2.225 should fix this, reopen if you find something.

closed 2021-06-18 by mikakaraila